istoreos/istoreos #950: 网络向导提示设备没有 WAN 口

0xJacky, 2023-08-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
网络向导-宽带拨号连接-提示因为您的设备没有 WAN 口，无法走本设置向导，但实际上是有 WAN 口的之前也配置过拨号。

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：网络向导-宽带拨号连接-提示因为您的设备没有 WAN 口，无法走本设置向导，但实际上是有 WAN 口的之前也配置过拨号。

### (2) 路由器型号和固件版本

路由器型号：超迷M1-8G-N100-4口
固件版本：iStoreOS 22.03.5 2023080410

### (3) 详细日志和/或截图

A：
[image]
[image]
[image]

jjm2473, 2023-08-18:
提供以下几个文件的内容：
* /etc/config/network 
* /etc/config/firewall

0xJacky, 2023-08-18:
/etc/config/network
```
root@iStoreOS:~# cat /etc/config/network

config interface 'loopback'
        option device 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'fd00:ba99:d8a5::/48'

config interface 'wan'
        option device 'eth0'
        option proto 'pppoe'
        option username '**********'
        option password '******'
        option peerdns '0'
        list dns '8.8.8.8'
        list dns ''

config interface 'wan6'
        option device 'eth3'
        option proto 'dhcpv6'

config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'eth1'
        list ports 'eth2'
        list ports 'eth3'

config interface 'lan'
        option device 'br-lan'
        option proto 'static'
        option ip6assign '60'
        option ipaddr '192.168.31.1'
        option netmask '255.255.255.0'
        option defaultroute '0'

config interface 'docker'
        option device 'docker0'
        option proto 'none'
        option auto '0'

```
/etc/config/firewall
```
root@iStoreOS:~# cat /etc/config/firewall

config defaults
        option syn_flood '1'
        option input 'ACCEPT'
        option output 'ACCEPT'
        option forward 'REJECT'
        option fullcone '1'

config zone
        option name 'lan'
        option input 'ACCEPT'
        option output 'ACCEPT'
        option forward 'ACCEPT'
        option auto_helper '0'
        list network 'lan'

config zone
        option name 'wan'
        option output 'ACCEPT'
        option forward 'REJECT'
[... 179 lines cut ...]

config redirect
        option dest 'lan'
        option target 'DNAT'
        option name 'ubuntu-ssh'
        list proto 'tcp'
        option src 'wan'
        option src_dport '10022'
        option dest_ip '192.168.31.152'
        option dest_port '22'

config redirect
        option dest 'lan'
        option target 'DNAT'
        option name 'wg'
        option src 'wan'
        option src_dport '51820'
        option dest_ip '192.168.31.2'
        option dest_port '51820'

config include 'openclash'
        option type 'script'
        option path '/var/etc/openclash.include'
        option reload '1'
```

jjm2473, 2023-08-19:
为什么你的wan跟wan6不是同一个eth0接口

0xJacky, 2023-08-19:
我就用过istore里的访问光猫/AP的插件，其他没动过

jjm2473, 2023-08-19:
我知道为什么了，你的network配置文件是有问题的，`list dns ''`这行是语法错误，是不是你自己手动改过配置文件，把这一行删掉。如果是从后台页面配置的dns，应该不可能出现这种空值。

0xJacky, 2023-08-19:
> 我知道为什么了，你的network配置文件是有问题的，`list dns ''`这行是语法错误，是不是你自己手动改过配置文件，把这一行删掉。如果是从后台页面配置的dns，应该不可能出现这种空值。

我知道了，我刚才把 list dns ''删掉之后就可以配置WAN口了，然后我用后台首页的DNS配置，选择自定义 DNS，然后只配置了一个DNS服务器，第二行留空，再保存就会出现这个问题。

然后如果把自定义DNS改回自动获取，此时获取到两个dns服务器ip，网络向导就能见得到WAN口。

还发现了一个问题，如果自定义DNS只填了一个IP，首页监控那里会显示DNS是自动获取的

jjm2473, 2023-08-21:
首页的dns配置不是只能填一个吗，怎么会有两行？

0xJacky, 2023-08-21:
> 

[image]

0xJacky, 2023-08-21:
> 首页显示的是当前网关所在接口的，如果你有多个网关，那可能就不准确了

确实是只有一个 WAN 口做网关

[image]

jjm2473, 2023-08-22:
OK，那估计是bug

jjm2473, 2023-08-22:
修复了，下个版本固件会发布

closed 2023-08-22 by jjm2473

uokooo, 2025-04-25:
能帮我也看一下吗，我是原来只配置了 Lan做旁路由，现在想用用来做主路由拨呈，添加了 Wan 口，网络向导里也显示无
WAN 口。
cat /etc/config/firewall

config defaults
        option syn_flood '1'
        option input 'ACCEPT'
        option output 'ACCEPT'
        option forward 'REJECT'
        option fullcone '1'

config zone
        option name 'lan'
        option input 'ACCEPT'
        option output 'ACCEPT'
        option forward 'ACCEPT'
        list network 'lan'

config zone
        option name 'wan'
        option input 'REJECT'
        option output 'ACCEPT'
        option forward 'REJECT'
        option masq '1'
        option mtu_fix '1'
        list network 'wan'
        list network 'wan6'
        list network 'wan0'

config forwarding
        option src 'lan'
        option dest 'wan'

config rule
        option name 'Allow-DHCP-Renew'
        option src 'wan'
        option proto 'udp'
        option dest_port '68'
        option target 'ACCEPT'
        option family 'ipv4'

config rule
        option name 'Allow-Ping'
        option src 'wan'
        option proto 'icmp'
        option icmp_type 'echo-request'
        option family 'ipv4'
        option target 'ACCEPT'

config rule
        option name 'Allow-IGMP'
        option src 'wan'
        option proto 'igmp'
        option family 'ipv4'
        option target 'ACCEPT'

config rule
        option name 'Allow-DHCPv6'
        option src 'wan'
        option proto 'udp'
        option dest_port '546'
        option family 'ipv6'
        option target 'ACCEPT'

config rule
        option name 'Allow-MLD'
        option src 'wan'
        option proto 'icmp'
        option src_ip 'fe80::/10'
        list icmp_type '130/0'
[... 136 lines cut ...]
        option name 'br-lan'
        option type 'bridge'
        list ports 'eth0'

config interface 'lan'
        option device 'br-lan'
        option proto 'static'
        option gateway '192.168.1.1'
        option metric '0'
        option ipaddr '192.168.1.4'
        option netmask '255.255.255.0'
        option peerdns '0'
        list dns '223.5.5.5'
        option defaultroute '0'

config interface 'wan0'
        option proto 'pppoe'
        option device 'eth1'
        option username '********'
        option password '********'
        option ipv6 'auto'
        option metric '1'

固件版本
iStoreOS 22.03.7 2025040711

jjm2473, 2025-06-06:
@uokooo `wan0`改成`wan`