istoreos/istoreos #951: 求大佬帮忙看一下

yaoyao19890513, 2023-08-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

opkg install 命令失败，代码 255

### (2) 路由器型号和固件版本

X86-64 物理机
iStoreOS 22.03.5 2023081110 / LuCI istoreos-22.03 branch git-23.103.23275-0193674

### (3) 详细日志和/或截图

Collected errors:
 * pkg_init_from_file: Malformed package file /tmp/upload.ipk.

a447948814, 2023-08-12:
我之前也是，无论什么包都升级失败，界面还会变成原生的，只能重启。解决办法：

yaoyao19890513, 2023-08-13:
> 我之前也是，无论什么包都升级失败，界面还会变成原生的，只能重启。解决办法： 1、软件中心下载作者的工具包。点击系统里，就看到了，选第二个修复系统软件。 [image] 2、终端里运行 opkg update opkg list-upgradable | cut -f 1 -d ' ' | xargs opkg upgrade 一键更新所有软件包，除了一个base的这个应该是作者定制的，无法更新。

谢谢   只要更新下就可以用了

closed 2023-08-13 by yaoyao19890513