istoreos/istoreos #958: 求助Docker被弄成这个样子了有什么办法可以恢复原样吗

1764364648, 2023-08-14:
[image]
被弄成了这样有谁知道怎么恢复原来的样子吗

rootphantomer, 2023-08-18:
[image]
首页的这个开关打开，再去看看

jjm2473, 2023-08-18:
看系统日志为什么dockerd无法启动

1764364648, 2023-08-18:
daemon.err dockerd[12532]: time="2023-08-18T17:12:23Z" level=warning msg="containerd config version `1` has been deprecated and will be removed in containerd v2.0, please switch to version `2`, see https://github.com/containerd/containerd/blob/main/docs/PLUGINS.md#version-header"
Sat Aug 19 01:12:23 2023 daemon.err modprobe: failed to find a module named aufs
Sat Aug 19 01:12:23 2023 daemon.err dockerd[12532]: time="2023-08-18T17:12:23.500267425Z" level=warning msg="failed to load plugin io.containerd.snapshotter.v1.devmapper" error="devmapper not configured"
Sat Aug 19 01:12:23 2023 daemon.err dockerd[12532]: time="2023-08-18T17:12:23.502166959Z" level=warning msg="could not use snapshotter devmapper in metadata plugin" error="devmapper not configured"
Sat Aug 19 01:12:23 2023 daemon.err dockerd[12532]: time="2023-08-18T17:12:23.516063336Z" level=error msg="failed to initialize a tracing processor \"otlp\"" error="no OpenTelemetry endpoint: skip plugin"
Sat Aug 19 01:12:23 2023 daemon.err dockerd[12532]: time="2023-08-18T17:12:23.592612134Z" level=error msg="failed to mount overlay: invalid argument" storage-driver=overlay2
Sat Aug 19 01:12:23 2023 daemon.err dockerd[12532]: time="2023-08-18T17:12:23.593080794Z" level=error msg="[graphdriver] prior storage driver overlay2 failed: driver not supported"
Sat Aug 19 01:12:23 2023 kern.err kernel: [22125.779880] overlayfs: filesystem on '/opt/docker/overlay2/check-overlayfs-support3355764298/upper' not supported as upperdir
Sat Aug 19 01:12:24 2023 daemon.err dockerd[12532]: failed to start daemon: error initializing graphdriver: driver not supported
Sat Aug 19 01:12:40 2023 daemon.err dockerd[12894]: time="2023-08-18T17:12:40Z" level=warning msg="containerd config version `1` has been deprecated and will be removed in containerd v2.0, please switch to version `2`, see https://github.com/containerd/containerd/blob/main/docs/PLUGINS.md#version-header"
Sat Aug 19 01:12:40 2023 daemon.err modprobe: failed to find a module named aufs
Sat Aug 19 01:12:40 2023 daemon.err dockerd[12894]: time="2023-08-18T17:12:40.355022021Z" level=warning msg="failed to load plugin io.containerd.snapshotter.v1.devmapper" error="devmapper not configured"
Sat Aug 19 01:12:40 2023 daemon.err dockerd[12894]: time="2023-08-18T17:12:40.356981233Z" level=warning msg="could not use snapshotter devmapper in metadata plugin" error="devmapper not configured"
Sat Aug 19 01:12:40 2023 daemon.err dockerd[12894]: time="2023-08-18T17:12:40.366803763Z" level=error msg="failed to initialize a tracing processor \"otlp\"" error="no OpenTelemetry endpoint: skip plugin"
Sat Aug 19 01:12:40 2023 daemon.err dockerd[12894]: time="2023-08-18T17:12:40.418888693Z" level=error msg="failed to mount overlay: invalid argument" storage-driver=overlay2
Sat Aug 19 01:12:40 2023 daemon.err dockerd[12894]: time="2023-08-18T17:12:40.419048106Z" level=error msg="[graphdriver] prior storage driver overlay2 failed: driver not supported"
Sat Aug 19 01:12:40 2023 kern.err kernel: [22142.608492] overlayfs: filesystem on '/opt/docker/overlay2/check-overlayfs-support2265559195/upper' not supported as upperdir
Sat Aug 19 01:12:41 2023 daemon.err dockerd[12894]: failed to start daem  这是日志

jjm2473, 2023-08-19:
你的docker数据目录是自己改的吗？默认肯定不是这个，恢复出厂设置

closed 2023-08-22 by jjm2473