istoreos/istoreos #960: r2s 老实断lan

simplysexyboy, 2023-08-15:
反馈bug/问题模板

### (1) 具体问题
A：老实断lan， 我按转了老版本 istoreos 也一样， 找不到什么原因。感觉是bug。

### (2) 路由器型号和固件版本

A：
model: FriendlyElec NanoPi R2S， 
rockchip/armv8， 
firmware version ：iStoreOS 22.03.5 2023080410 / LuCI istoreos-22.03 branch git-23.103.23275-0193674， 
kernel version： 5.10.176，

### (3) 详细日志和/或截图

A：就是断lan lan灯不亮

内核日志

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.10.176 (build@buildserver) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 2023070614) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 SMP PREEMPT Thu Aug 3 02:21:10 2023
[    0.000000] Machine model: FriendlyElec NanoPi R2S
[    0.000000] earlycon: uart8250 at MMIO32 0x00000000ff130000 (options '')
[    0.000000] printk: bootconsole [uart8250] enabled
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000200000-0x000000003fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000200000-0x000000003fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000200000-0x000000003fffffff]
[    0.000000] On node 0 totalpages: 261632
[    0.000000]   DMA zone: 4088 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 261632 pages, LIFO batch:63
[    0.000000] cma: Reserved 8 MiB at 0x000000003f400000
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 21 pages/cpu s47064 r8192 d30760 u86016
[    0.000000] pcpu-alloc: s47064 r8192 d30760 u86016 alloc=21*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 257544
[    0.000000] Kernel command line: console=ttyS2,1500000 earlycon=uart8250,mmio32,0xff130000 root=PARTUUID=5452574f-02 rw rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 1001708K/1046528K available (8638K kernel code, 2404K rwdata, 2424K rodata, 1728K init, 734K bss, 36628K reserved, 8192K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000009] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.001403] Console: colour dummy device 80x25
[    0.001876] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=240000)
[    0.002824] pid_max: default: 32768 minimum: 301
[    0.003485] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[... 461 lines cut ...]
[   23.125057] br-lan: port 1(eth1) entered forwarding state
[   23.204507] rk_gmac-dwmac ff540000.ethernet eth0: PHY [stmmac-0:00] driver [RTL8211E Gigabit Ethernet] (irq=POLL)
[   23.205460] rk_gmac-dwmac ff540000.ethernet: Set MAC tx_delay: 0x24 rx_delay: 0x18 for PHY 0x001cc915
[   23.213531] rk_gmac-dwmac ff540000.ethernet eth0: No Safety Features support found
[   23.214252] rk_gmac-dwmac ff540000.ethernet eth0: PTP not supported by HW
[   23.214868] rk_gmac-dwmac ff540000.ethernet eth0: configuring for phy/rgmii link mode
[   24.171340] br-lan: port 1(eth1) entered disabled state
[   26.334200] rk_gmac-dwmac ff540000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[   26.335060] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   26.832995] r8152 4-1:1.0 eth1: Promiscuous mode enabled
[   26.834107] r8152 4-1:1.0 eth1: carrier on
[   26.835272] br-lan: port 1(eth1) entered blocking state
[   26.835758] br-lan: port 1(eth1) entered forwarding state
[   30.455331] NFSD: Using /var/lib/nfs/v4recovery as the NFSv4 state recovery directory
[   30.456132] NFSD: Using legacy client tracking operations.
[   30.456628] NFSD: starting 10-second grace period (net f0000080)
[   38.202677] nf_conntrack: default automatic helper assignment has been turned off for security reasons and CT-based  firewall rule not found. Use the iptables CT target to attach helpers instead.
[  102.407288] overlayfs: upperdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
[  102.408466] overlayfs: workdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
[  102.409871] overlayfs: lowerdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
[  102.411041] overlayfs: lowerdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
[  103.531864] overlayfs: upperdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
[  103.533182] overlayfs: workdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
[  103.534359] overlayfs: lowerdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
[  103.535522] overlayfs: lowerdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.

jjm2473, 2023-08-18:
是不是电源的电流不够

simplysexyboy, 2023-08-25:
我换了网线 好像好了

jjm2473, 2023-08-25:
OK，那估计是网线质量不行或者有严重干扰

simplysexyboy, 2023-08-26:
也不是网线， 我家里有两个r2s，都是同一个卖家taobao买的，2台都换了网线还有只一台换了sd卡，结果没换过sd卡的机器端lan了。

大家都看啊， 千万不要用垃圾品牌sd卡，卖家给我的真垃圾 himory品牌 都没听说过。 

我买了sandisk 64g 140m/s 目前没什么问题了！

谢谢jjm

jjm2473, 2023-08-27:
神奇，sd卡会影响lan口，却不影响wan口和其他功能

closed 2023-08-27 by jjm2473