istoreos/istoreos #961: 手动安装design主题后报错

wangqingyuan233, 2023-08-15:
如题，安装并使用design主题后报错，并无法访问luci页面。
请求官方适配更多其他luci主题，比如design等。

以下为报错内容：
/usr/lib/lua/luci/template.lua:97: Failed to execute template 'sysauth'.
A runtime error occurred: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'header'.
A runtime error occurred: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'themes/design/header'.
A runtime error occurred: [string "/usr/lib/lua/luci/view/themes/design/header..."]:25: attempt to call field 'node_childs' (a nil value)
stack traceback:
	[string "/usr/lib/lua/luci/view/themes/design/header..."]:25: in main chunk
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:97: in function 'render'
	/usr/lib/lua/luci/dispatcher.lua:700: in function 'include'
	[string "/usr/lib/lua/luci/view/header.htm"]:3: in main chunk
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:97: in function 'render'
	/usr/lib/lua/luci/dispatcher.lua:700: in function 'include'
	[string "/usr/lib/lua/luci/view/sysauth.htm"]:1: in main chunk
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:97: in function </usr/lib/lua/luci/template.lua:85>
	(tail call): ?
	(tail call): ?
	/usr/lib/lua/luci/dispatcher.lua:483: in function </usr/lib/lua/luci/dispatcher.lua:482>

jjm2473, 2023-08-18:
应该让主题来适配。你用的固件什么版本，相应版本的openwrt能用这个主题吗？

closed 2023-08-18 by jjm2473

xiehuc, 2024-07-20:
design 选择 js 分支安装

xiehuc, 2024-07-20:
[image]