istoreos/istoreos #962: 不支持pcie模式的5g模块

andy524148733, 2023-08-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志和/或截图

A：

andy524148733, 2023-08-17:
不支持PCIe模式的5G模块

jjm2473, 2023-08-18:
什么型号的，知道驱动的名称吗？

andy524148733, 2023-08-18:
[image]
 这是广和通的

jjm2473, 2023-08-18:
执行 `lspci -nnk` ，把结果贴出来看看

jjm2473, 2023-08-18:
你的设备是什么型号？

jjm2473, 2023-08-18:
软路由的型号

andy524148733, 2023-08-18:
x86的

jjm2473, 2023-08-18:
看openwrt的项目，pcie的模块下个版本才有驱动，还不一定支持fibocom的模块

andy524148733, 2023-08-18:
有很多其他编译的op固件都支持pcie驱动了啊

jjm2473, 2023-08-18:
那你可以看看这个测试固件 https://github.com/istoreos/istoreos/issues/965 ，启动以后再软件包那边自己搜索驱动安装

jjm2473, 2023-08-22:
测试固件集成了MHI驱动

andy524148733, 2023-08-23:
[image]
需要第一个驱动才能用pcie

andy524148733, 2023-08-23:
[image]
移远的PCIe模块驱动

Siriling, 2023-08-23:
我来补充一下
移远以下模块支持使用PCIe模式
[image]

PCIe模式下一共有两种拨号模式QMI和MBIM，如何区分这两个呢？
默认为QMI，按照下面官方文档说明开启后便可使用MBIM了
[image]

jjm2473, 2023-08-23:
不会集成第三方的驱动，又不是什么通用的东西，如果内核提供的mhi不够用，那就不支持。

jjm2473, 2023-08-23:
测试固件已经加了mhi驱动，自己按文档配置

andy524148733, 2023-08-23:
是通用的 x55的pcie模块有这个驱动就都能用

jjm2473, 2023-08-23:
那你刷了测试固件没，有没有出现相关的设备？

andy524148733, 2023-08-23:
刷了
[image]
 没驱动

jjm2473, 2023-08-23:
不是看这个，是看/dev/里面有没有mhi相关的设备

andy524148733, 2023-08-23:
没驱动dev里面也有设备的

andy524148733, 2023-08-23:
[image]
有第二个驱动没第一个用不了

andy524148733, 2023-08-23:
[image]
加个这个驱动就可以用了

jjm2473, 2023-08-23:
pciutils只是用来看检查pci设备的，跟mhi或者拨号完全没关系。你执行 `lspci -nnk` 然后把输出贴出来看看，还有 `ls -l /dev/` 的输出

andy524148733, 2023-08-23:
[image]
[image]

andy524148733, 2023-08-23:
[image]

jjm2473, 2023-08-23:
贴图干嘛啊，复制文本不就行了

jjm2473, 2023-08-23:
你的第二张图没传完

andy524148733, 2023-08-23:
crw-------    1 root     root      239,   1 Aug 23 16:49 wwan0mbim0
crw-------    1 root     root      239,   0 Aug 23 16:49 wwan0qcdm0
crw-------    1 root     root      239,   2 Aug 23 16:49 wwan0qmi0
crw-rw-rw-    1 root     root        1,   5 Aug 23 16:49 zero

jjm2473, 2023-08-23:
你自己看不到上传失败的图吗

andy524148733, 2023-08-23:
就两张图啊

andy524148733, 2023-08-23:
[image]

andy524148733, 2023-08-23:
[image]

jjm2473, 2023-08-23:
算了，你还是执行 `lspci -nnk > /tmp/lspci.log; ls -l /dev/ >/tmp/lsdev.log`，然后想办法把/ tmp/lspci.log 和 /tmp/lsdev.log 发过来

andy524148733, 2023-08-23:
[image]

jjm2473, 2023-08-23:
不用管这个错误，把文件发过来就行了

jjm2473, 2023-08-23:
OK，再执行下 `ls -l /sys/dev/char/` 把输出发过来

jjm2473, 2023-08-23:
还有 `ls -l /sys/class/net/` 也发一下

andy524148733, 2023-08-23:
0:00:10.0/mhi0/mhi0_IP_HW0/net/mhi_hwip0
lrwxrwxrwx    1 root     root             0 Aug 23 16:49 siit0 -> ../../devices/virtual/net/siit0
lrwxrwxrwx    1 root     root             0 Aug 23 16:49 teql0 -> ../../devices/virtual/net/teql0
lrwxrwxrwx    1 root     root             0 Aug 23 16:49 tunl0 -> ../../devices/virtual/net/tunl0
root@iStoreOS:~#

andy524148733, 2023-08-23:
root@iStoreOS:~# ls -l /sys/class/net/
lrwxrwxrwx    1 root     root             0 Aug 23 16:49 br-lan -> ../../devices/virtual/net/br-lan
lrwxrwxrwx    1 root     root             0 Aug 23 16:49 docker0 -> ../../devices/virtual/net/docker0
lrwxrwxrwx    1 root     root             0 Aug 23 16:49 dummy0 -> ../../devices/virtual/net/dummy0
lrwxrwxrwx    1 root     root             0 Aug 23 16:49 erspan0 -> ../../devices/virtual/net/erspan0
lrwxrwxrwx    1 root     root             0 Aug 23 16:49 eth0 -> ../../devices/pci0000:00/0000:00:12.0/virtio1/net/eth0
lrwxrwxrwx    1 root     root             0 Aug 23 16:49 gre0 -> ../../devices/virtual/net/gre0
lrwxrwxrwx    1 root     root             0 Aug 23 16:49 gretap0 -> ../../devices/virtual/net/gretap0
lrwxrwxrwx    1 root     root             0 Aug 23 16:49 ip6gre0 -> ../../devices/virtual/net/ip6gre0
lrwxrwxrwx    1 root     root             0 Aug 23 16:49 ip6tnl0 -> ../../devices/virtual/net/ip6tnl0
lrwxrwxrwx    1 root     root             0 Aug 23 16:49 lo -> ../../devices/virtual/net/lo
lrwxrwxrwx    1 root     root             0 Aug 23 16:49 mhi_hwip0 -> ../../devices/pci0000:00/0000:00:10.0/mhi0/mhi0_IP_HW0/net/mhi_hwip0
lrwxrwxrwx    1 root     root             0 Aug 23 16:49 siit0 -> ../../devices/virtual/net/siit0
lrwxrwxrwx    1 root     root             0 Aug 23 16:49 teql0 -> ../../devices/virtual/net/teql0
lrwxrwxrwx    1 root     root             0 Aug 23 16:49 tunl0 -> ../../devices/virtual/net/tunl0
root@iStoreOS:~#

jjm2473, 2023-08-23:
那很清楚了啊，`/dev/wwan0` 开头的就是5G模块的tty控制台，可以执行AT命令，网络接口 `mhi_hwip0` 就是5G模块提供的网络接口。剩下的就是按文档执行AT命令和配置网络接口，AT命令看厂家的说明，网络接口通常只需要配置dhcp客户端就行了。

jjm2473, 2023-08-23:
如果想试试图形界面，打开后台页面的网络-接口，然后新增一个接口，协议选择“调试解调器管理器”，然后看看能不能选择到你的5G模块

andy524148733, 2023-08-23:
这个接口 mhi_hwip0是什么模式呢  qmi？mbim？

jjm2473, 2023-08-23:
@andy524148733 你没看到有后缀吗？wwan0mbim0 wwan0qmi0

jjm2473, 2023-08-23:
不知道什么模式，用AT命令查

jjm2473, 2023-08-23:
可能同时支持多个模式

jjm2473, 2023-08-23:
如果不知道怎么配置，那就用图形界面试试，毕竟图形界面是通用的，也就是“调试解调器管理器”

andy524148733, 2023-08-23:
调试解调器管理器没设备

andy524148733, 2023-08-23:
[image]
dao's倒是有这个接口

jjm2473, 2023-08-23:
还有可以在软件包那安装 `qmi-utils`，应该可以操作qmi控制设备

jjm2473, 2023-08-23:
还有，图形界面上有QMI协议，也可以试试

andy524148733, 2023-08-23:
图形界面QMI协议也是没设备

jjm2473, 2023-08-23:
哦，那个QMI只支持USB的

jjm2473, 2023-08-23:
执行`mmcli -L` 看看有没有列出设备

andy524148733, 2023-08-23:
root@iStoreOS:~# mmcli -L
No modems were found
root@iStoreOS:~#

jjm2473, 2023-08-23:
那“调试解调器管理器”确实不支持，只能看看文档怎么用命令拨号了

Siriling, 2023-08-23:
官方文档里面有写到，因为PCIe模式貌似只有QMI和MBIM两个拨号模式，这两个模式都需要使用他们的拨号工具，无法使用AT命令进行拨号。如果使用他们的拨号工具，必须要有他们的PCIE_MHI驱动才行（这个驱动经测试移远和广和通两家x55模块都能支持）。前面也看到出现的那些wwan0接口都不是AT接口，无法发送AT命令。

下面附上他们的PCIE拨号相关文档
[Quectel_RG50xQ&RM5xxQ系列_数据拨号应用指导_V1.0.pdf](https://github.com/istoreos/istoreos/files/12418139/Quectel_RG50xQ.RM5xxQ._._V1.0.pdf)
[FIBOCOM PCIe驱动集成及使用指南_Linux_V1.0.6.pdf](https://github.com/istoreos/istoreos/files/12418147/FIBOCOM.PCIe._Linux_V1.0.6.pdf)

jjm2473, 2023-08-23:
还有可以在软件包那安装 `qmi-utils`，就有qmicli了

Siriling, 2023-09-09:
# 说一下这几天的测试结果

**1.PCIE模式的5G模块可通过`qmi-utils`拨号成功**

**2.支持IPv4和IPv6，无法同时拨4和6，不清楚怎么配置**

**3.IPv6地址无法下发给内网设备。这个情况使用移远拨号工具`quectel_cm_5G`一样，不清楚怎么设置**

**4.IPv6拨号成功后还需要手动重启网络接口才能，OpenWrt才能获取到IPv6的地址**

**5.没有找到模块的AT接口，不知道是不是驱动的问题，如果用移远的PCIE驱动是有AT接口的**

**下面贴出PCIE模式的5G模块拨IPv6相关配置**

## 网口配置

```
config interface 'lan'
	option device 'br-lan'
	option proto 'static'
	option ipaddr '192.168.100.1'
	option netmask '255.255.255.0'
	option ip6assign '64'

config interface 'wwan_5g'
	option proto 'dhcp'
	option device 'mhi_hwip0'

config interface 'wwan6_5g'
	option proto 'dhcpv6'
	option device 'mhi_hwip0'
	option reqaddress 'try'
	option reqprefix 'auto'
```

## 查看模块接口

> root@iStoreOS:~# ls /dev/wwan0*
> /dev/wwan0mbim0  /dev/wwan0qcdm0  /dev/wwan0qmi

## 使用qmi-utils拨号

### profile文件

> root@iStoreOS:~# cat /etc/qmi-network.conf
> IP_TYPE=6
> APN=3gnet

### 启动网络连接

```shell
qmi-network --profile=/etc/qmi-network.conf /dev/wwan0qmi0 start
```

或

```shell
qmicli -d /dev/wwan0qmi0 --device-open-auto --wds-start-network="apn='3gnet',ip-type=6" --client-no-release-cid
```

> root@iStoreOS:~# qmi-network --profile=/etc/qmi-network.conf /dev/wwan0qmi0 start
> Loading profile at /etc/qmi-network.conf...
>     APN: 3gnet
>     APN user: unset
>     APN password: unset
>     qmi-proxy: no
>     IP_TYPE: 6
>     PROFILE: unset
> Checking data format with 'qmicli -d /dev/wwan0qmi0 --wda-get-data-format '...
> Device link layer protocol retrieved: raw-ip
> Getting expected data format with 'qmicli -d /dev/wwan0qmi0 --get-expected-data-format'...
> error: cannot get expected data format: Setting expected data format management is unsupported by the driver
> Expected link layer protocol not retrieved: kernel unsupported
> Updating device link layer protocol with 'qmicli -d /dev/wwan0qmi0 --wda-set-data-format=802-3 '...
> New device link layer protocol retrieved: raw-ip
> '  --client-no-release-cid '...d /dev/wwan0qmi0 --wds-start-network=apn='3gnet',ip-type='6
> Saving state at /tmp/qmi-network-state-wwan0qmi0... (CID: 44)
> Saving state at /tmp/qmi-network-state-wwan0qmi0... (PDH: 4161929664)
> Network started successfully

#### 已连接

> root@iStoreOS:~# ping -6 240c::6666
> PING 240c::6666 (240c::6666): 56 data bytes
> 64 bytes from 240c::6666: seq=0 ttl=52 time=81.904 ms
> 64 bytes from 240c::6666: seq=1 ttl=52 time=80.319 ms
> 64 bytes from 240c::6666: seq=2 ttl=52 time=80.169 ms
> 64 bytes from 240c::6666: seq=3 ttl=52 time=39.915 ms
> 64 bytes from 240c::6666: seq=4 ttl=52 time=79.797 ms

[image]

### 查看网络连接状态

```shell
qmi-network --profile=/etc/qmi-network.conf /dev/wwan0qmi0 status
```

或

```shell
qmicli -d /dev/wwan0qmi0 --wds-get-packet-service-status --client-cid=44 --client-no-release-cid
```

> root@iStoreOS:~# qmi-network --profile=/etc/qmi-network.conf /dev/wwan0qmi0 status
> Loading profile at /etc/qmi-network.conf...
>     APN: 3gnet
>     APN user: unset
>     APN password: unset
>     qmi-proxy: no
>     IP_TYPE: 6
>     PROFILE: unset
> Loading previous state from /tmp/qmi-network-state-wwan0qmi0...
>     Previous CID: 44
>     Previous PDH: 4161929664
> Getting status with 'qmicli -d /dev/wwan0qmi0 --wds-get-packet-service-status --client-cid=44 --client-no-release-cid '...
> Status: connected

### 停止网络连接

```shell
qmi-network --profile=/etc/qmi-network.conf /dev/wwan0qmi0 stop
```

或

```shell
qmicli -d /dev/wwan0qmi0 --wds-stop-network=4161929664 --client-cid=44
```

> root@iStoreOS:~# qmi-network --profile=/etc/qmi-network.conf /dev/wwan0qmi0 stop
> Loading profile at /etc/qmi-network.conf...
>     APN: 3gnet
>     APN user: unset
>     APN password: unset
>     qmi-proxy: no
>     IP_TYPE: 6
>     PROFILE: unset
> Loading previous state from /tmp/qmi-network-state-wwan0qmi0...
>     Previous CID: 44
>     Previous PDH: 4161929664
> Stopping network with 'qmicli -d /dev/wwan0qmi0 --wds-stop-network=4161929664 --client-cid=44 '...
> Network stopped successfully
> Clearing state at /tmp/qmi-network-state-wwan0qmi0...

Siriling, 2023-09-09:
补充一下，使用的5G模块是广和通FM150-AE-00

[27 more replies]