istoreos/istoreos #965: 23.05 固件测试反馈

jjm2473, 2023-08-18:
iStoreOS 适配此版本的主要目标是**支持 12 代 Intel CPU 的核显**，由于上游的 OpenWRT 23.05 还在 RC2 阶段，功能还未稳定下来，非 12 代 Intel CPU 用户不着急测试。测试固件下载地址[https://fw.koolcenter.com/iStoreOS/alpha/](https://fw.koolcenter.com/iStoreOS/alpha/)

目前已经在 N100 测试过。
理论上固件应该支持 N95/N100/N200/N305 等 CPU 的硬解。

### 注意事项：
1. 不要使用在线升级，不然会降级到线上正式版本
2. 如果之前使用其他版本固件，刷机前备份，尽量不保留配置升级。推荐使用 U盘多启动神器 Ventoy 来测试固件，无需刷机
3. 发现有其他问题可以在这个issue跟帖，不要单独开issue
4. 如果在这里反馈以后不想接收github的回复通知，可以在右边取消订阅

### 如何测试是否支持硬解
对于 N95 以上CPU，从iStore安装Jellyfin插件，并且部署以后，在Jellyfin的“控制台”-“播放”选择 QuickSync 转码器，之后的硬解选项都可以勾选上（一直到“色调映射”选项都可以勾选）。使用 Jellyfin 的网页播放器播放较高码率的 HEVC/AV1 等视频，并且将播放器的码率限制调低（例如 4Mbps），测试能否正常播放。如果硬解有问题，请先检查 Jellyfin 控制台显示的版本号在 10.8 以上。对于更老的CPU可以直接套用其他教程的做法，不是本次测试的重点。

### 请求适配其他 12 代 Intel CPU 核显：
先检查自己的CPU是不是有核显 https://en.wikipedia.org/wiki/Alder_Lake#List_of_12th_generation_Alder_Lake_processors
如果遇到不支持硬解的 12 代 CPU，又希望适配硬解，请在反馈时包括以下信息：
1. CPU 型号
2. 开机以后执行 `grep -F 'model name' /proc/cpuinfo | head -1; lspci -nn | grep -F VGA; dmesg | grep -F i915`，输出的内容

#### 一些测试视频下载地址
Jellyfish（H264/HEVC）： http://www.larmoire.info/jellyfish/
其他：https://test-videos.co.uk/

wuyufeng1989, 2023-08-18:
明天机器到了，测试下，牛

molin830, 2023-08-21:
小宝大佬 我cpu是i5 11320h 安装Jellyfin第一次播放会调用核显硬解 停播后 再播就没有硬件解码了 只有纯cpu解码 重新安装也不行 只能重刷固件才能第一次播放有硬件 之后就都不行了 问题这是什么原因

jjm2473, 2023-08-21:
@molin830 确认jellyfin的版本是10.8以上，执行`grep -F 'model name' /proc/cpuinfo | head -1; lspci -nn | grep -F VGA; dmesg | grep -F i915`把结果贴出来

molin830, 2023-08-21:
结果是这样的
# grep -F 'model name' /proc/cpuinfo | head -1; lspci -nn | grep -F VGA; dmesg | grep -F i915
model name      : 11th Gen Intel(R) Core(TM) i5-11320H @ 3.20GHz

molin830, 2023-08-21:
就好像第二次就没有调用的权限了

jjm2473, 2023-08-21:
@molin830 就只有一行？你用的是这个测试固件吗？

molin830, 2023-08-21:
我用的5.10的版本

molin830, 2023-08-21:
就只有这一行 显示cpu型号

jjm2473, 2023-08-21:
**不是这个测试固件的问题不要来这个帖子反馈！**

老固件不支持部分核显很正常。要么你就刷这个测试固件，有问题再反馈。

molin830, 2023-08-21:
嗯嗯 谢谢大佬的解答

molin830, 2023-08-21:
再请教大佬一个问题 我测试新固件 固件先装在u盘上测试 效果一样吗

jjm2473, 2023-08-21:
嗯。推荐用ventoy，可以测试多个不同版本的固件。

molin830, 2023-08-21:
好的 感谢大佬

molin830, 2023-08-21:
可以啦大佬 不过我是用Emby测试 因为我有购买Emby  Emby可以 说明Jellyfin也没问题 不过就好卡 应该是u盘当系统盘的原因 期待大佬的正式版

molin830, 2023-08-21:
还发现个bug iStore商店本地安装不成功 所以就没法安装那么啥

molin830, 2023-08-21:
[image]

jjm2473, 2023-08-21:
istore安装失败是因为上传的文件不完整，估计是luci的bug，需要排查下

jjm2473, 2023-08-22:
文件上传的bug已经修好了

benzo970210, 2023-08-25:
passwall 在这个版本上没发用

benzo970210, 2023-08-25:
我是n100 显示客户端与媒体格式不兼容

jjm2473, 2023-08-25:
@benzo970210 你看看jellyfin的版本是不是10.8以上，我测试过n100肯定没问题的

benzo970210, 2023-08-25:
> 

是10.8.10

benzo970210, 2023-08-25:
> 

没办法，忽略报错时候可以安装，但是把节点倒入之后用不起，我现在退回之前的版本了，一切正常

jjm2473, 2023-08-25:
看日志啊

Superskyli, 2023-08-27:
-------------------------------------------------------
 iStoreOS 22.03.5, 2023082509
 -------------------------------------------------------
root@HomeCenter:~# grep -F 'model name' /proc/cpuinfo | head -1; lspci -nn | grep -F VGA; dmesg | grep -F i915
model name      : 12th Gen Intel(R) Core(TM) i3-12100
00:02.0 VGA compatible controller [0300]: Intel Corporation Device [8086:4692] (rev 0c)
[   43.052404]  nf_log_ipv6 nf_log_ipv4 nf_log_common nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount ne2k_pci mmc_spi mlx5_core mlx4_en mlx4_core mdio_gpio mdio_bitbang macvlan lzo_rle lzo libcurve25519_generic libchacha kvm_intel kvm kaweth irqbypass iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ECN ipheth ip_tables igc iavf i915 i40e hso hid_cp2112 gpu_sched forcedeth ezusb ethoc et131x e1000e e100 dmx_usb crc7 crc_ccitt compat cm109 cls_flower chaoskey cdc_wdm cdc_acm br_netfilter bnx2x bnx2 be2net ax88796b atl2 atl1e atl1c atl1 alx 8390 8139too 8139cp fuse sch_teql sch_sfq sch_multiq sch_gred sch_fq sch_dsmark sch_codel em_text em_nbyte em_meta em_cmp act_simple act_pedit act_csum em_ipset cls_bpf act_bpf act_ctinfo act_connmark sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall

jjm2473, 2023-08-28:
@Superskyli 你这个不支持吗，把完整内核启动日志发出来

Superskyli, 2023-08-28:
> @Superskyli 你这个不支持吗，把完整内核启动日志发出来

重启了一下，这个是内核日志
[https://raw.githubusercontent.com/Superskyli/Superskyli.github.io/master/20230829.txt](url)

谢谢

jjm2473, 2023-08-28:
@Superskyli 你这个肯定不是完整日志

Superskyli, 2023-08-28:
> @Superskyli 你这个肯定不是完整日志

哦哦好的，我刚更新了一下，这下应该是完整的日志了。

jjm2473, 2023-08-29:
@Superskyli 你这个不是测试固件，你来这反馈，你来搞笑吗？你是没看issue的内容吗？测试固件的内核版本应该是5.15，你这个却是5.10

jjm2473, 2023-08-29:
12代CPU使用测试固件发现不支持核显再来反馈，使用正式固件不支持的话不要反馈到这里。

jjm2473, 2023-08-30:
passwall等插件的透明代理修复了

wgxatc, 2023-09-03:
model name      : Intel(R) Pentium(R) Gold 7505 @ 2.00GHz
00:02.0 VGA compatible controller [0300]: Intel Corporation Device [8086:9a78] (rev 01)
我这边好像没法硬解

jjkle, 2023-09-03:
amd的核显会支持么？

jjm2473, 2023-09-04:
@wgxatc 你是在使用这个测试固件吗？

jjm2473, 2023-09-04:
@jjkle 应该支持，自己测试，只是jellyfin的配置不一样

wgxatc, 2023-09-04:
> @wgxatc 你是在使用这个测试固件吗？

我知道了，我pve里还装了一个正式版，是不是显示的不准确？

wgxatc, 2023-09-04:
> @wgxatc 你是在使用这个测试固件吗？

是用的这个固件，感觉是不是7505核显有点问题？试了好几种方法都不行

jjm2473, 2023-09-04:
@wgxatc 我看了内核代码，[8086:9a78]是支持的，你是在这个测试固件里面执行的命令吗。

wgxatc, 2023-09-04:
> @wgxatc 我看了内核代码，[8086:9a78]是支持的，你是在这个测试固件里面执行的命令吗。
model name      : Intel(R) Pentium(R) Gold 7505 @ 2.00GHz
00:02.0 VGA compatible controller [0300]: Device [1234:1111] (rev 02)

wgxatc, 2023-09-04:
我还有一个那N305机器，运行代码是这样
root@iStoreOS:~# uname -r; grep -F 'model name' /proc/cpuinfo | head -1; lspci -nn | grep -F VGA; dmesg | grep -F i915
5.15.125
model name      : Intel(R) Core(TM) i3-N305
00:02.0 VGA compatible controller [0300]: Device [1234:1111] (rev 02)
00:10.0 VGA compatible controller [0300]: Intel Corporation Device [8086:46d0]
[    7.460621] [drm:intel_step_name [i915]] *ERROR* blitter: timed out waiting for forcewake ack request.
[    7.461170] i915 0000:00:10.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by intel_step_name+0x1172/0x4d70 [i915]
[    8.062356] [drm:intel_step_name [i915]] *ERROR* blitter: timed out waiting for forcewake ack request.
[    8.063022] i915 0000:00:10.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by intel_step_name+0x1172/0x4d70 [i915]
[    8.664269] [drm:intel_step_name [i915]] *ERROR* blitter: timed out waiting for forcewake ack request.
[    8.664926] i915 0000:00:10.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by intel_step_name+0x1172/0x4d70 [i915]
[    9.266219] [drm:intel_step_name [i915]] *ERROR* blitter: timed out waiting for forcewake ack request.
[    9.267003] i915 0000:00:10.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by intel_step_name+0x1172/0x4d70 [i915]
[    9.868387] [drm:intel_step_name [i915]] *ERROR* blitter: timed out waiting for forcewake ack request.
[    9.869105] i915 0000:00:10.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by intel_step_name+0x1172/0x4d70 [i915]
[   10.470451] [drm:intel_step_name [i915]] *ERROR* blitter: timed out waiting for forcewake ack request.
[   10.471196] i915 0000:00:10.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by intel_step_name+0x1172/0x4d70 [i915]
[   11.072562] [drm:intel_step_name [i915]] *ERROR* render: timed out waiting for forcewake ack to clear.
[   11.073328] i915 0000:00:10.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by intel_step_name+0x11b8/0x4d70 [i915]
[   11.674778] [drm:intel_step_name [i915]] *ERROR* blitter: timed out waiting for forcewake ack request.
[   11.675575] i915 0000:00:10.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by intel_step_name+0x1172/0x4d70 [i915]
[   12.276972] [drm:intel_step_name [i915]] *ERROR* blitter: timed out waiting for forcewake ack request.
[   12.277779] i915 0000:00:10.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by intel_step_name+0x1172/0x4d70 [i915]
[   12.879199] [drm:intel_step_name [i915]] *ERROR* blitter: timed out waiting for forcewake ack request.
[   12.880122] i915 0000:00:10.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by intel_step_name+0x1172/0x4d70 [i915]
[   13.481678] [drm:intel_step_name [i915]] *ERROR* blitter: timed out waiting for forcewake ack request.
[   13.482747] i915 0000:00:10.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by intel_step_name+0x1172/0x4d70 [i915]
[   14.084379] [drm:intel_step_name [i915]] *ERROR* render: timed out waiting for forcewake ack to clear.
[   14.085302] i915 0000:00:10.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by intel_step_name+0x11b8/0x4d70 [i915]
[   14.686839] [drm:intel_step_name [i915]] *ERROR* render: timed out waiting for forcewake ack to clear.
[   14.687646] i915 0000:00:10.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by intel_step_name+0x11b8/0x4d70 [i915]
[   15.289056] [drm:intel_step_name [i915]] *ERROR* render: timed out waiting for forcewake ack to clear.
[   15.289880] i915 0000:00:10.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by intel_step_name+0x11b8/0x4d70 [i915]
[   15.891350] [drm:intel_step_name [i915]] *ERROR* render: timed out waiting for forcewake ack to clear.
[   15.892502] i915 0000:00:10.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by intel_step_name+0x11b8/0x4d70 [i915]
[   16.494061] [drm:intel_step_name [i915]] *ERROR* render: timed out waiting for forcewake ack to clear.
[   16.494998] i915 0000:00:10.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by intel_step_name+0x11b8/0x4d70 [i915]
[   17.096529] [drm:intel_step_name [i915]] *ERROR* render: timed out waiting for forcewake ack to clear.
[   17.097693] i915 0000:00:10.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by intel_step_name+0x11b8/0x4d70 [i915]
[   17.699503] [drm:intel_step_name [i915]] *ERROR* blitter: timed out waiting for forcewake ack request.
[   17.700499] i915 0000:00:10.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by intel_step_name+0x1172/0x4d70 [i915]
[   18.302075] [drm:intel_step_name [i915]] *ERROR* blitter: timed out waiting for forcewake ack request.
[   18.303022] i915 0000:00:10.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by intel_step_name+0x1172/0x4d70 [i915]
[   18.904513] [drm:intel_step_name [i915]] *ERROR* blitter: timed out waiting for forcewake ack request.
[   18.905448] i915 0000:00:10.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by intel_step_name+0x1172/0x4d70 [i915]
[   19.507010] [drm:intel_step_name [i915]] *ERROR* render: timed out waiting for forcewake ack to clear.
[   19.508239] i915 0000:00:10.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by intel_step_name+0x11b8/0x4d70 [i915]
[   20.110067] [drm:intel_step_name [i915]] *ERROR* blitter: timed out waiting for forcewake ack request.
[   20.111107] i915 0000:00:10.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by intel_step_name+0x1172/0x4d70 [i915]
[   20.112874] i915 0000:00:10.0: [drm] *ERROR* rcs'0 reset request timed out: {request: 00000004, RESET_CTL: ffffffff}
[... cut ...]
[   24.341250] i915 0000:00:10.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by intel_step_name+0x1172/0x4d70 [i915]
[   24.942755] [drm:intel_step_name [i915]] *ERROR* blitter: timed out waiting for forcewake ack request.
[   24.943715] i915 0000:00:10.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by intel_step_name+0x1172/0x4d70 [i915]
[   24.944685] i915 0000:00:10.0: [drm] *ERROR* Failed to map the ggtt page table
[   25.112817] i915 0000:00:10.0: Device initialization failed (-12)
[   25.113834] i915: probe of 0000:00:10.0 failed with error -12

jjm2473, 2023-09-04:
@wgxatc 在7505上重新执行下命令看看。

n305这个是使用了虚拟机吗？你的宿主机的内核是否支持核显的sriov？有没有测试过非虚拟机环境？麻烦用物理机测试看看，推荐用ventoy。

wgxatc, 2023-09-04:
> @wgxatc 在7505上重新执行下命令看看。
> 
> n305这个是使用了虚拟机吗？你的宿主机的内核是否支持核显的sriov？有没有测试过非虚拟机环境？麻烦用物理机测试看看，推荐用ventoy。

7505执行命令就只显示这么多

BusyBox v1.36.1 (2023-08-15 15:52:35 UTC) built-in shell (ash)

                                                        
   ▀     ▄▄▄▄    ▄                          ▄▄▄▄   ▄▄▄▄ 
 ▄▄▄    █▀   ▀ ▄▄█▄▄   ▄▄▄    ▄ ▄▄   ▄▄▄   ▄▀  ▀▄ █▀   ▀
   █    ▀█▄▄▄    █    █▀ ▀█   █▀  ▀ █▀  █  █    █ ▀█▄▄▄ 
   █        ▀█   █    █   █   █     █▀▀▀▀  █    █     ▀█
 ▄▄█▄▄  ▀▄▄▄█▀   ▀▄▄  ▀█▄█▀   █     ▀█▄▄▀   █▄▄█  ▀▄▄▄█▀
                                                        
                                                        
 -------------------------------------------------------
 iStoreOS 23.05-SNAPSHOT, 2023082212
 -------------------------------------------------------
root@iStoreOS:~# uname -r; grep -F 'model name' /proc/cpuinfo | head -1; lspci -nn | grep -F VGA; dmesg | grep -F i915
5.15.125
model name      : Intel(R) Pentium(R) Gold 7505 @ 2.00GHz
00:02.0 VGA compatible controller [0300]: Device [1234:1111] (rev 02)
root@iStoreOS:~# 

i305后面有机会我再试试，现在里面虚拟机装了好多东西，暂时不想动，感谢

jjm2473, 2023-09-04:
@wgxatc 用u盘或者ventoy启动的话，就不会影响到虚拟机宿主系统了。

你这个7505很明显都没把核显直通到虚拟机里，肯定没发驱动核显啊。不过还是不要用虚拟机测试才准确，谁知道虚拟机到底有什么配置错误或者不支持显卡直通。

wgxatc, 2023-09-04:
> @wgxatc 用u盘或者ventoy启动的话，就不会影响到虚拟机宿主系统了。
> 
> 你这个7505很明显都没把核显直通到虚拟机里，肯定没法驱动核显啊，或者说核显能否驱动跟虚拟机内的iStoreOS没有任何关系了。不过还是不要用虚拟机测试才准确，谁知道虚拟机到底有什么配置错误或者不支持显卡直通。

好的，我后面把东西搬完用物理机试试，谢谢

ChrisLiu0343, 2023-09-09:
11 代 Intel CPU 核显同新支持AV1硬解

hxfgoodnow, 2023-09-15:
model name      : Intel(R) Xeon(R) CPU D-1581 @ 1.80GHz
0c:00.0 VGA compatible controller [0300]: Intel Corporation DG1 [Iris Xe Graphics] [8086:4908] (rev 01)

luyufan498, 2023-09-16:
测试jellyfin好像成功了，EMBY识别不到，奇怪

jjm2473, 2023-09-22:
@hxfgoodnow 最新的测试固件应该支持了

dnpl, 2023-09-23:
> > @wgxatc 用u盘或者ventoy启动的话，就不会影响到虚拟机宿主系统了。
> > 你这个7505很明显都没把核显直通到虚拟机里，肯定没法驱动核显啊，或者说核显能否驱动跟虚拟机内的iStoreOS没有任何关系了。不过还是不要用虚拟机测试才准确，谁知道虚拟机到底有什么配置错误或者不支持显卡直通。
> 
> 好的，我后面把东西搬完用物理机试试，谢谢

你好，你用7505物理测试了这个版本的固件了吗，是否支持核显

jjm2473, 2023-09-23:
@dnpl 最新的固件应该支持

hxfgoodnow, 2023-09-23:
> @hxfgoodnow最新的测试固件应该支持了

DG1还是无法硬解

root@iStoreOS:~# ls /dev/dri
card0       renderD128
root@iStoreOS:~# uname -r; grep -F 'model name' /proc/cpuinfo | head -1; lspci -nn | grep -F VGA; dmesg | grep -F i915
5.15.125
model name      : Intel(R) Xeon(R) CPU D-1581 @ 1.80GHz
0d:00.0 VGA compatible controller [0300]: Intel Corporation DG1 [Iris Xe Graphics] [8086:4908] (rev 01)
[   20.878980] i915 0000:0d:00.0: [drm] GT count: 1, enabled: 1
[   20.879341] i915 0000:0d:00.0: [drm] Unsupported device. GVT-g is disabled
[   20.879387] i915 0000:0d:00.0: vgaarb: deactivate vga console
[   20.879410] i915 0000:0d:00.0: [drm] Using Transparent Hugepages
[   20.883414] i915 0000:0d:00.0: [drm] Local memory IO size: 0x00000000fb800000
[   20.883418] i915 0000:0d:00.0: [drm] Local memory available: 0x00000000fb800000
[   20.967266] i915 0000:0d:00.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=io+mem:owns=none
[   20.993033] i915 0000:0d:00.0: [drm] Finished loading DMC firmware i915/dg1_dmc_ver2_02.bin (v2.2)
[   21.027512] i915 0000:0d:00.0: [drm] GT0: GuC firmware i915/dg1_guc_70.7.0.bin version 70.7.0
[   21.027519] i915 0000:0d:00.0: [drm] GT0: HuC firmware i915/dg1_huc_7.9.3.bin version 7.9.3
[   21.042059] i915 0000:0d:00.0: [drm] GT0: HuC: authenticated!
[   21.042764] i915 0000:0d:00.0: [drm] GT0: GUC: submission enabled
[   21.042768] i915 0000:0d:00.0: [drm] GT0: GUC: SLPC enabled
[   21.043016] i915 0000:0d:00.0: [drm] GT0: GUC: RC enabled
[   21.051595] i915 0000:0d:00.0: GT0: local0 bcs'0.0 clear bandwidth:54274 MB/s
[   21.052280] [drm] Initialized i915 1.6.0 20201103 for 0000:0d:00.0 on minor 0
[   21.153806] i915 0000:0d:00.0: [drm] Cannot find any crtc or sizes
[   21.154027] i915 0000:0d:00.0: [drm] Cannot find any crtc or sizes
[image]

[image]

jjm2473, 2023-09-24:
@dnpl 你是用istore里面的插件部署的jellyfin吗？检查下jellyfin的版本是不是10.8以上？

dnpl, 2023-09-25:
> @dnpl 你是用istore里面的插件部署的jellyfin吗？检查下jellyfin的版本是不是10.8以上？还有勾选jellyfin的低电压模式编码

我的问题已经解决，采用这个测试可以实现cpu 7505的硬解

jjm2473, 2023-09-29:
最新的正式版固件已经支持最新的 Intel 核显，详情 https://github.com/istoreos/istoreos/discussions/1022

这个测试版固件暂时没啥用了

jiaxinonly, 2023-10-16:
请问我安装的是eSir编译的openwrt (Powered by LuCI Master (git-23.082.27304-adda8b2) / ArgonTheme v1.7.3 / OpenWrt R23.3.3 GDQ V1[2023] Compiled by eSir) 是否有办法安装n100核显的驱动？

jjm2473, 2023-10-19:
@jiaxinonly 内核5.10以上应该都支持，但是需要编译，不太可能直接使用istoreos编译好的驱动，因为内核版本和内核配置都不一样。

wangqianyang, 2024-09-28:
/dev/dri
root@iStoreOS:/dev/dri# uname -r; grep -F 'model name' /proc/cpuinfo | head -1; lspci -nn | grep -F VGA; dmesg | grep -F i915
5.10.201
model name      : Intel(R) N100
00:02.0 VGA compatible controller [0300]: Intel Corporation Device [8086:46d1]
[    6.375668] enable_kmod.group i915-oot
[    7.286292] i915 0000:00:02.0: [drm] GT count: 1, enabled: 1
[    7.286486] i915 0000:00:02.0: [drm] Unsupported device. GVT-g is disabled
[    7.286525] i915 0000:00:02.0: [drm] Transparent Hugepage support is recommended for optimal performance on this platform!
[    7.291459] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adlp_dmc_ver2_16.bin (v2.16)
[    7.313999] i915 0000:00:02.0: [drm] GT0: GuC firmware i915/tgl_guc_70.9.1.bin version 70.9.1
[    7.314005] i915 0000:00:02.0: [drm] GT0: HuC firmware i915/tgl_huc_7.9.3.bin version 7.9.3
[    7.318809] i915 0000:00:02.0: [drm] GT0: HuC: authenticated!
[    7.319402] i915 0000:00:02.0: [drm] GT0: GUC: submission enabled
[    7.319404] i915 0000:00:02.0: [drm] GT0: GUC: SLPC enabled
[    7.319745] i915 0000:00:02.0: [drm] GT0: GUC: RC enabled
[    7.321615] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
[    8.824958] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[   10.277686] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[   11.734389] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
root@iStoreOS:/dev/dri#
[image]

wangqianyang, 2024-09-28:
这个输出正常吗，还是硬解不了