istoreos/istoreos #969: 局域网cifs挂载不稳定

lockfan, 2023-08-23:
设备为R4S，版本为2023080410
挂载局域网的samba，过一会就无法打开了，去界面上重新点一次保存，可以重新挂载，过一会再次掉线如此反复

内核日志：
[28718.031474] CIFS: Attempting to mount //192.168.3.252/图片
[39387.886398] CIFS: reconnect tcon failed rc = -6
[39612.952927] CIFS: Attempting to mount //192.168.3.252/图片
[44055.990992] CIFS: reconnect tcon failed rc = -2

终端上进入目录ls：
root@iStoreOS:/mnt/smb# ls
ls: .: No such file or directory

jjm2473, 2023-08-25:
看起来cifs还是有尝试重连的，是不是网络不稳定或者服务端掉线了

lockfan, 2023-08-26:
> 看起来cifs还是有尝试重连的，是不是网络不稳定或者服务端掉线了

应该可以排除网络不稳定或者NAS有问题，家里的mac book在期间也挂载了nas的smb，期间是没有出现掉线问题的

是否有方法让cifs记录更详细的日志用于排查问题呢？只是看到一个reconnect tcon failed基本无从下手

这几天尝试了另外几个编译的版本
官方最新23.05：挂载后45分钟左右掉线
LEDE分支最新R23.8.23：挂载45分支后掉线
DHDAXCW二次编译的lede分支（好像是5.4内核）：稳定挂载，运行了3 4天 都未掉过线

lockfan, 2023-08-26:
[39387.886398] CIFS: reconnect tcon failed rc = -6

jjm2473, 2023-08-27:
cifs是内核的功能，不知道怎么打印更详细的日志，可以试试修改printk级别，自己搜索一下。

5.4内核的R4S固件我们也有，其实就是上一个大版本 https://fw.koolcenter.com/iStoreOS/r4s/ ，`istoreos-21.02.3-*` 开头的都是

jjm2473, 2023-08-30:
你的r4s的内存是多大的

lockfan, 2023-09-01:
> 你的r4s的内存是多大的

4G内存

lockfan, 2023-09-01:
又花了点时间测了两种情况

情况1，就是这个issue里提到的，nas没断线，但openwrt这边掉线了，只需要/etc/init.d/cifs restart即可重新挂载

情况2，直接重启nas，openwrt里对于掉线的表现一致，但这种情况无法再次mount，命令行手动umount，然后输入mount -t cifs 后会得到special device does not exist，查了下mount.cifs的官方文档，cat /proc/fs/cifs/Stats，可以看到已掉线，且没有释放的session

1) \\192.168.3.252\图片	DISCONNECTED
SMBs: 129101
Bytes read: 0  Bytes written: 33226665
Open files: 0 total (local), 0 open on server
TreeConnects: 44232 total 44231 failed
TreeDisconnects: 0 total 0 failed
Creates: 28107 total 0 failed
Closes: 28107 total 0 failed
Flushes: 3 total 0 failed
Reads: 0 total 0 failed
Writes: 33 total 0 failed
Locks: 0 total 0 failed
IOCTLs: 1 total 1 failed
QueryDirectories: 958 total 0 failed
ChangeNotifies: 0 total 0 failed
QueryInfos: 27645 total 1 failed
SetInfos: 15 total 0 failed
OplockBreaks: 0 sent 0 failed

然后找到网上一个说法，umount -l 对于cifs是无效的，必须等到内核释放，或者直接重启openwrt

最终只好用了一个很脏的办法来曲线救国，先iptables指定一个空ip到nas，然后用mount -t cifs命令从这个新ip再次挂载，mount.cifs就会启动一个新的session

写了个脚本，放在crontab里每分钟1次：

#!/bin/bash

function rand(){
    min=$1
    max=$(($2-$min+1))
    num=$(date +%s%N)
    echo $(($num%$max+$min))
}

DIR="/mnt/smb/wj"
CDIR="192.168.3."
R=$(rand 150 170)
FAKEIP=${CDIR}${R}

if [ -d "$DIR" ]; then
  # Take action if $DIR exists. #
  echo "dir exists"
else
  echo "dir not exists,remounting to:"$FAKEIP
  umount /mnt/smb -l
  iptables -t nat -A OUTPUT -d $FAKEIP -j DNAT --to-destination 192.168.3.252
  mount -t cifs -o vers=3.0,iocharset=utf8,rw,sec=none //$FAKEIP/图片 /mnt/smb
fi

lockfan, 2023-09-01:
又花了点时间测了两种情况

情况1，就是这个issue里提到的，nas没断线，但openwrt这边掉线了，只需要/etc/init.d/cifs restart即可重新挂载

情况2，直接重启nas，openwrt里对于掉线的表现一致，但这种情况无法再次mount，命令行手动umount，然后输入mount -t cifs 后会得到special device does not exist，查了下mount.cifs的官方文档，cat /proc/fs/cifs/Stats，可以看到已掉线，且没有释放的session

1) \\192.168.3.252\图片	DISCONNECTED
   SMBs: 129101
   Bytes read: 0  Bytes written: 33226665
   Open files: 0 total (local), 0 open on server
   TreeConnects: 44232 total 44231 failed
   TreeDisconnects: 0 total 0 failed
   Creates: 28107 total 0 failed
   Closes: 28107 total 0 failed
   Flushes: 3 total 0 failed
   Reads: 0 total 0 failed
   Writes: 33 total 0 failed
   Locks: 0 total 0 failed
   IOCTLs: 1 total 1 failed
   QueryDirectories: 958 total 0 failed
   ChangeNotifies: 0 total 0 failed
   QueryInfos: 27645 total 1 failed
   SetInfos: 15 total 0 failed
   OplockBreaks: 0 sent 0 failed

然后找到网上一个说法，umount -l 对于cifs是无效的，必须等到内核释放，或者直接重启openwrt

最终只好用了一个很脏的办法来曲线救国，先iptables指定一个空ip到nas，然后用mount -t cifs命令从这个新ip再次挂载，mount.cifs就会启动一个新的session

写了个脚本，放在crontab里每分钟1次：
#!/bin/bash

OFFSET="/tmp/remount-offset"
touch $OFFSET
IP=$(cat $OFFSET)

if [ ! -n "$IP" ]; then
  IP=101
fi

DIR="/mnt/smb/wj"
CDIR="192.168.3."
FAKEIP=${CDIR}${IP}
echo $FAKEIP
if [ -d "$DIR" ]; then
  # Take action if $DIR exists. #
  echo "dir exists"
else
  echo "dir not exists,remounting to:"$FAKEIP
  umount /mnt/smb
  iptables -t nat -A OUTPUT -d $FAKEIP -j DNAT --to-destination 192.168.3.252
  mount -t cifs -o vers=3.0,user=guest,password=,iocharset=utf8,rw,sec=none //$FAKEIP/图片 /mnt/smb
  echo $(($IP+1)) > $OFFSET
fi

jjm2473, 2023-09-01:
umount应该有强制卸载的参数吧，试试`umount -f`

lockfan, 2023-09-07:
umount -f -a -t cifs -l

lockfan, 2023-09-08:
有点奇怪，直接重启nas后，之前挂载的session应该就变成僵尸了，尝试过umount -f -t cifs -l -a，虽然可以umount，但原来的ip再次挂载，一定会得到special device does not exist这个提示，过了3天再尝试去之前僵尸session的ip上mount，依然是不行，除非重启设备；

翻了一圈各种文档，没找到怎么让内核释放僵尸cifs session的方法，所以目前依然是上边那个曲线救国的脚本好使

jjm2473, 2023-09-08:
NFS的僵尸session很常见，smb我没见过。以前有同事开发过虚拟机作为沙箱扫描病毒的系统，病毒文件通过网络共享给沙箱，每扫完一个病毒就要重置虚拟机，使用NFS必定卡死，smb则没事。

我觉得你可以试试换成不带中文的共享，例如把“图片”换成“Pictures”。

Jasonsey, 2024-08-12:
所以有解决方案了吗？我使用r4s 使用最新系统，再次出现了这个问题