istoreos/istoreos #970: H68K输入istoreos无法启动，引导问题

lan-zz, 2023-08-24:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述
在iStoreOS刚支持H68K的时候我就将iStoreOS刷入了EMMC，是能正常使用的，后面陆陆续续版本在线升级到最新版都是没问题的，就昨天点了web端的重启以后就无法启动了，只有ETH1网卡灯常亮，没有其他状态，HDMI无输出信号。
### (1) 具体问题

[A：后面经过不同的尝试，下载了别人编译的支持H68K的Openwrt刷入TF卡可以启动正常使用的，然后又将Openwrt刷入EMMC也是可以正常启动使用的。]openwrt系统地址(https://www.right.com.cn/forum/forum.php?mod=viewthread&tid=8260147&highlight=h68k)

B：后面又尝试将iStoreOS输入TF与EMMC都无法正常启动，期间使用瑞芯微开发工具擦除flash也是没有用；然后又将Openwrt刷入TF卡启动系统，查看EMMC的数据是已经分好区有系统数据的，在EMMC里的IStoreOS的分区表类型是MBR，在TF卡里的Openwrt的分区表类型是GPT。

C：后面又尝试在Openwrt中使用命令
dd if=/mnt/mmcblk0p4/MiniLoaderAll.bin of=/dev/mmcblk1 bs=4M
dd if=/mnt/mmcblk0p4/istoreos-22.03.5-2023081110-h6xk-squashfs-combined.img of=/dev/mmcblk1来将IStoreOS直接写入EMMC看能否启动，结果还是无法启动；不知道MiniLoaderAll.bin启动引导是否有用。

      但是同样的方法，使用命令
dd if=/mnt/mmcblk0p4/openwrt_rk3568_h68k_R23.2.14_k6.1.18-Version666.img of=/dev/mmcblk1来将Openwrt写入EMMC就可以在EMMC中正常启动；启动引导没有加也是没有问题的。

### (2) 路由器型号和固件版本

A：软路由：hinlink H68K F版、4G+32G、带WiFi   
      IStoreOS固件版本：istoreos-22.03.5-2023081110-h6xk-squashfs-combined.img
      [[Openwrt固件版本：openwrt_rk3568_h68k_R23.2.14_k6.1.18-Version666]](https://pan.baidu.com/s/1L9T90hqB2fmZFKqTrksInw) 密码：6666

### (3) 详细日志和/或截图

A：
[image]

[image]
[image]

jjm2473, 2023-08-25:
大概率是因为你使用了不同的引导，也就是你的设备里刷了多个固件，例如tf卡是istoreos，emmc里面是其他固件。由于istore使用的是厂商的uboot，其他openwrt可能使用的是上游的原版uboot，两者不兼容。
总而言之，不能有不兼容的引导程序存在，不然大概率会启动不了。

closed 2023-08-25 by jjm2473

lan-zz, 2023-08-25:
>  一开始一直都是用的刷入EMMC的IStoreOS，没有插TF卡的，因为重启以后起不来才换TF卡尝试，后面按照官方教程，C to C 连电脑线刷方式重新往EMMC刷了最新的IStoreOS没有插TF卡，单独用EMMC也是无法启动。
  
>     既然是uboot的问题导致无法启动，该如何解决呢？
     IStoreOS采用的厂商的uboot是存在某个除EMMC的地方的，然后因为是硬件的损坏才导致无法启动还是怎么回事，或者能否自己改成上游的原版uboot来启动IStoreOS @jjm2473 🥲🥲

jjm2473, 2023-08-25:
不能使用原版uboot，你看那些使用原版uboot的固件都不支持npu。

本来应该是有tf卡的使用优先引导tf卡，但是rk的ATF并不是这样做的，然后uboot里面又是先查找tf卡，导致出现emmc的uboot引导了tf卡的系统，如果系统跟这个uboot不兼容，可能就会启动失败卡住等。

总之就是不要混用不同的uboot。

jjm2473, 2023-08-25:
也就是说，你把tf卡拔掉就一切正常了