istoreos/istoreos #972: 编译出现问题    ERROR: toolchain/gdb failed to build.

Dr-chen99, 2023-08-24:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：    ERROR: toolchain/gdb failed to build.

### (2) 路由器型号和固件版本

A：

### (3) 详细日志和/或截图

A：

jjm2473, 2023-08-25:
自己往前查日志到底哪个文件有问题

Dr-chen99, 2023-08-26:
> 自己往前查日志到底哪个文件有问题

Building backport-include/backport/autoconf.h ... done.
  CC [M]  /media/ta/Ext4/istoreos/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/drivers/net/wireless/broadcom/brcm80211/brcmfmac/common.o
In file included from /media/ta/Ext4/istoreos/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/drivers/net/wireless/broadcom/brcm80211/brcmfmac/common.c:20:
/media/ta/Ext4/istoreos/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/drivers/net/wireless/broadcom/brcm80211/brcmfmac/of.h:19:1: error: 'brcmf_of_fwnames' defined but not used [-Werror=unused-function]
   19 | brcmf_of_fwnames(struct device *dev, u32 *map_count)
      | ^~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[12]: *** [scripts/Makefile.build:289: /media/ta/Ext4/istoreos/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/drivers/net/wireless/broadcom/brcm80211/brcmfmac/common.o] Error 1
make[11]: *** [scripts/Makefile.build:552: /media/ta/Ext4/istoreos/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/drivers/net/wireless/broadcom/brcm80211/brcmfmac] Error 2
make[10]: *** [scripts/Makefile.build:552: /media/ta/Ext4/istoreos/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/drivers/net/wireless/broadcom/brcm80211] Error 2
make[9]: *** [scripts/Makefile.build:552: /media/ta/Ext4/istoreos/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/drivers/net/wireless/broadcom] Error 2
make[8]: *** [scripts/Makefile.build:552: /media/ta/Ext4/istoreos/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/drivers/net/wireless] Error 2
make[7]: *** [Makefile:1906: /media/ta/Ext4/istoreos/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24] Error 2
make[6]: *** [Makefile.build:13: modules] Error 2
make[5]: *** [Makefile.real:93: modules] Error 2
make[4]: *** [Makefile:121: modules] Error 2
make[4]: Leaving directory '/media/ta/Ext4/istoreos/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24'
make[3]: *** [Makefile:401: /media/ta/Ext4/istoreos/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/.built] Error 2
make[3]: Leaving directory '/media/ta/Ext4/istoreos/package/kernel/mac80211'
time: package/kernel/mac80211/regular/compile#1.83#0.62#3.00
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[2]: *** [package/Makefile:120: package/kernel/mac80211/compile] Error 1
make[2]: Leaving directory '/media/ta/Ext4/istoreos'
make[1]: *** [package/Makefile:114: /media/ta/Ext4/istoreos/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/media/ta/Ext4/istoreos'
make: *** [/media/ta/Ext4/istoreos/include/toplevel.mk:236：world] 错误 2

jjm2473, 2023-08-27:
你用的是什么分支？

jjm2473, 2023-08-27:
怀疑是因为使用的gcc版本有问题，因为那是个警告，只是把警告当成错误。 你用这里提供的config文件和feed文件重新编译看看   https://fw.koolcenter.com/iStoreOS/alpha/x86_64_efi/

Dr-chen99, 2023-08-27:
> 你用的是什么分支？

istoreos-23.05

Dr-chen99, 2023-08-27:
> 怀疑是因为使用的gcc版本有问题，因为那是个警告，只是把警告当成错误。 你用这里提供的config文件和feed文件重新编译看看 https://fw.koolcenter.com/iStoreOS/alpha/x86_64_efi/

晚点试试

Dr-chen99, 2023-08-27:
> 怀疑是因为使用的gcc版本有问题，因为那是个警告，只是把警告当成错误。 你用这里提供的config文件和feed文件重新编译看看 https://fw.koolcenter.com/iStoreOS/alpha/x86_64_efi/

不编译brcmfmac驱动没事，我用的istoreos-23.05这个分支。东西还是一样，为啥我用的官网最新openwrt main分支没事，加这个驱动brcmfmac

Dr-chen99, 2023-08-27:
>

closed 2023-08-27 by Dr-chen99

jjm2473, 2023-08-28:
那你试试openwrt的23.05分支

jjm2473, 2023-08-28:
我这里是有编译kmod-brcmfmac的，也没报错。
[image]