istoreos/istoreos #975: 更新固件后无法使用iStore opkg报错 Segmentation fault

break123aqwq, 2023-08-26:
## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ X] 没有类似的issue

## 2. 详细叙述
更新固件后无法使用iStore opkg报错
更新失败: Segmentation fault
### (1) 具体问题
A：
[image]
[image]
任何操作都会提示 Segmentation fault
重启后以及升级最新固件都无法解决问题
### (2) 路由器型号和固件版本
A：
[image]

### (3) 详细日志和/或截图
A：

break123aqwq, 2023-08-26:
[image]

break123aqwq, 2023-08-26:
[image]

jjm2473, 2023-08-28:
你是不是添加了非x86的软件源？

closed 2023-09-14 by jjm2473