istoreos/istoreos #977: istoreos-23.05自编译防火墙问题

Dr-chen99, 2023-08-27:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
防火墙显示   #旧版本与系统上存在旧版 iptables 规则。 不鼓励混合使用 iptables 和 nftables 规则，这可能会导致流量过滤不完整。

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志和/或截图

A：

jjm2473, 2023-08-28:
这是因为你混用了nftables和iptables导致的。istoreos默认使用的是iptables，因为兼容性好。如果不想自己查找哪些包依赖了nftables，那从这里下载对应机型的config文件  https://fw.koolcenter.com/iStoreOS/ ，编译的时候就可以完全去除掉nftables

jjm2473, 2023-08-28:
如果你想用nftables也可以，把所有iptables相关的软件包切换到nftables的版本，例如upnp是有两个版本的。

closed 2023-08-28 by jjm2473

Dr-chen99, 2023-08-28:
> 如果你想用nftables也可以，把所有iptables相关的软件包切换到nftables的版本，例如upnp是有两个版本的。

pass2和pass问题解决了

jjm2473, 2023-08-28:
不想折腾的话，就直接使用我们的config文件，之后编译其他插件时取消掉nftables相关的功能就行了

Dr-chen99, 2023-08-28:
> 不想折腾的话，就直接使用我们的config文件，之后编译其他插件时取消掉nftables相关的功能就行了

这个问题解决了，在pass2不够选那个就好