istoreos/istoreos #981: 为什么升级了最新istoreos

jasonnmy, 2023-08-29:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
istoreso做旁路由出现了tcp被拦截情况，外网无法转口转发到内网，抓包看了下，tcp请求来，电脑返回了，但是到旁路由被中断了，发送端没收到回复，然后一直tcp retransmission。把旁路由换其他系统没问题，直连主路由也没问题

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本
iStoreOS 21.02.3, 2023022415 和22.03.5 2023082509
软路由
A：

### (3) 详细日志和/或截图

A：

jasonnmy, 2023-08-29:
[image]

jjm2473, 2023-08-30:
是否使用了小米路由器作为AP？

jasonnmy, 2023-08-30:
查到原因了防火墙的问题，我之前没问题，突然升级某个版本后出现，好像是开了IP 动态伪装

closed 2023-08-30 by jasonnmy

jjm2473, 2023-08-30:
按道理来说，做旁路由的话，根本就不会有流量经过了，因为主路由会直接往内网设备发。

jjm2473, 2023-08-30:
你的主路由的端口转发是怎么做的，DMZ到旁路由吗，还是直接转发到目标内网设备

jjm2473, 2023-08-30:
如果不开lan口的IP动态伪装的话，旁路由的docker容器没法上网

jasonnmy, 2023-08-30:
爱快的直接端口转发，没开dmz

jjm2473, 2023-08-30:
那理论上流量都不需要旁路由转发了啊，你抓包看到的tcp发送失败的目的ip是什么

jjm2473, 2023-08-30:
爱快的端口转发是没改源ip吗？

jasonnmy, 2023-08-30:
什么源ip

jjm2473, 2023-08-30:
端口转发不就是NAT吗，至少目标ip要改吧，那源ip改了吗，如果源ip改成了爱快的ip，那内网设备回复的时候不就直接往爱快发了吗，怎么也不会经过旁路由

jasonnmy, 2023-08-30:
ip为爱快和旁路由都能转发，其他设备就不行

jjm2473, 2023-08-30:
所以说看看抓包的失败tcp的目的ip是什么啊

jasonnmy, 2023-08-30:
等会

jasonnmy, 2023-08-30:
[image]

jjm2473, 2023-08-30:
为什么你的电脑收到的源ip不是局域网ip？正常的端口转发会把源ip也改掉，你还是看看爱快的端口转发是不是有什么特殊设置吧

jasonnmy, 2023-08-30:
[image]

jjm2473, 2023-08-30:
参考下这个 https://bbs.ikuai8.com/thread-139586-1-1.html ，可能爱快的端口转发不是通常的使用CT维护连接

jasonnmy, 2023-08-30:
可能爱快不一样吧