istoreos/istoreos #982: AMD首页不显示温度

xiaoermeizai, 2023-08-29:
-------------------------------------------------------
 iStoreOS 22.03.5, 2023082509
 -------------------------------------------------------
root@iStoreOS:~# sensors
drivetemp-scsi-0-0
Adapter: SCSI adapter
temp1:        +45.0°C  

fam15h_power-pci-00c4
Adapter: PCI adapter
power1:      315.06 mW (interval =   0.01 s, crit =   5.99 W)

radeon-pci-0008
Adapter: PCI adapter
temp1:        +53.0°C  (crit = +120.0°C, hyst = +90.0°C)

k10temp-pci-00c3
Adapter: PCI adapter
temp1:        +51.8°C  (high = +70.0°C)
                       (crit = +105.0°C, hyst = +104.0°C)

-------------------------------------------------------------------
root@iStoreOS:~# cat /sys/class/hwmon/hwmon3/temp1_input
53000
4个传感器都可以正常读取
-----------------------------------------------------------------
root@iStoreOS:~# cat /sys/class/thermal/thermal_zone0/temp
cat: can't open '/sys/class/thermal/thermal_zone0/temp': No such file or directory
没有thermal_zone文件夹，只有cooling_device
-----------------------------------------------------------------
辛苦修复

jjm2473, 2023-08-30:
所以，你觉得哪个是cpu的温度？

xiaoermeizai, 2023-08-30:
> 那么，你觉得cpu的温度是哪个？

k10temp-pci-00c3啊    我这里也就是/hwmon3/temp1_input

请问首页读取/sys/class/thermal/thermal_zone0/temp的代码文件在哪个路径，我自己替换成/hwmon3/temp1_input也是可以

jjm2473, 2023-08-30:
你以为写死/hwmon3/temp1_input就行了？hwmon3这些都是内核按顺序生成的，你怎么知道不是hwmon0呢？需要判断条件。例如你说k10temp就是cpu温度，那么/hwmon3/里面哪个文件能找到关联呢？

xiaoermeizai, 2023-08-30:
> 你以为写死/hwmon3/temp1_input就行了？hwmon3这些都是内核按顺序生成的，你怎么知道不是hwmon0呢？需要判断条件。例如你说k10temp就是cpu温度，那么/hwmon3/里面哪个文件能找到关联呢？

我在终端里每秒打印sensors中的k10temp和/hwmon3/temp1_input，发现温度是一致的。我的意思的我自己手动改一下   整体逻辑判断你们团队可以长远计议   辛苦辛苦

jjm2473, 2023-08-30:
目前没有amd的测试机，所以需要你提供线索。不打算依赖sensor，太大了。你看看/hwmon3/里面哪个文件包含k10temp，或者/hwmon3/里面是否有文件软连接到k10temp什么的。或者你提供一个远程访问的方式给我看看。

jjm2473, 2023-08-30:
或者hwmon3本身是不是一个软连接？

xiaoermeizai, 2023-08-30:
/sys/class/hwmon/下的0-3  一共四个  都是软连接   我正在down下来原目录便于你们研究  稍等

jjm2473, 2023-08-30:
根据这个文章 https://blog.csdn.net/ColdZera/article/details/119908561 ，是不是/hwmon3/name包含了k10temp

xiaoermeizai, 2023-08-30:
> 根据这个文章 https://blog.csdn.net/ColdZera/article/details/119908561 ，是不是/hwmon3/name包含了k10temp

是的   name里就是组件名称   和sensors里的组件名称也是对应的

xiaoermeizai, 2023-08-30:
没有这个/sys/class/thermal/thermal_zone0/路径，应该是amd在初始化的时候的问题

xiaoermeizai, 2023-08-30:
抱歉，/hwmon1/name是k10temp，我在朋友的amd小主机上也是hwmon1，不知道是不是巧合，或许amd机型都可以写死hwmon1

jjm2473, 2023-08-30:
那你看看/hwmon1/里面的哪个temp*_input是对应cpu温度

xiaoermeizai, 2023-08-30:
root@iStoreOS:~# cat /sys/class/hwmon/hwmon1/name
k10temp

3是集显  其实和cpu温度无限接近

xiaoermeizai, 2023-08-30:
> 那你看看/hwmon1/里面的哪个temp*_input是对应cpu温度

root@iStoreOS:~# cat /sys/class/hwmon/hwmon1/temp1_input
53750
------------------------
hwmon1下面只有temp1_input没有234...
或许AMD的一些机型不分核心显示温度

我是双核  朋友是四核   都只有temp1_input 而且是hwmon1   不知道AMD是不是共性

xiaoermeizai, 2023-08-30:
hwmon*应该是代表不同元器件
temp*_input应该是代表不同核心

不过我这里两台AMD无论是读取sensors还是hwmon，都无法读取到不同核心分开的温度数据

xiaoermeizai, 2023-08-30:
root@iStoreOS:/sys/class/hwmon# ls -al
drwxr-xr-x    2 root     root             0 Aug 29 19:36 .
drwxr-xr-x   74 root     root             0 Aug 29 19:35 ..
lrwxrwxrwx    1 root     root             0 Aug 29 19:36 hwmon0 -> ../../devices/pci0000:00/0000:00:18.4/hwmon/hwmon0
lrwxrwxrwx    1 root     root             0 Aug 29 19:36 hwmon1 -> ../../devices/pci0000:00/0000:00:18.3/hwmon/hwmon1
lrwxrwxrwx    1 root     root             0 Aug 30 08:09 hwmon2 -> ../../devices/pci0000:00/0000:00:11.0/ata1/host0/target0:0:0/0:0:0:0/hwmon/hwmon2
lrwxrwxrwx    1 root     root             0 Aug 30 08:09 hwmon3 -> ../../devices/pci0000:00/0000:00:01.0/hwmon/hwmon3

xiaoermeizai, 2023-08-30:
root@iStoreOS:/sys/devices/pci0000:00/0000:00:18.3/hwmon/hwmon1# ls -al
drwxr-xr-x    3 root     root             0 Aug 29 19:36 .
drwxr-xr-x    3 root     root             0 Aug 29 19:36 ..
lrwxrwxrwx    1 root     root             0 Aug 30 08:09 device -> ../../../0000:00:18.3
-r--r--r--    1 root     root          4096 Aug 30 08:09 name
drwxr-xr-x    2 root     root             0 Aug 30 10:49 power
lrwxrwxrwx    1 root     root             0 Aug 30 10:49 subsystem -> ../../../../../class/hwmon
-r--r--r--    1 root     root          4096 Aug 30 08:09 temp1_crit
-r--r--r--    1 root     root          4096 Aug 30 08:09 temp1_crit_hyst
-r--r--r--    1 root     root          4096 Aug 30 08:09 temp1_input
-r--r--r--    1 root     root          4096 Aug 30 08:09 temp1_max
-rw-r--r--    1 root     root          4096 Aug 30 10:49 uevent

xiaoermeizai, 2023-08-30:
root@iStoreOS:~# ls -lL /sys/class/thermal
drwxr-xr-x    3 root     root             0 Aug 29 19:36 cooling_device0
drwxr-xr-x    3 root     root             0 Aug 29 19:36 cooling_device1
-----------
两台AMD都是这样，没有thermal_zone*

jjm2473, 2023-08-30:
ok，我等下编一个quickstart给你试试

jjm2473, 2023-08-30:
软路由上下载这个文件，解压运行，运行之前先执行`/etc/init.d/quickstart stop`停止系统里面的quickstart。
[quickstart.amd64.zip](https://github.com/istoreos/istoreos/files/12471549/quickstart.amd64.zip)

xiaoermeizai, 2023-08-30:
> ok，我等下编一个quickstart给你试试

辛苦小哥

xiaoermeizai, 2023-08-30:
我一会吃完饭去试试

xiaoermeizai, 2023-08-30:
刚试了两遍没成功   温度进度条一直还是0

jjm2473, 2023-08-30:
都还没跟你说怎么运行呢

jjm2473, 2023-08-30:
`STTRACE=main,service ./quickstart.amd64 serve`

xiaoermeizai, 2023-08-30:
root@iStoreOS:~# cat /sys/class/hwmon/hwmon1/temp1_input
53750
---------
读取还是正常的

> 都还没跟你说怎么运行呢

啊？  不是stop之后直接解压运行么

xiaoermeizai, 2023-08-30:
> 都还没跟你说怎么运行呢
啊？ 不是stop之后直接解压运行么

jjm2473, 2023-08-30:
你直接运行有加参数吗，没有参数的话不是web服务，那你在首页应该看不到任何数据，如果有数据那肯定是因为没刷新。

xiaoermeizai, 2023-08-30:
> 你直接运行有加参数吗，没有参数的话不是web服务，那你在首页应该看不到任何数据，如果有数据那肯定是因为没刷新。

nb   成功  谢谢

xiaoermeizai, 2023-08-30:
root@iStoreOS:~# STTRACE=main,service ./quickstart.amd64 serve
2023/08/30 05:23:49.969662 backend-main.go:24: DEBUG: start serve at 127.0.0.1:3038
2023/08/30 05:23:49.970475 service-backend.go:43: DEBUG: arch x86_64
2023/08/30 05:23:49.972386 service-backend.go:72: DEBUG: hwmonTemperature k10tem

jjm2473, 2023-08-30:
OK，那就没问题了，之后的固件会加上

xiaoermeizai, 2023-08-30:
> OK，那就没问题了，之后的固件会加上

十分感谢！  期待之后的固件~

我去加一下群

xiaoermeizai, 2023-08-30:
重启就会失效了   是需要等之后的固件吗？

jjm2473, 2023-08-30:
可以手动替换掉系统的quickstart，但是以后就没法跟随固件更新了。

jjm2473, 2023-08-30:
为了避免以后的麻烦，还是等固件更新比较好

xiaoermeizai, 2023-08-30:
> 为了避免以后的麻烦，还是等固件更新比较好

ok！

xiaoermeizai, 2023-09-01:
istoreos-22.03.5-2023090112-x86-64-squashfs-combined-efi.img.gz   是不是没有替换quickstart文件 ？  和上个版本一模一样   文件大小和你单发给我的不一样

kuoluo1995, 2024-06-17:
能询问一下大佬们，你们是从哪里下载到openwrt中的k10temp驱动的？
我发现opkg list里面默认是没有的。额外下载不知道到哪里寻找比较合适？
https://rpmfind.net/linux/rpm2html/search.php?query=kmod%28k10temp%29&submit=Search+...&system=&arch=x86_64

jjm2473, 2024-06-18:
@kuoluo1995 执行下 `/etc/init.d/quickstart stop && STTRACE=main,service quickstart serve` 看看输出什么。

kuoluo1995, 2024-06-21:
> /etc/init.d/quickstart stop && STTRACE=main,service quickstart serve

kuoluo@iStoreOS:~$ sudo /etc/init.d/quickstart stop && STTRACE=main,service quickstart serve
Password:
2024/06/21 03:28:00.112766 backend-main.go:24: DEBUG: start serve at 127.0.0.1:3038 unix at 127.0.0.1:3038
2024/06/21 03:28:00.120634 service-backend.go:43: DEBUG: arch x86_64
2024/06/21 03:28:00.121284 service-backend.go:96: DEBUG: hwmonTemperature default 1 1

kuoluo1995, 2024-06-21:
> kuoluo@iStoreOS:~$ sudo /etc/init.d/quickstart stop && STTRACE=main,service quickstart serve Password: 2024/06/21 03:28:00.112766 backend-main.go:24: DEBUG: start serve at 127.0.0.1:3038 unix at [...]

我opkg list里面的结果：
kmod-hwmon-core - 5.10.201-1
kmod-hwmon-drivetemp - 5.10.201-1
kmod-hwmon-it87 - 5.10.201-1
kmod-hwmon-pwmfan - 5.10.201-1
kmod-hwmon-vid - 5.10.201-1

以及sensors的结果：
root@iStoreOS:/sys/module# sensors
amdgpu-pci-0500
Adapter: PCI adapter
vddgfx:      668.00 mV
vddnb:       668.00 mV
edge:         +42.0°C
power1:      1000.00 mW

iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +40.0°C

jjm2473, 2024-06-24:
执行这个命令看看 `ls /sys/class/hwmon/ | xargs -n1 sh -c 'echo $0; cat /sys/class/hwmon/$0/name; echo'`

xfghvgnfyjssjgte, 2025-04-06:
> 能询问一下大佬们，你们是从哪里下载到openwrt中的k10temp驱动的？ 我发现opkg list里面默认是没有的。额外下载不知道到哪里寻找比较合适？ https://rpmfind.net/linux/rpm2html/search.php?query=kmod%28k10temp%29&submit=Search+...&system=&arch=x86_64

解决了吗？我也是这个问题