istoreos/istoreos #984: R4S刷istoreos-22以上的版本无法启动

lsys2, 2023-08-30:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

R4S刷istoreos-22以上版本系统无法启动，sys灯都不闪

### (2) 路由器型号和固件版本
路由：R4S
固件版本：istoreos-22.03.5-2023082509-r4s-squashfs.img

### (3) 详细日志和/或截图

A：

jjm2473, 2023-08-30:
你是怎么刷机的？

lsys2, 2023-08-30:
非常感谢没想到回复这么快，一样的刷法，21.02的我已经在用了，然后22的就无法启动，现象就是R4S只有电源灯亮，sys、lan灯都不亮。
[image0.jpeg][image1.jpeg]

发自我的iPad

jjm2473, 2023-08-30:
我这里没有r4s，要有人提供启动日志才行，使用ttl转usb连接r4s的调试脚，然后开机把ttl的日志发出来

jjm2473, 2023-08-31:
你的r4s是在升级以后出问题，还是重新制作tf卡以后出问题？有试过旧版本升级或者重做系统吗？

lsys2, 2023-08-31:
我这没有ttl转接的，我是重新制作的系统，升级行不行不清楚。

发自我的iPad

ohyeswind, 2023-09-13:
同是r4s情况一样，刷机最新istoreos固件后无法启动，刷机工具：balenaEtcher（之前版本一直用这个刷均正常），无奈换其他版本固件刷入正常启动，希望istoreOS能尽快解决，谢谢

DeadpoolRain, 2023-11-11:
我的r4s也有这个问题，刷istoreos最新版22.03.5-2023102711固件，系统sys灯都不亮，刷别的固件没问题

jjm2473, 2024-02-23:
根据[用户反馈](https://github.com/istoreos/istoreos/issues/1210#issuecomment-1952755038)，openwrt官方固件也有这个问题，那就让openwrt官方修复吧 ，istoreos的r4s固件目前使用的uboot来自openwrt官方

closed 2024-02-23 by jjm2473