istoreos/istoreos #985: 0825 的固件 Samba网络共享有严重BUG 会直接卡掉挂载

1klakla1, 2023-08-31:
播放视频几分钟就会断线，然后必须重启路由才能解决

jjm2473, 2023-08-31:
试试不保留配置升级，反正你的系统可以回滚

1klakla1, 2023-08-31:
好的，明天我测试下回复。

1klakla1, 2023-09-01:
> 试试不保留配置升级，反正你的系统可以回滚

大佬，刚才测试过了。不保留配置也是一样问题。

----------------------------------

iStoreOS 21.02.3 2023042111
我上一个版本用的是这个版本号，没有问题。
应该是这个版本固件的问题。

jjm2473, 2023-09-01:
samba是跟着上游openwrt版本升级的，我们不会闲着没事去修改samba。我今天测试下看看

1klakla1, 2023-09-01:
> samba是跟着上游openwrt版本升级的，我们不会闲着没事去修改samba。我今天测试下看看

辛苦了！

jjm2473, 2023-09-02:
我放了快一个小时的电影都没问题。

closed 2023-09-02 by jjm2473

jjm2473, 2023-09-02:
服务端是x86_64，N100，iStoreOS 22.03.5 2023082509
这是我的samba配置文件， /etc/config/samba4 

```ini

config samba
        option workgroup 'WORKGROUP'
        option charset 'UTF-8'
        option description 'Samba on OpenWRT'
        option interface 'lan wan'
        option macos '1'

config sambashare
        option path '/mnt/data'
        option read_only 'no'
        option force_root '1'
        option guest_ok 'yes'
        option inherit_owner 'yes'
        option create_mask '0666'
        option dir_mask '0777'
        option name 'data'

config sambashare
        option name 'media'
        option path '/mnt/media'
        option read_only 'no'
        option force_root '1'
        option guest_ok 'yes'
        option inherit_owner 'yes'
        option create_mask '0666'
        option dir_mask '0777'
```

客户端是苹果笔记本，使用“访达”直接打开

1klakla1, 2023-09-06:
config samba
	option workgroup 'WORKGROUP'
	option charset 'UTF-8'
	option description 'Samba on OpenWRT'
	option interface 'lan wan'

config sambashare
	option name 'Disk'
	option path '/mnt/usb2-1'
	option read_only 'no'
	option users 'jiazhen'
	option create_mask '0666'
	option dir_mask '0777'
	option guest_ok 'no'

---------------------------------------------------
这是我的设置，我设置了用户访问权限 jiazhen 这个ID加了密码
你发的那个配置好像没有加权限。

对了，补充一下，我的硬盘是NTFS格式的。WIN11系统
[image]

jjm2473, 2023-09-06:
那你注意下 usb2-1 有没有掉线，出问题以后看看内核日志，搜索分区（例如sda1）