istoreos/istoreos #987: 添加rtl8812BU USB WIFI 无限网卡支持

theLIKImk, 2023-09-03:
如果可以的话....添加rtl8812BU的支持吧

需要驱动参考的话可以到这个链接，这个驱动在树莓派4B（aarch64）编译试过可用
https://github.com/morrownr/88x2bu-20210702

jjm2473, 2023-09-14:
看看有没有其他openwrt分支版本已经打包的，集成更快。不过realtek的代码比较垃圾，可能没法同时开双频

jjm2473, 2023-10-27:
已增加rtw88-oot驱动，支持此网卡

closed 2023-10-27 by jjm2473

xuxu8023, 2023-11-07:
> 看看有没有其他openwrt分支版本已经打包的，集成更快。不过realtek的代码比较垃圾，可能没法同时开双频

咨询下，最新包已经支持rtl8822au了那是否支持双频wifiap呢 我自己试了8822ce无法开双频WiFi ap只能二选一，按照网上教程改配置文件出两个radio第二个无法使用，望回复

jjm2473, 2023-11-07:
@xuxu8023 驱动代码来自 https://github.com/lwfinger/rtw88 ，可以去这里问问。如果你在无线页面只看到一个无线设备，那一般就不支持双频共存（可能是驱动不完善，也可能是硬件不支持），支持双频共存的会显示两个无线设备，例如 MT7916 这些。

xuxu8023, 2023-11-08:
> @xuxu8023 驱动代码来自 https://github.com/lwfinger/rtw88 ，可以去这里问问。如果你在无线页面只看到一个无线设备，那一般就不支持双频共存（可能是驱动不完善，也可能是硬件不支持），支持双频共存的会显示两个无线设备，例如 MT7916 这些。

收到，感谢