istoreos/istoreos #990: 不好意思，再次反馈8168问题，这回是降速使用，请见谅。

hx801217, 2023-09-06:
之前搜索了issue，关于8168使用8169驱动，后来大大提供了驱动，我看他的问题也解决了。
但是我的问题跟他的不同，我因为是办公室百兆网，一直降速使用，从0609升级到0901之后，反应的是flow control off，也就是光有链接，但不走数据，ap无法访问，wifi无连接因特网问题。
## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：8168网卡降速使用，升级22.03.05之后，表现为不走数据，ap无法访问，wifi无连接因特网。

### (2) 路由器型号和固件版本

A：hp t620 thinpc，固件版本0910

### (3) 详细日志和/或截图

A：此外网络基本信息：
root@iStoreOS:~# cat /etc/config/network

config interface 'loopback'
        option device 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'fd3e:8cff:4ee9::/48'

config interface 'wan'
        option device 'eth0'
        option proto 'static'
        option netmask '255.255.255.0'
        option gateway '172.14.33.1'
        option peerdns '0'
        option ipaddr '172.14.33.95'
        list dns '61.132.163.68'
        list dns '202.102.213.68'

config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'eth1'

config interface 'lan'
        option proto 'static'
        option ip6assign '60'
        option ipaddr '192.168.1.1'
        option netmask '255.255.255.0'
        option device 'br-lan'

config interface 'docker'
        option device 'docker0'
        option proto 'none'
        option auto '0'

config interface 'vpn0'
        option proto 'none'
        option device 'tun0'

root@iStoreOS:~# ip link
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: ip6tnl0@NONE: <NOARP> mtu 1452 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/tunnel6 :: brd ::
3: tunl0@NONE: <NOARP> mtu 1480 qdisc noop state DOWN mode DEFAULT group default qlen 1000
[... 105 lines cut ...]
[   97.258974] r8169 0000:02:00.0 eth1: rtl_rxtx_empty_cond == 0 (loop: 42, delay: 100).
[   97.263714] r8169 0000:02:00.0 eth1: Link is Down
[   97.263887] br-lan: port 1(eth1) entered disabled state
[  102.703265] Generic FE-GE Realtek PHY r8169-0-200:00: Downshift occurred from negotiated speed 1Gbps to actual speed 100Mbps, check cabling!
[  102.703310] r8169 0000:02:00.0 eth1: Link is Up - 100Mbps/Full (downshifted) - flow control off
[  102.703356] br-lan: port 1(eth1) entered blocking state
[  102.703364] br-lan: port 1(eth1) entered forwarding state
[  108.782665] r8169 0000:02:00.0 eth1: rtl_rxtx_empty_cond == 0 (loop: 42, delay: 100).
[  108.787340] r8169 0000:02:00.0 eth1: Link is Down
[  108.787524] br-lan: port 1(eth1) entered disabled state

安装大大另一个issue里面提供的8168驱动后日志：
  19.628765] eth1: 0xffffc90000065000, 7c:d3:0a:29:d6:1c, IRQ 35
[   19.720180] br-lan: port 1(eth1) entered blocking state
[   19.720257] br-lan: port 1(eth1) entered disabled state
[   19.720484] device eth1 entered promiscuous mode
[   19.735532] eth0: 0xffffc90000055000, 00:e8:4c:68:03:88, IRQ 33
[   21.873394] r8168: eth0: link up
[   21.873646] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   25.046892] ujail[9762]: segfault at 5e ip 00007f5a8ed519fe sp 00007ffe31129540 error 4 in libubox.so.20220515[7f5a8ed50000+5000]
[   25.046987] Code: 38 76 00 00 41 58 ff 25 b8 74 00 00 31 c0 c3 55 48 89 fd 53 48 83 ec 18 40 f6 c6 03 75 0c 48 83 c4 18 5b 5d e9 81 ff ff ff 90 <80> 7d 0e 00 8b 7f 08 89 f3 75 37 40 f6 c6 08 74 07 be 01 00 00 00
[   25.950751] r8168: eth1: link up
[   25.950950] br-lan: port 1(eth1) entered blocking state
[   25.951015] br-lan: port 1(eth1) entered forwarding state
[   25.951288] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready

jjm2473, 2023-09-14:
是虚拟机吗

jjm2473, 2023-09-14:
你安装的8168驱动可能不兼容了

hx801217, 2023-09-14:
实体机，而且用的您提供的8168驱动

hx801217, 2023-09-14:
> 你安装的8168驱动可能不兼容了

> 实体机，而且用的您提供的8168驱动

hx801217, 2023-09-14:
@jjm2473 用的大佬您issue904里面提供的驱动

hx801217, 2023-09-14:
@jjm2473
01:00.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller [10ec:8168] (rev 07)
        Subsystem: Realtek Semiconductor Co., Ltd. Device [10ec:0123]
        Kernel driver in use: r8168
02:00.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller [10ec:8168] (rev 0c)
        Subsystem: Hewlett-Packard Company Device [103c:2187]

2块8168，下面HP这块不正常，上面8169驱动好好的

jjm2473, 2023-09-14:
使用8169驱动时，看日志应该是网线质量不行 `[ 36.820440] Generic FE-GE Realtek PHY r8169-0-200:00: Downshift occurred from negotiated speed 1Gbps to actual speed 100Mbps, check cabling!`

jjm2473, 2023-09-14:
使用8169驱动时执行`ethtool eth0`看看

jjm2473, 2023-09-14:
你应该没有强制设置千兆吧

jjm2473, 2023-09-14:
看日志，同样是8168网卡，你的eth0又没问题吗？

jjm2473, 2023-09-22:
是不是内存太小，0908固件有个内存异常的bug，会占用多 512M 内存

jjm2473, 2023-09-28:
新固件增加了可选驱动的功能，同时修复了r8168的优先级

closed 2023-09-28 by jjm2473