istoreos/istoreos #991: 网速差的离谱

aishuaikang, 2023-09-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

我用h69k刷的is固件，网速在200～300M，其他固件能达到500～700M，请问这是怎么回事

### (2) 路由器型号和固件版本
设备：H69K max +
固件：istoreos-22.03.5-2023082509-h6xk-squashfs-combined

### (3) 详细日志和/或截图
前者用的is，后者用的其他的固件
[image]
[image]

xuehuaxuyu, 2023-09-10:
已经更换esir系统了，瞬间跑满千兆，

jjm2473, 2023-09-14:
wifi测试的吗，还是有线？有没有使用5G？

aishuaikang, 2023-09-20:
> wifi测试的吗，还是有线？有没有使用5G？

是5G，wifi和有线都测试了，wifi在200-300左右，有线也是的

jjm2473, 2023-09-29:
我昨天测试了没有什么问题，起码千兆可以跑满的，你是不是使用5G模块联网的，所以速度这么慢

jjm2473, 2023-09-29:
而且看你的两个测试截图，有很大差别吗，都是100M到500M之间波动

closed 2023-09-29 by jjm2473