istoreos/istoreos #997: 更新iStoreOS 22.03.5 2023090810后网口降速为百兆，手动用ethtool改为千兆重启无效

david446747602, 2023-09-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：更新后接口降速为100 Mbit/s 全双工，手动用ethtool改为千兆重启无效

### (2) 路由器型号和固件版本

A：X86 iStoreOS 22.03.5 2023090810

### (3) 详细日志和/或截图

A：
[image]

tuntun2019, 2023-09-13:
这个版本我升级了，有问题，随时显示千兆，但是速度最快跑200M，回退9月1日版本 回复正常，两台都遇到这个问题

jjm2473, 2023-09-14:
什么网卡，执行 `lspci -nnk` 看看结果

tuntun2019, 2023-09-14:
我是两台，一台是i225，这台是升级到 08日版本的，
01:00.0 Ethernet controller [0200]: Intel Corporation Ethernet Controller I225-V                                    [8086:15f3] (rev 03)
        Subsystem: Intel Corporation Device [8086:0000]
        Kernel driver in use: igc
02:00.0 Ethernet controller [0200]: Intel Corporation Ethernet Controller I225-V                                    [8086:15f3] (rev 03)
        Subsystem: Intel Corporation Device [8086:0000]
        Kernel driver in use: igc
03:00.0 Ethernet controller [0200]: Intel Corporation Ethernet Controller I225-V                                    [8086:15f3] (rev 03)
        DeviceName: Onboard - RTK Ethernet
        Subsystem: Intel Corporation Device [8086:0000]
        Kernel driver in use: igc
04:00.0 Ethernet controller [0200]: Intel Corporation Ethernet Controller I225-V                                    [8086:15f3] (rev 03)
        Subsystem: Intel Corporation Device [8086:0000]
        Kernel driver in use: igc

另外一台是 8111H 手动安装了R8168的驱动，现在回退到1日版本了，升级到8日版本，症状是一样的
02:00.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL8111/8168                                                                        /8411 PCI Express Gigabit Ethernet Controller [10ec:8168] (rev 15)
        DeviceName: Onboard - RTK Ethernet
        Subsystem: Realtek Semiconductor Co., Ltd. Device [10ec:0123]
        Kernel driver in use: r8168
03:00.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL8111/8168                                                                        /8411 PCI Express Gigabit Ethernet Controller [10ec:8168] (rev 15)
        Subsystem: Realtek Semiconductor Co., Ltd. Device [10ec:0123]

家里是千兆网络，两台都是回退到9月1日版本 恢复正常   升级到8日版本显示接口千兆，但是网速实际只能跑200m左右的下行，9月1日版本可以跑满940m。

jjm2473, 2023-09-14:
升级以后恢复出厂设置试过吗

jjm2473, 2023-09-14:
这一段怎么会出现`DeviceName: Onboard - RTK Ethernet`，你复制的文本是完整的嘛
```
01:00.0 Ethernet controller [0200]: Intel Corporation Ethernet Controller I225-V                                    [8086:15f3] (rev 03)
        Subsystem: Intel Corporation Device [8086:0000]
        Kernel driver in use: igc
02:00.0 Ethernet controller [0200]: Intel Corporation Ethernet Controller I225-V                                    [8086:15f3] (rev 03)
        Subsystem: Intel Corporation Device [8086:0000]
        Kernel driver in use: igc
03:00.0 Ethernet controller [0200]: Intel Corporation Ethernet Controller I225-V                                    [8086:15f3] (rev 03)
        DeviceName: Onboard - RTK Ethernet
        Subsystem: Intel Corporation Device [8086:0000]
        Kernel driver in use: igc
04:00.0 Ethernet controller [0200]: Intel Corporation Ethernet Controller I225-V                                    [8086:15f3] (rev 03)
        Subsystem: Intel Corporation Device [8086:0000]
        Kernel driver in use: igc
```

jjm2473, 2023-09-14:
在1号的固件里面也执行`lspci -nnk`看看

tuntun2019, 2023-09-14:
没有试过，目前为了正常使用是 选择回退上一版本，升级上一个版本 就恢复正常。
哦对了还有一个现象是，不正常的版本我的udpxy转发IPTV会出现掉包卡，我都是这样发现的。上次出现时更新5.15内核因为8111网卡 没有8168驱动，我也是这样发现的，手动安装了驱动就正常了。这次出现这个问题我也试过了  8168 8169几个驱动卸载重新安装尝试，没有任何变化。

tuntun2019, 2023-09-14:
是完整的，有可能是这个名字出现RTK 有可能是我配置在多台备份 恢复造成的 intel网卡那个我刚回退版本了 
结果如下

00:1f.1 SMBus [0c05]: Intel Corporation Celeron/Pentium Silver Processor Gaussian Mixture Model [8086:31d4] (rev 06)
        DeviceName: Onboard - Other
        Subsystem: Intel Corporation Device [8086:7270]
01:00.0 Ethernet controller [0200]: Intel Corporation Ethernet Controller I225-V [8086:15f3] (rev 03)
        Subsystem: Intel Corporation Device [8086:0000]
        Kernel driver in use: igc
02:00.0 Ethernet controller [0200]: Intel Corporation Ethernet Controller I225-V [8086:15f3] (rev 03)
        Subsystem: Intel Corporation Device [8086:0000]
        Kernel driver in use: igc
03:00.0 Ethernet controller [0200]: Intel Corporation Ethernet Controller I225-V [8086:15f3] (rev 03)
        DeviceName: Onboard - RTK Ethernet
        Subsystem: Intel Corporation Device [8086:0000]
        Kernel driver in use: igc
04:00.0 Ethernet controller [0200]: Intel Corporation Ethernet Controller I225-V [8086:15f3] (rev 03)
        Subsystem: Intel Corporation Device [8086:0000]
        Kernel driver in use: igc
[image]

jjm2473, 2023-09-14:
系统的备份功能是不包括驱动的，只备份配置文件。

我等下给你个测试固件看看有没有问题

jjm2473, 2023-09-14:
你的固件是efi还是非efi的

jjm2473, 2023-09-14:
试试这个固件（EFI版本）： https://we.tl/t-o9L6mCvq1Y

tuntun2019, 2023-09-14:
测试了，还是不行，8111那台机器用1号版本通过南大测速脚本 下行可以跑到800m以上，8号 14号只能跑不到200m，下行。上行正常。
地址test.nju.edu.cn，我是直接部署到路由器，用命令行测速的

jjm2473, 2023-09-14:
奇怪，10号只是改了驱动加载顺序啊，如果没有光口，应该都不受影响的

tuntun2019, 2023-09-14:
确实是挺奇怪的， 感觉就和我8111H网卡不单独安装r8168驱动的感觉差不多，但是速度会好一点，而且这次是intel  RTK网卡两个都中招了，但是也没看到其他人由反馈，有些人可能还没升级吧。目前RTK那台我还是退到1号的版本了， intel那台还是8号版本，目前恢复出厂了，一会儿我映射出来 我再测试下

jjm2473, 2023-09-21:
是不是内存问题，0908版本的固件有个异常占用内存的bug，需要多消耗500M内存，如果是虚拟机你试试给分配多点内存，要么就等下个版本固件

tuntun2019, 2023-09-25:
今天尝试901升级到922的版本，从901直接升级之后，测速变成20m宽带，今天采用了新思路，手动仅卸载kmod-r8169驱动 保留r8169-firmware，重启，速度恢复正常，同一个 兰州电信测速点可以跑到900m以上。。。
初步判断应该是901之后的版本由于调整了驱动加载的优先级，导致8168网卡驱动没有被驱动，使用的是8169的驱动，导致网络速度异常。
由于是主力路由，通过这个解决了懒得全家断网，就不回退再升级验证测试了，如果有类似的网卡的机器 升级后建议查看下lspci -nnk 是否是8169而不是8168驱动

jjm2473, 2023-09-25:
OK，那确实可能是因为优先级改变。

jjm2473, 2023-09-27:
但是跟intel的网卡有什么关系呢，r8168的优先级也只影响realtek的网卡啊

tuntun2019, 2023-09-27:
这个不清楚，但是intel那台 我通过恢复出厂设置和升级922的版本解决了

jjm2473, 2023-09-28:
新固件增加了可选驱动的功能，同时修复了r8168的优先级

closed 2023-09-28 by jjm2473

maxxia29, 2023-11-01:
你好这里出现同样的问题，N5105-I226网卡
root@iStoreOS:~# lspci -nnk
00:00.0 Host bridge [0600]: Intel Corporation 440BX/ZX/DX - 82443BX/ZX/DX Host bridge [8086:7190] (rev 01)
        Subsystem: VMware Virtual Machine Chipset [15ad:1976]
        Kernel driver in use: agpgart-intel
lspci: Unable to load libkmod resources: error -12
00:01.0 PCI bridge [0604]: Intel Corporation 440BX/ZX/DX - 82443BX/ZX/DX AGP bridge [8086:7191] (rev 01)
00:07.0 ISA bridge [0601]: Intel Corporation 82371AB/EB/MB PIIX4 ISA [8086:7110] (rev 08)
        Subsystem: VMware Virtual Machine Chipset [15ad:1976]
00:07.1 IDE interface [0101]: Intel Corporation 82371AB/EB/MB PIIX4 IDE [8086:7111] (rev 01)
        Subsystem: VMware Virtual Machine Chipset [15ad:1976]
        Kernel driver in use: ata_piix
00:07.3 Bridge [0680]: Intel Corporation 82371AB/EB/MB PIIX4 ACPI [8086:7113] (rev 08)
        Subsystem: VMware Virtual Machine Chipset [15ad:1976]
00:07.7 System peripheral [0880]: VMware Virtual Machine Communication Interface [15ad:0740] (rev 10)
        Subsystem: VMware Virtual Machine Communication Interface [15ad:0740]
        Kernel driver in use: vmw_vmci
00:0f.0 VGA compatible controller [0300]: VMware SVGA II Adapter [15ad:0405]
        Subsystem: VMware SVGA II Adapter [15ad:0405]
00:11.0 PCI bridge [0604]: VMware PCI bridge [15ad:0790] (rev 02)
        Subsystem: VMware PCI bridge [15ad:0790]
00:15.0 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
00:15.1 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
00:15.2 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
00:15.3 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
00:15.4 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
00:15.5 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
00:15.6 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
00:15.7 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
00:16.0 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
00:16.1 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
00:16.2 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
00:16.3 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
00:16.4 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
00:16.5 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
00:16.6 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
00:16.7 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
00:17.0 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
[... 42 lines cut ...]
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
00:18.7 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
03:00.0 Ethernet controller [0200]: Intel Corporation Device [8086:125c] (rev 04)
        DeviceName: pciPassthru0
        Subsystem: Intel Corporation Device [8086:0000]
        Kernel driver in use: igc
04:00.0 USB controller [0c03]: VMware USB3 xHCI 1.0 Controller [15ad:0779]
        DeviceName: usb_xhci
        Subsystem: VMware USB3 xHCI 1.0 Controller [15ad:0779]
        Kernel driver in use: xhci_hcd
0b:00.0 Ethernet controller [0200]: Intel Corporation Device [8086:125c] (rev 04)
        DeviceName: pciPassthru1
        Subsystem: Intel Corporation Device [8086:0000]
        Kernel driver in use: igc
13:00.0 Ethernet controller [0200]: Intel Corporation Device [8086:125c] (rev 04)
        DeviceName: pciPassthru2
        Subsystem: Intel Corporation Device [8086:0000]
        Kernel driver in use: igc
1b:00.0 Ethernet controller [0200]: VMware VMXNET3 Ethernet Controller [15ad:07b0] (rev 01)
        DeviceName: Ethernet0
        Subsystem: VMware VMXNET3 Ethernet Controller [15ad:07b0]
        Kernel driver in use: vmxnet3

jjm2473, 2023-11-02:
@maxxia29 你是用的虚拟机？试试不要直通网卡，改成桥接的虚拟网卡，反正虚拟网卡都是万兆的

apple4105, 2023-11-04:
禁用i915_oot驱动可以解决

jjm2473, 2023-11-07:
@apple4105 你的cpu是什么型号？固件啥版本？

ioojcn, 2023-11-30:
我说我最近网速好像变100M了..上路由器看一眼的确是百兆口了.来回插拔N次网线.变回来了.

但是发现我的1000M宽带.最高只能跑到500M了.

gzxiexl, 2023-12-11:
我也遇到了 我还以为是电信的问题 我是3215U 4口I211物理机安装istoreos
开机后，wan口反复重连，最后变100M才稳定下来。
lspci -nnk结果如下：
```
root@iStoreOS:~# lspci -nnk
00:00.0 Host bridge [0600]: Intel Corporation Broadwell-U Host Bridge -OPI [8086:1604] (rev 09)
        Subsystem: Intel Corporation Broadwell-U Host Bridge -OPI [8086:1604]
        Kernel driver in use: bdw_uncore
lspci: Unable to load libkmod resources: error -12
00:02.0 VGA compatible controller [0300]: Intel Corporation HD Graphics [8086:1606] (rev 09)
        DeviceName:  Onboard IGD
        Subsystem: Intel Corporation HD Graphics [8086:1606]
        Kernel driver in use: i915
00:03.0 Audio device [0403]: Intel Corporation Broadwell-U Audio Controller [8086:160c] (rev 09)
        Subsystem: Intel Corporation Broadwell-U Audio Controller [8086:160c]
00:14.0 USB controller [0c03]: Intel Corporation Wildcat Point-LP USB xHCI Controller [8086:9cb1] (rev 03)
        Subsystem: Intel Corporation Device [8086:7270]
        Kernel driver in use: xhci_hcd
00:16.0 Communication controller [0780]: Intel Corporation Wildcat Point-LP MEI Controller #1 [8086:9cba] (rev 03)
        Subsystem: Intel Corporation Device [8086:7270]
00:1b.0 Audio device [0403]: Intel Corporation Wildcat Point-LP High Definition Audio Controller [8086:9ca0] (rev 03)
        Subsystem: Intel Corporation Device [8086:7270]
00:1c.0 PCI bridge [0604]: Intel Corporation Wildcat Point-LP PCI Express Root Port #1 [8086:9c90] (rev e3)
        Subsystem: Intel Corporation Device [8086:7270]
        Kernel driver in use: pcieport
00:1c.1 PCI bridge [0604]: Intel Corporation Wildcat Point-LP PCI Express Root Port #2 [8086:9c92] (rev e3)
        Subsystem: Intel Corporation Device [8086:7270]
        Kernel driver in use: pcieport
00:1c.2 PCI bridge [0604]: Intel Corporation Wildcat Point-LP PCI Express Root Port #3 [8086:9c94] (rev e3)
        Subsystem: Intel Corporation Device [8086:7270]
        Kernel driver in use: pcieport
00:1c.4 PCI bridge [0604]: Intel Corporation Wildcat Point-LP PCI Express Root Port #5 [8086:9c98] (rev e3)
        Subsystem: Intel Corporation Device [8086:7270]
        Kernel driver in use: pcieport
00:1d.0 USB controller [0c03]: Intel Corporation Wildcat Point-LP USB EHCI Controller [8086:9ca6] (rev 03)
        Subsystem: Intel Corporation Device [8086:7270]
        Kernel driver in use: ehci-pci
00:1f.0 ISA bridge [0601]: Intel Corporation Wildcat Point-LP LPC Controller [8086:9cc5] (rev 03)
        Subsystem: Intel Corporation Device [8086:7270]
        Kernel driver in use: lpc_ich
00:1f.2 SATA controller [0106]: Intel Corporation Wildcat Point-LP SATA Controller [AHCI Mode] [8086:9c83] (rev 03)
        Subsystem: Intel Corporation Device [8086:7270]
        Kernel driver in use: ahci
00:1f.3 SMBus [0c05]: Intel Corporation Wildcat Point-LP SMBus Controller [8086:9ca2] (rev 03)
        Subsystem: Intel Corporation Device [8086:7270]
01:00.0 Ethernet controller [0200]: Intel Corporation I211 Gigabit Network Connection [8086:1539] (rev 03)
        Subsystem: Intel Corporation Device [8086:0000]
        Kernel driver in use: igb
02:00.0 Ethernet controller [0200]: Intel Corporation I211 Gigabit Network Connection [8086:1539] (rev 03)
        Subsystem: Intel Corporation Device [8086:0000]
        Kernel driver in use: igb
03:00.0 Ethernet controller [0200]: Intel Corporation I211 Gigabit Network Connection [8086:1539] (rev 03)
        Subsystem: Intel Corporation Device [8086:0000]
        Kernel driver in use: igb
04:00.0 Ethernet controller [0200]: Intel Corporation I211 Gigabit Network Connection [8086:1539] (rev 03)
        Subsystem: Intel Corporation Device [8086:0000]
        Kernel driver in use: igb
```

Meilinhost, 2023-12-14:
N100 i226 同样遇到了这个问题，下载官网最新的版本，早上测试变成100M
为此还刷了AP固件，WIFI一顿操作没效果，最后重启istoreos就恢复了

jjm2473, 2023-12-15:
@ioojcn 打开后台网络-全局网络配置，勾选数据包引导，然后保存应用，重启系统，看看速度有没有变化

jjm2473, 2023-12-15:
网络协商降速的问题，一般都是网线不好。

ioojcn, 2023-12-15:
> @ioojcn 打开后台网络-全局网络配置，勾选数据包引导，然后保存应用，重启系统，看看速度有没有变化

看别人说降到0901就好了.的确好了.先不动了...网心云缓存了7天才满...

Meilinhost, 2023-12-16:
> @ioojcn 打开后台网络-全局网络配置，勾选数据包引导，然后保存应用，重启系统，看看速度有没有变化

全局网络配置，勾选数据包引导 并没有起到效果，更换了一根超5的线 管了一天又再次掉到100M
[image]
[image]

Meilinhost, 2023-12-16:
刚刚试了ikuai 拨号，ikuai也会自适应失败变成100M，难道真的是网线问题？

我先取消自适应 强制1000M用段时间看看

`ethtool eth3 #查看当前速率`

`ethtool -s eth3 speed 1000 duplex full autoneg off  # 取消eth3协商速率 调整为1000M固定速率`

然后持久化就加入istoreos 开机自启

jjm2473, 2023-12-16:
总之，网络协商降速100M的一般都是网线质量不行。

IsColin, 2024-06-26:
[image]
同问怎么解决 网线插到电脑就是千兆,插到istoreos的wan口就是百兆 
最新固件
iStoreOS 22.03.6 2024061415
CPU: N100
网卡：板载 INTEL I226×4 2.5G 网卡芯片

jjm2473, 2024-06-26:
可能i226驱动比较挑网线，反正换换网线试试

aionroy, 2024-07-12:
J1900 4网口工控机 ，也是一样的问题，千M网卡显示百M了。

walkerfirst, 2024-09-26:
我也安装的最新版本，国内下载百兆，上传 130，而用 openclash 国外下载能上千兆，上行 150，请问这是什么原因？wan 和 lan 都显示千兆

jjm2473, 2024-10-24:
@walkerfirst 确定不是你的远程测速节点不行吗？

Meilinhost, 2025-07-04:
买了一大堆成品网线和一些品牌的扁平都不好使，最终还是决定更换一个光猫试试。联系运营商要收钱更换，还是旧款，索性某鱼110买了一个光猫7615v3b 彻底解决问题。无论如何重启 插拔，和之前一样各种虐，都是协商成功。希望对还在苦恼的XD有帮助。

[image]

[image]