istoreos/istoreos #999: 升级到22.03.5 2023090810后，部署的docker无法访问内部局域网

dick86114, 2023-09-11:
昨晚更新新版本后，在iStoreos上部署的所有容器都无法访问局域网，能访问互联网

kirafreedom, 2023-09-11:
我这里反过来，能局域网访问，不能访问互联网，aira2都不能下载了

kirafreedom, 2023-09-11:
> 我这里反过来，能局域网访问，不能访问互联网，aira2都不能下载了

自行解决了，在防火墙把IP动态伪勾上就行

jjm2473, 2023-09-14:
> 我的还是不行哦。能给个防火墙配置的截图吗？

防火墙勾选lan区域的ip动态伪装

TeddyXiong79, 2023-09-20:
我发现不是在Docker上部署的容器不能访问外部，而是容器之间无法访问。比如TeslaMate，app访问不了数据库

dick86114, 2023-09-20:
> 我发现不是在Docker上部署的容器不能访问外部，而是容器之间无法访问。比如TeslaMate，app访问不了数据库

勾选下ip动态伪装看看行不行，我们都是勾选后解决了各种访问异常

jjm2473, 2023-09-21:
对于单网口设备，如果不想打开lan口的ip动态伪装（可能影响其他功能），可以在终端执行这段命令（整段复制执行）：
```bash
uci -q get firewall.docker_nat >/dev/null || uci -q batch <<-EOF
        set firewall.docker_nat=nat
        set firewall.docker_nat.name='DockerNAT'
        set firewall.docker_nat.proto='all'
        set firewall.docker_nat.src='lan'
        set firewall.docker_nat.target='MASQUERADE'
        set firewall.docker_nat.extra='-i docker0'
EOF
/etc/init.d/firewall reload
```

这个命令只对docker默认网络有效，如果是自己创建的docker网络，在防火墙的NAT页面模仿配置即可。

TeddyXiong79, 2023-09-21:
我是R5S，多网卡设备，目前的问题就是在Docker中的容器之间无法访问。嗨。实在是搞不懂。

jjm2473, 2023-09-21:
@TeddyXiong79 最新固件还有问题？
执行这段脚本试试：
```
uci -q set dockerd.globals.iptables=0

uci -q delete firewall.docker.network

DOCKER0_DEVICE=`uci show network | grep -E "^network\\.@device\\[\\d+\\]\\.name='docker0'" | head -n1 | head -c -16`

if [ -n "$DOCKER0_DEVICE" ]; then
    uci delete "$DOCKER0_DEVICE"
fi
uci -q delete network.docker
uci commit
```

然后重启设备

fankangsong, 2023-10-13:
> > 
> 
> 感谢指点，我在lan转发到docker这一行勾上后，也能顺利访问内网了。。。

困扰我好几天的问题终于解决了。

gowy222, 2023-10-17:
R5S 最新版本固件，还是这个问题, 到底咋办？ docker 部署K3S host模式 重启 网络 直接崩，有没有彻底解决方案?

jjm2473, 2023-10-19:
防火墙的全局设置里面把“转发”设置为“允许”试试。

如果是自己添加过docker网络配置，将实际的接口名称加入到防火墙的docker区域试试。默认docker网络桥都会经过防火墙，如果没配置桥内允许转发，防火墙的全局设置默认又是禁止转发的，那就没法互通了。

TeddyXiong79, 2023-10-19:
防火墙的全局设置里面把“转发”设置为“允许”后，就可以了。谢谢。
我还想知道如何单独配置docker网络桥内部允许转发？

jjm2473, 2023-10-20:
@TeddyXiong79 防火墙的docker区域编辑看看，高级设置里面可以添加你自定义的docker网络接口。要么升级今天发布的固件，默认就禁用网络桥的过滤，也就是不会再阻止桥内部转发。

closed 2023-10-20 by jjm2473