istoreos/istoreos #1002: 首页一直提示软件源错误

lcuwx2016, 2023-09-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
首页一直提示软件源错误

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：修改定制源后回滚修改，首页一直提示软件源错误

### (2) 路由器型号和固件版本

A：X86 iStoreOS 22.03.5 2023090112

### (3) 详细日志和/或截图

A：
[image]

[image]

lcuwx2016, 2023-09-12:
目前只有清华源有这个问题，切换到中国科大源后显示正常

jjm2473, 2023-09-14:
是清华源返回了403，之后会改变检测方式

closed 2023-09-28 by jjm2473

yangpin97, 2023-11-29:
@lcuwx2016 你好，我也遇到了相同的问题，换了很多个国内dns，以及切换不同的软件源，都不行，切终端运行opkg update都是失败的，见下图
[image]
[image]

wx2020, 2023-11-29:
你这是虚拟机安装的吧，安装的旁路由吗

yangpin97, 2023-11-29:
是的

wx2020, 2023-11-29:
手动设置一下上游dns和网关

yangpin97, 2023-11-29:
我进设备底层，用nslookup和dig解析域名都可以正常加载出来。

yangpin97, 2023-11-29:
好的，那我试试吧，谢谢

 
重装试试吧。。我有也不好说是什么问题。。

jjm2473, 2023-12-15:
@yangpin97 你的配置文件语法都错了吧，恢复出厂设置吧

yangpin97, 2023-12-15:
@jjm2473 重装了