istoreos/istoreos #1004: 三星1T固态盘装完系统，剩下900多G空间无法分区

starlitCoding, 2023-09-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
我是个三星980pro 1T固态，装完系统，系统只占到5个G左右。我一旦把另外的空闲空间分区然后格式化，金系统后就会出新啊无法配置lan口的问题。如果不初始化那块空间就没有任何问题。

## 2. 详细叙述

### (1) 具体问题

A：quickstart装完系统之后，系统正常启动，但是系统盘只有四块分区，占了5个G的空间，剩下的929G空间没有分区。
我如果给剩下的空间分区，重启系统后就会/etc/config/下的network文件就会消失，系统无法配置网口。

### (2) 路由器型号和固件版本

A：x86小主机：N100

### (3) 详细日志和/或截图

A：
[image]

starlitCoding, 2023-09-15:
不好意思，我试了好几个入口去挂载这个分区和格式化这个分区都显示格式化失败，然后就提交了这个issue。结果刚提出来，就发现有个入口可以完成格式化。
[image]

jjm2473, 2023-09-21:
可能磁盘管理插件不支持GPT的备份表移动，导致GPT分区表损坏，先用首页的分区功能吧。