istoreos/istoreos #1006: 接的usb3.0的硬盘盒，时间一长就失去响应，uas报错。

owl6660sky, 2023-09-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
接的usb3.0的硬盘盒，时间一长就失去响应，uas报错。

Q：是否用关键词搜索了issue? (使用 " 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：接的usb3.0的硬盘盒，时间一长就失去响应，uas报错。看着网上有种说法是uas驱动的问题。可以把uas调整为usb_storage可能可以解决，但不知istoreOS如何设置。  lsusb -t 显示 Port 3: Dev 3, If 0, Class=, Driver=uas, 5000M

### (2) 路由器型号和固件版本

A：x86 
iStoreOS 22.03.5 2023090810

### (3) 详细日志和/或截图

A：
日志
Fri Sep 15 23:34:43 2023 kern.info kernel: [28845.422112] sd 0:0:0:0: [sda] tag#25 CDB: opcode=0x2a 2a 00 29 47 bc 48 00 00 08 00
Fri Sep 15 23:34:43 2023 kern.info kernel: [28845.422151] sd 0:0:0:0: [sda] tag#7 uas_eh_abort_handler 0 uas-tag 30 inflight: CMD OUT
Fri Sep 15 23:34:43 2023 kern.info kernel: [28845.422152] sd 0:0:0:0: [sda] tag#7 CDB: opcode=0x2a 2a 00 2a 40 bb d8 00 00 08 00
Fri Sep 15 23:34:43 2023 kern.info kernel: [28845.422191] sd 0:0:0:0: [sda] tag#6 uas_eh_abort_handler 0 uas-tag 29 inflight: CMD OUT
Fri Sep 15 23:34:43 2023 kern.info kernel: [28845.422192] sd 0:0:0:0: [sda] tag#6 CDB: opcode=0x2a 2a 00 29 84 97 60 00 00 08 00
Fri Sep 15 23:34:43 2023 kern.info kernel: [28845.422231] sd 0:0:0:0: [sda] tag#5 uas_eh_abort_handler 0 uas-tag 28 inflight: CMD OUT
Fri Sep 15 23:34:43 2023 kern.info kernel: [28845.422232] sd 0:0:0:0: [sda] tag#5 CDB: opcode=0x2a 2a 00 29 84 79 b0 00 00 08 00
Fri Sep 15 23:34:43 2023 kern.info kernel: [28845.422271] sd 0:0:0:0: [sda] tag#4 uas_eh_abort_handler 0 uas-tag 27 inflight: CMD OUT
Fri Sep 15 23:34:43 2023 kern.info kernel: [28845.422272] sd 0:0:0:0: [sda] tag#4 CDB: opcode=0x2a 2a 00 29 45 f5 30 00 00 08 00
Fri Sep 15 23:34:43 2023 kern.info kernel: [28845.481963] scsi host0: uas_eh_device_reset_handler start
Fri Sep 15 23:34:43 2023 kern.info kernel: [28845.582318] usb 2-2.3: reset SuperSpeed Gen 1 USB device number 3 using xhci_hcd
Fri Sep 15 23:34:44 2023 kern.info kernel: [28845.613692] scsi host0: uas_eh_device_reset_handler success

反复报错后reset

jjm2473, 2023-09-21:
你的硬盘盒是什么型号，比较新的吗？执行`lsusb -tv`看看结果

owl6660sky, 2023-09-22:
> 你的硬盘盒是什么型号，比较新的吗？执行`lsusb -tv`看看结果

root@iStoreOS:~# lsusb -tv
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 10000M
    ID 1d6b:0003  
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/12p, 480M
    ID 1d6b:0002  
    |__ Port 6: Dev 2, If 0, Class=, Driver=hub/4p, 480M
        ID 05e3:0610  
    |__ Port 10: Dev 3, If 0, Class=, Driver=, 12M
        ID 8087:0026  
    |__ Port 10: Dev 3, If 1, Class=, Driver=, 12M
        ID 8087:0026  
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 10000M
    ID 1d6b:0003  
    |__ Port 2: Dev 2, If 0, Class=, Driver=hub/4p, 5000M
        ID 05e3:0626  
        |__ Port 3: Dev 3, If 0, Class=, Driver=uas, 5000M
            ID 0080:a001  
        |__ Port 4: Dev 4, If 0, Class=, Driver=r8152, 5000M
            ID 0bda:8153  
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    ID 1d6b:0002  

并不是很新的硬盘盒。先typec转usb，再usb转2.5寸的。Port 3: Dev 3, If 0, Class=, Driver=uas, 5000M   ID 0080:a001  这个上的

jjm2473, 2023-09-22:
你会修改grub配置文件吗？也就是修改 `/boot/grub/grub.cfg`，在第一个 `noinitrd` 后面添加 ` usb-storage.quirks=0080:a001:u` ，然后重启设备

jjm2473, 2023-09-22:
还有个临时测试的方法，拔掉硬盘盒，终端执行 `echo '0080:a001:u' > /sys/module/usb_storage/parameters/quirks`，然后再插上硬盘盒，应该就不会启用 UAS 了

owl6660sky, 2023-09-22:
> 你会修改grub配置文件吗？也就是修改 `/boot/grub/grub.cfg`，在第一个 `noinitrd` 后面添加 ` usb-storage.quirks=0080:a001:u` （注意`noinitrd` 后面需要空格），然后重启设备

多谢指教。已经按照这个方法改过来了，lsusb -tv已是usb-storage。再观察下长时间还会不会失去响应。
原来是这个文件路径，我找的资料都是linux下的解决方法，都是说的在/etc/modprobe.d下，要么boot.ini。
再次感谢！

jjm2473, 2023-09-22:
@owl6660sky 但是这个修改方式在升级以后会被覆盖，所以如果测试资格方案可以用，还是要写到内核里去

seacatwu, 2023-11-26:
这个可能解决问题，虽然我的yda主板被我搞废了，不知道能不能修好，但是这个应该是个解决方案。

jjm2473, 2024-02-23:
最新固件支持了内核参数的调整，可以自己添加UAS黑名单了 https://github.com/istoreos/istoreos/discussions/1359

closed 2024-02-23 by jjm2473

owl6660sky, 2024-03-19:
已测试，已解决问题。

bluedevils23, 2024-10-10:
找了好几天终于找到解决办法了，优越者硬盘盒主控asm235cm, 系统日志

TMODER, 2025-01-12:
> 找了好几天终于找到解决办法了，优越者硬盘盒主控asm235cm非常常见的主控了, 系统日志
> 
> Thu Oct 10 12:36:26 2024 kern.info kernel: [ 888.233884] scsi host1: uas_eh_device_reset_handler start Thu Oct 10 12:36:26 2024 kern.info kernel: [ 888.233960] sd 1:0:0:0: [sdb] tag#1 uas_zap_pending 0 uas-tag 1 inflight: CMD Thu Oct 10 12:36:26 2024 kern.info kernel: [ 888.233967] sd 1:0:0:0: [sdb] tag#1 CDB: opcode=0x2a 2a 00 76 08 10 00 00 00 08 00 Thu Oct 10 12:36:26 2024 kern.info kernel: [ 888.384475] usb 2-1: reset SuperSpeedPlus Gen 2 USB device number 2 using xhci_hcd Thu Oct 10 12:36:26 2024 kern.info kernel: [ 888.416828] scsi host1: uas_eh_device_reset_handler success Thu Oct 10 12:36:26 2024 kern.info kernel: [ 888.454306] sd 1:0:0:0: [sdb] tag#1 data cmplt err -71 uas-tag 1 inflight: CMD Thu Oct 10 12:36:26 2024 kern.info kernel: [ 888.454316] sd 1:0:0:0: [sdb] tag#1 CDB: opcode=0x2a 2a 00 76 08 10 00 00 00 08 00 Thu Oct 10 12:36:57 2024 kern.info kernel: [ 918.952923] scsi host1: uas_eh_device_reset_handler start Thu Oct 10 12:36:57 2024 kern.info kernel: [ 918.952999] sd 1:0:0:0: [sdb] tag#1 uas_zap_pending 0 uas-tag 1 inflight: CMD Thu Oct 10 12:36:57 2024 kern.info kernel: [ 918.953007] sd 1:0:0:0: [sdb] tag#1 CDB: opcode=0x2a 2a 00 76 08 10 00 00 00 08 00 Thu Oct 10 12:36:57 2024 kern.info kernel: [ 919.103409] usb 2-1: reset SuperSpeedPlus Gen 2 USB device number 2 using xhci_hcd Thu Oct 10 12:36:57 2024 kern.info kernel: [ 919.135910] scsi host1: uas_eh_device_reset_handler success Thu Oct 10 12:36:57 2024 kern.info kernel: [ 919.173259] sd 1:0:0:0: [sdb] tag#1 data cmplt err -71 uas-tag 1 inflight: CMD Thu Oct 10 12:36:57 2024 kern.info kernel: [ 919.173269] sd 1:0:0:0: [sdb] tag#1 CDB: opcode=0x2a 2a 00 76 08 10 00 00 00 08 00 Thu Oct 10 12:37:28 2024 kern.info kernel: [ 949.681795] scsi host1: uas_eh_device_reset_handler start Thu Oct 10 12:37:28 2024 kern.info kernel: [ 949.681871] sd 1:0:0:0: [sdb] tag#1 uas_zap_pending 0 uas-tag 1 inflight: CMD Thu Oct 10 12:37:28 2024 kern.info kernel: [ 949.681879] sd 1:0:0:0: [sdb] tag#1 CDB: opcode=0x2a 2a 00 76 08 10 00 00 00 08 00 Thu Oct 10 12:37:28 2024 kern.info kernel: [ 949.832300] usb 2-1: reset SuperSpeedPlus Gen 2 USB device number 2 using xhci_hcd Thu Oct 10 12:37:28 2024 kern.info kernel: [ 949.864766] scsi host1: uas_eh_device_reset_handler success

我也遇到这个问题了！ 离谱啊！还不知道怎么改默认配置

TMODER, 2025-01-12:
> 还有个临时测试的方法，拔掉硬盘盒，终端执行 `echo '0080:a001:u' > /sys/module/usb_storage/parameters/quirks`，然后再插上硬盘盒，应该就不会启用 UAS 了

我这尝试使用了 这个方法是OK的，但我这设备 没有 /boot/grub/grub.cfg 这个路径  ，重启就失效了 苦涩中....

owl6660sky, 2025-03-26:
> > 还有个临时测试的方法，拔掉硬盘盒，终端执行 `echo '0080:a001:u' > /sys/module/usb_storage/parameters/quirks`，然后再插上硬盘盒，应该就不会启用 UAS 了
> 
> 我这尝试使用了 这个方法是OK的，但我这设备 没有 /boot/grub/grub.cfg 这个路径 ，重启就失效了 苦涩中....

如果是最新版的固件，可以在系统-杂项设置-内核启动参数-自定义内核参数，这里填写参数即可。类似：usb-storage.quirks=0080:a001:u