istoreos/istoreos #1007: 有没有办法实现truenas或群辉那样用nvme做缓存啊？

jakesolo, 2023-09-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志和/或截图

A：

jjm2473, 2023-09-21:
不知道他们用的什么方案。

有个比较简单的方式，你可以给nvme分个区，在磁盘管理那边格式化成swap，作为交换区使用

jakesolo, 2023-09-24:
> 不知道他们用的什么方案。
> 
> 有个比较简单的方式，你可以给nvme分个区，在磁盘管理那边格式化成swap，作为交换区使用

我试过一下，感觉好像并没有缓存盘那种效果

qq10373815, 2023-09-24:
我用的方法是丢了个16G的傲腾进去，
修改使用交换分区的比例 
echo "vm.swappiness=90" >> /etc/sysctl.conf

然后挂载swap交换分区
dd if=/dev/zero of=/mnt/swapfile bs=1M count=4096
chmod 0644 /mnt/swapfile
mkswap /mnt/swapfile
swapon /mnt/swapfile

================================================================================
然而我现在的问题是不知道这么开zram，CPU够用，物理内存不够用，我在树莓派上用zram很明显的。但是这个R66S版本的openwrt实在是不知道怎么开，折腾了快两周了，貌似只能在编译内核的时候开启。其他方法实在是实现。。。

jjm2473, 2023-09-24:
@qq10373815 zram的模块没集成，以后可以加上。加上以后，后台页面的“系统”-“系统”-“Zram设置”就可以设置

jjm2473, 2023-09-24:
@qq10373815 你的创建swapfile路径是错误的，/mnt是临时文件夹，总共只有4MB，起码应该创建到硬盘里

qq10373815, 2023-09-24:
我mount挂载在mnt  从安卓你那边带过来的习惯