istoreos/istoreos #1008: iStoreOS 设为旁路由模式，Nginx Proxy Manager 反向代理安装在istoreos。内外访问正常，外网无法访问，请问是哪里需要设置吗？

stkroot, 2023-09-19:
iStoreOS 设为旁路由模式，Nginx Proxy Manager 反向代理安装在istoreos。内外访问正常，外网无法访问，请问是哪里需要设置吗？

主路由为ikuai，已开启端口转发和DMZ主机，都无效。先以为ikuai出问题了，后来测试端口转发到其它主机，正常。

感觉像istoreos拒绝了外网的访问请求，但没找到开启外网访问的设置在哪？

stkroot, 2023-09-19:
openwrt----web管理----只允许内网访问，iStoreOS 没有这个设置吗？

2ipm1nd, 2023-10-11:
打开 etc/config 目录下的 uhttpd 文件，找到 option rfc1918_filter '1'，把 1 改成 0 即可

試試看這個行不行

jjm2473, 2023-10-25:
从istore里面安装uhttpd，就可以配置了，应该就是楼上说的，取消勾选“忽略公共接口上的私有 IP”

jjm2473, 2023-10-25:
ikuai的端口转发有没有DNAT？

caideone, 2023-11-09:
> > iStoreOS 设为旁路由模式，Nginx Proxy Manager 反向代理安装在istoreos。内外访问正常，外网无法访问，请问是哪里需要设置吗？
> > 主路由为ikuai，已开启端口转发和DMZ主机，都无效。先以为ikuai出问题了，后来测试端口转发到其它主机，正常。
> > 感觉像istoreos拒绝了外网的访问请求，但没找到开启外网访问的设置在哪？
> 
> 我也是这个问题，我的home assistant旁路指向istore os ，外网无法访问，内网可以

把ipv6关了，又好了 奇怪

closed 2024-02-23 by jjm2473