istoreos/istoreos #1010: frp使用xtcp时配置错误问题

Xuchengen, 2023-09-21:
## 官方文档中配置例子
[例子地址](https://gofrp.org/docs/examples/xtcp/)
```
[common]
server_addr = x.x.x.x
server_port = 7000
# 当默认的 stun server 不可用时，可以配置一个新的
# nat_hole_stun_server = xxx

[p2p_ssh_visitor]
type = xtcp
# xtcp 的访问者
role = visitor
# 要访问的 xtcp 代理的名字
server_name = p2p_ssh
sk = abcdefg
# 绑定本地端口用于访问 ssh 服务
bind_addr = 127.0.0.1
bind_port = 6000
# 当需要自动保持隧道打开时，设置为 true
# keep_tunnel_open = false
```

## 通过luci配置生成如下
```
[MySQL]
type = xtcp
use_encryption = false
use_compression = false
local_ip = 127.0.0.1
local_port = 3306
role = visitor
server_name = Local-MySQL2
sk = [密码]
```

## 问题
### 系统日志输出如下
```
Thu Sep 21 11:51:04 2023 daemon.info frpc[16116]: bind_port is required
Thu Sep 21 11:51:04 2023 daemon.info procd: Instance frpc::instance1 s in a crash loop 14 crashes, 0 seconds since last crash
```
### 配置差异
```
bind_addr = 127.0.0.1
bind_port = 6000

local_ip = 127.0.0.1
local_port = 3306
```
## 解决方案
需要修复luci页面逻辑！

jjm2473, 2023-09-21:
frpc是op官方仓库的插件，你先去那边看看有没有相关的问题  https://github.com/openwrt/luci/tree/master/applications/luci-app-frpc

davidGangGang, 2024-07-31:
我也遇到了这个问题，不知道改怎么修复？

jjm2473, 2024-08-02:
你们是不是用了第三方的仓库，导致frpc的版本跟luci不匹配

jjm2473, 2024-08-02:
还有，你们是不是搞错了frpc和frps？

xtcp是要在两个frpc上设置的 https://gofrp.org/zh-cn/docs/examples/xtcp/ ，肯定不需要 bind_addr，并且还需要一个frps作为打洞服务器

closed 2024-08-02 by jjm2473