istoreos/istoreos #1011: 鸭子的h28k有适配计划么？

jjkle, 2023-09-21:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：目前有个大雕套皮的，不怎么好用

### (2) 路由器型号和固件版本

A：

### (3) 详细日志和/或截图

A：

jjm2473, 2023-09-21:
目前还没测试机，好像是rk3399方案的吧

jjkle, 2023-09-21:
> 目前还没测试机，好像是rk3399方案的吧

是rk3528。

slty7, 2023-09-23:
最近想入手一台rk3538的h28k，相等istore os，2G内存和8g emmc。可以外置内存卡不装docker的话可玩性很高价格也还行，就不用担心安装不了软件了。随便安装软件高大全。

jjkle, 2023-09-23:
> 最近想入手一台rk3538的h28k，相等istore os，2G内存和8g emmc。可以外置内存卡不装docker的话可玩性很高价格也还行，就不用担心安装不了软件了。随便安装软件高大全。

现在就大雕和Flippy的固件，而且大雕的还是没得机子的情况下适配的

jjkle, 2023-09-23:
> 最近想入手一台rk3538的h28k，相等istore os，2G内存和8g emmc。可以外置内存卡不装docker的话可玩性很高价格也还行，就不用担心安装不了软件了。随便安装软件高大全。

而且鸭王说有bug 的固件不要发的，哪有没得bug的系统

slty7, 2023-09-23:
这平台其实应该是istore的很好的载体，可以测试的软件就丰富了，至于bug吗也就是有人开发就有bug，想要istore快速的话就得兼容各个商家。不然一直都是小众系统而已应该也不是istore初衷吧。179元/2g/8g(emmc)感觉挺理想的载台的

slty7, 2023-09-23:
> > 最近想入手一台rk3538的h28k，相等istore os，2G内存和8g emmc。可以外置内存卡不装docker的话可玩性很高价格也还行，就不用担心安装不了软件了。随便安装软件高大全。
> 
> 而且鸭王说有bug 的固件不要发的，哪有没得bug的系统

这平台其实应该是istore的很好的载体，可以测试的软件就丰富了，至于bug吗也就是有人开发就有bug，想要istore快速的话就得兼容各个商家。不然一直都是小众系统而已应该也不是istore初衷吧。179元/2g/8g(emmc)感觉挺理想的载台的

slty7, 2023-09-23:
顺便问一下这里有没有非官方的turenas scale的那种可以安装自己docker的源吗，最近安装truenas scale自己部署各种服务总是报错提示没有权限之类的

jjm2473, 2023-10-23:
h28k固件上周发布了

closed 2023-10-23 by jjm2473