istoreos/istoreos #1012: 关于RK35xx系列机型混用不同作者固件可能出现的问题

jjm2473, 2023-09-22:
RK35xx 系列同时带 eMMC 和 SD/TF卡 的机型，包括：
* 友善 R5S/R5C，R6S/R6C
* 恒领 H66K/H68K/H69K，H88K
* 莱因特 T68M 

等，在混用不同作者固件的情况下，可能出现无法启动，型号识别错误，网口错乱（LAN口和WAN口调换，或缺少网口），无法联网，NPU/GPU报错 等问题，特定条件下还可能损坏硬件。

PS：R4SE 可能也有这个问题。

### 什么是“混用不同作者固件”？
也就是 eMMC 和 SD/TF 卡分别刷了不同作者的固件，例如 eMMC 里面刷了 lean 的 lede 固件，SD/TF 卡刷了 iStoreOS，或者反过来。

### 为什么“混用不同作者固件”会导致问题？
因为 RK35xx 有多种不同的 uboot，不同的 uboot 支持的功能不一样，且同一个 uboot 也可能有不同编译配置。

RK35xx 至少支持这两种 uboot：1. RK 厂家提供的 uboot，2. 上游 uboot（也就是 uboot 官方的，可能会加些补丁）。RK 的 uboot 又有不同的配置，例如是否使用 Android 闪存布局，是否支持 ADC 命令，是否提前打开 NPU 电源，是否初始化 GMAC 等。

iStoreOS 的 RK35xx 机型使用的的是 RK 的 uboot，配置了不使用 Android 闪存布局，需要读取 ADC 的值判断机型，需要提前打开 NPU 电源，需要初始化 GMAC。

RK35xx 系列机型启动时会优先加载 eMMC 的 uboot，然后 uboot 又是优先加载 SD/TF 卡里的固件（为了实现插上 SD/TF 卡就启动里面的系统，拔掉 SD/TF 卡就启动 eMMC 的系统），这种混乱的逻辑导致了 uboot 跟固件系统不一致的情况。
例子1: R5C 是跟 R5S 共用同一个固件的，需要通过 ADC 的值来加载不同机型的配置。假设 R5C  的 eMMC 刷了 lede，SD/TF 卡刷了 iStoreOS，机器启动时加载的是 eMMC 的 uboot，然后启动 SD/TF 卡的 iStoreOS，iStoreOS 在判断机型的时候使用 ADC 命令，由于 lede 的 uboot 不支持 ADC 命令，那么 iStoreOS 就会加载默认机型 R5S 的配置启动，启动以后可能机型显示错误，网口错乱等。
例子2: R5S 的 eMMC 刷了 iStoreOS，SD/TF 卡刷了友善官方的 FriendlyWrt，由于 FriendlyWrt 固件使用的是 Android 闪存布局，iStoreOS 的 uboot 不支持 Android 闪存布局，可能导致无法启动。

### 如何清掉 eMMC 或者 SD/TF 卡的固件（包括 uboot）？
只是删除分区不能清掉 uboot，要清掉 uboot 就要把存储器的前面 16MB 擦除。对于 RK35xx 机型，在启动到 iStoreOS 的情况下：
* 如果要清掉 eMMC 里的固件，执行 `dd if=/dev/zero of=/dev/mmcblk2 bs=4096 count=4096 conv=notrunc`
* 如果要清掉 SD/TF 卡里的固件，执行 `dd if=/dev/zero of=/dev/mmcblk0 bs=4096 count=4096 conv=notrunc`

ks-bao, 2023-09-25:
厉害厉害，有空我加入 FAQ！

gowy222, 2023-10-16:
R5S TF卡完全不起作用, 
机器拿回来默认出厂时FriendlyELEC的，IstoreOS 又是U-boot ，所以 永远走emmc启动。无解。

eMMC当前系统 | TF卡当前系统 | 启动优先级
-- | -- | --
无系统 | 任意固件 | TF卡
FriendlyELEC的固件 | FriendlyELEC的固件 | TF卡
FriendlyELEC的固件 | 采用主线U-boot的固件 | eMMC
采用主线U-boot的固件 | FriendlyELEC的固件 | eMMC
采用主线U-boot的固件 | 采用主线U-boot的固件 | eMMC

aquawius, 2023-10-20:
我不知道 @jjm2473 说的混刷uboot错乱在我的机器上应该怎样验证, 我使用的是hinlink h68k (2023版的).

> RK35xx 系列机型启动时会优先加载 eMMC 的 uboot，然后 uboot 又是优先加载 SD/TF 卡里的固件（为了实现插上 SD/TF 卡就启动里面的系统，拔掉 SD/TF 卡就启动 eMMC 的系统），这种混乱的逻辑导致了 uboot 跟固件系统不一致的情况。

jjm2473, 2023-10-20:
@aquawius 不需要验证啊，这是毕现的，是由于rk的引导逻辑混乱导致的问题。

eronez, 2024-09-07:
R4SE也是混乱的，
emmc->FriendlyELEC的固件
sd->无固件，启动失败

GCY123123, 2024-09-24:
@jjm2473 大佬，求更新适配 友善R3S的iStoreOS

jjm2473, 2024-09-24:
@GCY123123 目前没收到测试机

Loyea, 2025-03-11:
R3S亲测这问题确实存在。R3S 用dd擦除16M也没用。甚者比如说immortalWrt，系统没有mmcblk2，这个命令也没法用。

R3S 在emmc装了iStoreOS之后，无法使用TF启动KWrt(openwrt.ai) 或 immortalWrt , 但是可以TF启动FriendlyWrt，此时给emmc写入immortalWrt之后，再也无法从TF卡启动FriendlyWrt。 最后使用了 @gowy222 的办法，启动时按住mask4秒，进入TF卡的FriendlyWrt.

真是有够蛋疼的，早知道不买emmc的版本了，插TF卡就从TF卡启动根本是诈骗宣传😅

jjm2473, 2025-05-23:
某些机型可能还有spi flash，也要清空掉。

kingwho, 2025-06-14:
您好，我这边测试E20C设备，宽带使用拨号上网无法获取IP地址。

[日志.txt](https://github.com/user-attachments/files/20737993/default.txt)

xu75, 2025-11-15:
怎么办，我也是R4SE，板上的系统就是FriendlyElec的系统。怎样才能换成iStoreOS呢？

tkfanswind, 2025-11-19:
eMMC 刷机助手在software里面是哪个ipk?为啥我搜索不到
我是R3S以前用eMMC启动的时候能看到eMMC 刷机助手.
R3S如何控制用SD卡启动?我的eMMC卡是ImmortalWrt.
我打算用eMMC 刷机助手刷新系统