istoreos/istoreos #1017: 科学上网插件问题

GolemZ2K, 2023-09-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
科学上网插件问题VSSR安装后，国内网站连不上
换SSR-Plus后，模式选绕过大陆IP，发现是全局走的节点，用显示外网IP网站显示是节点的国外IP，换用“GFW列表模式”才能国内的不走节点
其他几个科学上网插件都有类似问题，难到是iStoreOS固件版本问题？版本是22.03.5 2023092209，x86和r2s版的，问题都一样

### (1) 具x体问题

A：VSSR安装后，国内网站连不上
B:  SSR-Plus后，模式选绕过大陆IP，发现是全局走的节点，用显示外网IP网站显示是节点的国外IP，换用“GFW列表模式”才能国内的不走节点

### (2) 路由器型号和固件版本

A：x86和r2s版的22.03.5 2023092209，问题都一样

### (3) 详细日志和/或截图

A：

jjm2473, 2023-09-25:
试试这里的插件 https://github.com/AUK9527/Are-u-ok

GolemZ2K, 2023-09-26:
就是用的这里的插件啊！ 还有别的地方有插件？

jjm2473, 2023-09-27:
怎么没有，还有插件的官方编译版本

jjm2473, 2023-09-27:
看看插件的日志

GolemZ2K, 2023-09-28:
不清楚官方编译版本，https://github.com/AUK9527/Are-u-ok 里的挨个试了，PassWall的全插件版可以正常使用，hyperv装的x86版和R2S版一样的情况。

jjm2473, 2023-09-28:
你是不是没更新geoip.dat之类的文件

jjm2473, 2023-09-28:
你说的“显示外网IP网站”是什么网站，国内还是国外的

GolemZ2K, 2023-09-29:
更新了ip信息，ip111.cn