istoreos/istoreos #1018: iStoreOS 做主路由   UPNP功能    群晖显示不兼容

Pei-Qi, 2023-09-25:
## 1. UPNP功能不兼容

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：群辉系统->路由配置->设置路由器   显示Upnp不兼容
B：其他系统好像是可以用  因为在Upnp界面看到有其他系统的转发

### (2) 路由器型号和固件版本

A： 路由器型  R86S
B： 固件版本  22.03.5 2023092209

### (3) 详细日志和/或截图

[image]
[image]

elvellonwg, 2023-11-08:
> 那问问群晖为什么不兼容，其他程序不都成功开放端口了吗

是现在用的upnp问题  别的openwrt编译出来的upnp群晖都可以识别 只有这个系统的upnp无法识别 是否可以考虑换下upnp版本呢

jjm2473, 2023-11-08:
不是吧，我看openwrt的upnp从22.03到master都是2.3.3。
你试试打开upnp的ipv6支持。执行这段命令：
```
uci -q batch <<-EOF >/dev/null
    set upnpd.config.ipv6_disable=1
    commit upnpd
EOF
/etc/init.d/miniupnpd restart
```

jjm2473, 2023-11-09:
如果打开ipv6还是不行，那么在upnp的高级设置里面关掉强制转发和stun试试

jjm2473, 2023-11-10:
测试了没啊

goeli-ibm, 2023-11-15:
我也遇到了同样的问题，我在服务——unup中设置：
1、启动 UPnP 与 NAT-PMP 服务（不知是否有关）；
2、高级设置——“MiniUPnP 访问控制列表”添加了NAS的IP地址，设置“allow”；
3、重启后群晖NAS——完成设置路由器——通过。

不会贴图，请谅解。

jjm2473, 2023-11-19:
`启动 UPnP 与 NAT-PMP 服务`是upnp服务的总开关，如果都没开，哪有upnp功能。
访问控制列表是默认值，跟openwrt那边一样。

Pei-Qi, 2023-11-23:
> 测试了没啊

最近入手了DS423+  测试了一下UPNP转发功能  
1、群晖识别的也是不兼容、但是多了一个自动和高级转发模式  可以转发成功（系统是DSM7.2 之前无法转发的是 DSM7.1  应该是群晖系统问题）
2、转发成功之后  遇到了  端口冲突问题    导致群晖无法将端口无法映射成功
3、端口冲突之后尝试删除原有端口  重新映射  发现端口无法删除
4、重启系统之后发现 UPUP原有端口都不见了
5、同时会遇到 NAT回流问题（没有找到比较好的解决方式 1、电脑修改hots文件（测试可以解决）  2、搭建DNS服务（未尝试））

解决方法：暂时放弃UPNP  手动转发端口   （之前用小米路由器UPNP转发很稳定  但是也会遇到NAT回流问题）

closed 2023-11-23 by Pei-Qi

jjm2473, 2023-11-24:
试试关掉UPNP的强制转发，因为只有这个功能才是跟openwrt官方不一致的。

moehz, 2024-03-09:
> 试试关掉UPNP的强制转发，因为只有这个功能才是跟openwrt官方不一致的。

我也出现了upnp的功能不起作用的情况，内网设备可以添加进upnp的规则，页面可以看到，但是所有的规则都没有生效。
群晖，plex均显示没有网络，从外网访问打开的upnp端口也确实无法访问

我查看了iptables规则，发现filter表里生成了MINIUPNPD链，nat表里生成了MINIUPNPD链和MINIUPNPD-POSTROUTING链，但是均没有引用，规则没有捕捉到任何包。

简单使用命令在nat表中添加MINIUPNPD添加到prerouting_rule链，MINIUPNPD-POSTROUTING添加到postrouting_rule链,如下：
```bash
iptables -t nat -A prerouting_rule -j MINIUPNPD
iptables -t nat -A postrouting_rule -j MINIUPNPD-POSTROUTING
```
此后再用`iptables -t nat -nvL MINIUPNPD`查看时就发现能捕捉到对应规则的包了，此时在外网通过upnp打通的端口访问Plex时，就可以成功访问了。我不知道是我的什么操作导致了iptables链出现了异常，请问需要我提供什么来帮助定位这个bug？

moehz, 2024-03-09:
请尝试使用命令：
```bash
iptables -t nat -nvL MINIUPNPD
```
看看返回的结果：
```bash
root@iStoreOS:~# iptables -t nat -nvL MINIUPNPD
Chain MINIUPNPD (2 references)
 pkts bytes target     prot opt in     out     source               destination         
    0    0       DNAT       tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            tcp dpt:12057 to:192.168.2.176:32400
    0    0       DNAT       tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            tcp dpt:6881 to:192.168.2.233:6881
    0    0       DNAT       udp  --  *      *       0.0.0.0/0            0.0.0.0/0            udp dpt:6881 to:192.168.2.233:6881
```
前面两列是不是计数都是0，如果是的话，你应该跟我是同一个问题，页面虽然显示了，但并没有生效。
如果有计数，那可能问题还不一样，我再去提一个bug

jjm2473, 2024-03-11:
@moehz 你是不是自己安装了miniupnpd？miniupnpd有两个版本，一个是iptables版，一个是nftables版。按FAQ第2点操作 
 https://github.com/istoreos/istoreos/wiki/FAQ  ，如果还是这样，那就恢复出厂设置，以后不要再乱升级系统软件。

404user, 2024-12-22:
我到现在还是不行，见鬼了，时好时坏的。