istoreos/istoreos #1033: NPS/NPC 商店中的插件没有正常工作

OzzyYang, 2023-10-01:
* [ x] 没有类似的issue

 (1) 具体问题
想把内网的Jellyfin服务开放给外网，所以打算部署NPS服务，本来想直接使用插件，但是出现了以下问题：
A：NPS服务器正常工作，在内网的另一个Linux系统中启动NPC客户端可以正常工作，在iStore OS所在的Linux 命令行中启动NPC也可以正常工作，这两种方式都可以让远程的NPS服务器显示在线，但是在商店的插件中配置无法正常工作，远程的NPS服务器显示离线。
PS：iStore中的两个NPC插件都用过了，都无法正常工作。

### (2) 路由器型号和固件版本

A：路由器型号是畅网的X86-p5，iStore OS和另一个Linux都工作在PVE虚拟机中。
iStore OS：iStoreOS 22.03.5 2023092815
PVE：8.0.3
NPC插件版本：v0.26.10

### (3) 详细日志和/或截图

不知道该在哪里弄日志，所以先给详细截图说明清楚情况吧

直接在iStore OS的后台中启用npc客户端可以正常访问：
[image]
远程服务也可正常显示在线：
[image]
在iStore的插件中进行配置，可以确保配置正常无误：
[image]
远程客户端显示离线：
[image]

OzzyYang, 2023-10-01:
`Mon Oct  2 03:07:26 2023 daemon.info npc[8709]: 2023/10/01 19:07:26.904 [1;34m[I][0m [control.go:102]  Reconnecting...
Mon Oct  2 03:07:32 2023 daemon.info npc[8709]: 2023/10/01 19:07:32.628 [1;31m[E][0m [control.go:231]  EOF
Mon Oct  2 03:07:32 2023 daemon.info npc[8709]: 2023/10/01 19:07:32.628 [1;31m[E][0m [control.go:111]  EOF
Mon Oct  2 03:07:32 2023 daemon.info npc[8709]: 2023/10/01 19:07:32.628 [1;34m[I][0m [control.go:102]  Reconnecting...
Mon Oct  2 03:07:37 2023 daemon.info npc[8709]: 2023/10/01 19:07:37.907 [1;31m[E][0m [control.go:231]  EOF
Mon Oct  2 03:07:37 2023 daemon.info npc[8709]: 2023/10/01 19:07:37.907 [1;31m[E][0m [control.go:111]  EOF
Mon Oct  2 03:07:37 2023 daemon.info npc[8709]: 2023/10/01 19:07:37.907 [1;34m[I][0m [control.go:102]  Reconnecting...
Mon Oct  2 03:07:40 2023 user.notice [NPC]: NPS Client(NPC) Service is now disabled ...
Mon Oct  2 03:07:40 2023 daemon.info npc[8709]: 2023/10/01 19:07:40.113 [1;33m[W][0m [npc.go:201]  stop...
Mon Oct  2 03:12:04 2023 user.notice [NPC]: Starting NPS Client(NPC) ...
Mon Oct  2 03:12:04 2023 daemon.info npc[608]: 2023/10/01 19:12:04.600 [1;34m[I][0m [npc.go:231]  the version of client is 0.26.10, the core version of client is 0.26.0
Mon Oct  2 03:12:04 2023 daemon.info npc[608]: 2023/10/01 19:12:04.605 [1;34m[I][0m [control.go:97]  Loading configuration file /tmp/etc/npc.conf successfully
Mon Oct  2 03:12:05 2023 daemon.info npc[608]: 2023/10/01 19:12:05.246 [1;32m[N][0m [control.go:174]  web access login username:user password:**********
Mon Oct  2 03:12:05 2023 daemon.info npc[608]: 2023/10/01 19:12:05.715 [1;34m[I][0m [client.go:72]  Successful connection with server *************:8025`
刚才尝试重新启动了一下服务npc，重启大法好，发现又正常工作了，报错和正常工作的日志如上，先把issue关了。

closed 2023-10-01 by OzzyYang

haowen2000, 2024-12-01:
> `Mon Oct 2 03:07:26 2023 daemon.info npc[8709]: 2023/10/01 19:07:26.904 �[1;34m[I]�[0m [control.go:102] Reconnecting... Mon Oct 2 03:07:32 2023 daemon.info npc[8709]: 2023/10/01 19:07:32.628 [...]

请教一下，你的istoreos商店里面的NPS的WEB管理端怎么打开的？我安装 后都没有相应的WEB端设置选项？