istoreos/istoreos #1036: BCM57810S不支持2.5G速率

jac88288, 2023-10-05:
在EXSi虚拟机上用bcm57810s网卡能识别到2.5G猫棒，单不能用到2.5G速率，只用到1G，如果使用爱快8就能用到2.5G！是否驱动不支持

jjm2473, 2023-10-09:
这个网卡对应的驱动是什么？执行`lspci -nnk`看看输出

jac88288, 2023-10-09:
Ethernet controller [0200]: Broadcom Inc. and subsidiaries NetXtreme II BCM57810 10 Gigabit Ethernet [14e4:168e] (rev 10)
        DeviceName: pciPassthru0
        Subsystem: Hitachi, Ltd Device [1054:3087]
        Kernel driver in use: bnx2x

jac88288, 2023-10-09:
你们的iStoreOS软件包里是有bnx2x，但就是不能用！

jac88288, 2023-10-09:
我用了
iStoreOS 23.05-SNAPSHOT 2023092209
iStoreOS 22.03.5-2023092815-x86-64-squashfs-combined
两个版本都试过了！不能使用

jac88288, 2023-10-15:
[993-bnx2x_warpcore_8727_2_5g_sgmii_txfault.patch](https://github.com/istoreos/istoreos/files/12909047/993-bnx2x_warpcore_8727_2_5g_sgmii_txfault.patch)
是否缺少了这驱动编译代码

jac88288, 2023-10-19:
我用iStoreOS 23.05源码编译了固件，加入[993-bnx2x_warpcore_8727_2_5g_sgmii_txfault.patch](https://github.com/istoreos/istoreos/files/12909047/993-bnx2x_warpcore_8727_2_5g_sgmii_txfault.patch)就支持2.5G速率了！
[image]
[image]
[image]

jjm2473, 2023-10-23:
谢谢，补丁加上了

closed 2023-10-23 by jjm2473

jac88288, 2023-10-23:
[image]
请教一下这里的版本日期序号怎么添加与修改

jjm2473, 2023-10-24:
就是版本号，例如r123456或者日期，正常情况下是会自动通过git产生r开头的版本号，可能你的系统缺少git。还有个简单的方法，就是直接写一串字符到`version`文件，例如在源代码目录执行`echo 2023102410 > version`，之后编译出来的固件就会带2023102410这个版本号

jjm2473, 2023-10-24:
最好还是使用22版分支吧，23版还是之前的测试版本

jac88288, 2023-10-24:
我用23版编译了自己需要的应用在用了，挺好的！

jac88288, 2023-10-31:
你好！再请教你一下，请问编译时怎么把lan端口改为默认eth0

jjm2473, 2023-10-31:
修改 `/etc/board.d` 里面的 network 相关文件