istoreos/istoreos #1037: R4SE有什么办法驱动CF-811AC呢

Jeafly, 2023-10-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志和/或截图

A：

jjm2473, 2023-10-06:
R4SE的固件应该集成了一些usb无线网卡驱动，如果用不了，那可能就是没驱动了。如果你知道驱动名称，可以在软件包里搜索看看。

Jeafly, 2023-10-07:
> R4SE的固件应该集成了一些usb无线网卡驱动，如果用不了，那可能就是没驱动了。如果你知道驱动名称，可以在软件包里搜索看看。

软件包可以下载到cf812ac的驱动，但是下载了没有反应

jjm2473, 2023-10-07:
@Jeafly 安装驱动以后重启下试试，看看“网络”菜单有没有出现“无线”

Jeafly, 2023-10-07:
> @Jeafly 安装驱动以后重启下试试，看看“网络”菜单有没有出现“无线”

没有，有没有已经实测在r4s的itoreos上驱动的usb无线网卡呢

jjm2473, 2023-10-07:
集成的驱动有这些，具体支持哪些网卡型号就不知道了。
```
  - kmod-mt76x0u
  - kmod-mt76x2u
  - kmod-mt7663u
  - kmod-mt7663-firmware-ap
  - kmod-mt7921u
  - kmod-mt7921-firmware
  - kmod-rtl8192cu
  - kmod-rtl8812au-ct
  - kmod-rtl8xxxu
 ```

jjm2473, 2023-10-07:
cf812ac的驱动名称是什么

Jeafly, 2023-10-07:
> cf812ac的驱动名称是什么

***kmod-rtl8812au-ct*** Driver for Realtek 8812 AU devices comfast 912-ac, etc

***设备型号*** :  FriendlyElec NanoPi R4SE 
***固件版本*** : iStoreOS 22.03.5 2023092815 
***内核版本*** : 5.10.176 

```shell
root@iStoreOS:~# lsusb
Bus 007 Device 001: ID 1d6b:0002 Linux 5.10.176 xhci-hcd xHCI Host Controller
Bus 002 Device 001: ID 1d6b:0002 Linux 5.10.176 ehci_hcd EHCI Host Controller
Bus 004 Device 001: ID 1d6b:0001 Linux 5.10.176 ohci_hcd Generic Platform OHCI controller
Bus 006 Device 001: ID 1d6b:0003 Linux 5.10.176 xhci-hcd xHCI Host Controller
Bus 008 Device 001: ID 1d6b:0003 Linux 5.10.176 xhci-hcd xHCI Host Controller
Bus 005 Device 002: ID 0bda:b812 Realtek 802.11ac NIC
Bus 001 Device 001: ID 1d6b:0002 Linux 5.10.176 ehci_hcd EHCI Host Controller
Bus 003 Device 001: ID 1d6b:0001 Linux 5.10.176 ohci_hcd Generic Platform OHCI controller
Bus 005 Device 001: ID 1d6b:0002 Linux 5.10.176 xhci-hcd xHCI Host Controller
```

```shell
root@iStoreOS:~# lsmod
· · · · · ·
mt76                   57344 13 mt76x0u,mt76x0_common,mt7921u,mt7921_common,mt76x2u,mt76x2_common,mt76x02_usb,mt76x02_lib,mt7663u,mt7663_usb_sdio_common,mt7615_common,mt76_usb,mt76_connac_lib
mt76_connac_lib        45056  5 mt7921u,mt7921_common,mt7663u,mt7663_usb_sdio_common,mt7615_common
mt76_usb               24576  5 mt76x0u,mt7921u,mt76x2u,mt76x02_usb,mt7663u
mt7615_common          77824  2 mt7663u,mt7663_usb_sdio_common
mt7663_usb_sdio_common   16384  1 mt7663u
mt7663u                16384  0 
mt76x0_common          40960  1 mt76x0u
mt76x02_lib            57344  5 mt76x0u,mt76x0_common,mt76x2u,mt76x2_common,mt76x02_usb
mt76x02_usb            20480  2 mt76x0u,mt76x2u
mt76x0u                16384  0 
mt76x2_common          24576  1 mt76x2u
mt76x2u                24576  0 
mt7921_common          57344  1 mt7921u
mt7921u                20480  0 
· · · · · ·
rtl_pci                32768  1 rtl8821ae
rtl_usb                20480  1 rtl8192cu
rtl8150                20480  0 
rtl8192c_common        40960  1 rtl8192cu
rtl8192cu              73728  0 
rtl8812au            1286144  0 
rtl8821ae             212992  0 
rtl8xxxu              102400  0 
rtlwifi                69632  6 rtl8821ae,rtl8192cu,rtl8192c_common,rtl_usb,rtl_pci,btcoexist
· · · · · ·
```

```shell
root@iStoreOS:~# wifi detect
WARNING: Wifi detect is deprecated. Use wifi config instead
For more information, see commit 5f8f8a366136a07df661e31decce2458357c167a
```
重启后网络里面没有出现无线

jjm2473, 2023-10-07:
重启下，把开机的日志发出来看看

jjm2473, 2023-10-07:
还有，执行`lsusb -tv`才能输出驱动名称

Jeafly, 2023-10-08:
> 重启下，把开机的日志发出来看看

系统日志还是内核日志呢

Jeafly, 2023-10-08:
```shell
root@iStoreOS:~# lsusb -tv
/:  Bus 08.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
    ID 1d6b:0003  
/:  Bus 07.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 480M
    ID 1d6b:0002  
/:  Bus 06.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
    ID 1d6b:0003  
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 480M
    ID 1d6b:0002  
    |__ Port 1: Dev 2, If 0, Class=, Driver=, 480M
        ID 0bda:b812  
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=ohci-platform/1p, 12M
    ID 1d6b:0001  
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=ohci-platform/1p, 12M
    ID 1d6b:0001  
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ehci-platform/1p, 480M
    ID 1d6b:0002  
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-platform/1p, 480M
    ID 1d6b:0002  
```

系统日志：
```shell
Sun Oct  8 18:10:37 2023 daemon.debug ModemManager[4124]: hotplug: event reported: action=add, name=ptypc, subsystem=tty
Sun Oct  8 18:10:37 2023 daemon.err ModemManager[4124]: hotplug: parent device sysfspath not found
Sun Oct  8 18:10:37 2023 daemon.debug ModemManager[4124]: hotplug: cached event found: action=add, name=tty9, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty9
Sun Oct  8 18:10:37 2023 daemon.debug ModemManager[4124]: hotplug: event reported: action=add, name=tty9, subsystem=tty
Sun Oct  8 18:10:37 2023 daemon.err ModemManager[4124]: hotplug: parent device sysfspath not found
Sun Oct  8 18:10:37 2023 daemon.notice netifd: wan (6491): udhcpc: broadcasting select for 192.168.1.2, server 192.168.1.1
Sun Oct  8 18:10:37 2023 daemon.debug ModemManager[4124]: hotplug: cached event found: action=add, name=tty53, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty53
Sun Oct  8 18:10:37 2023 daemon.debug ModemManager[4124]: hotplug: event reported: action=add, name=tty53, subsystem=tty
Sun Oct  8 18:10:37 2023 daemon.err ModemManager[4124]: hotplug: parent device sysfspath not found
Sun Oct  8 18:10:37 2023 daemon.debug ModemManager[4124]: hotplug: cached event found: action=add, name=tty15, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty15
Sun Oct  8 18:10:37 2023 daemon.debug ModemManager[4124]: hotplug: event reported: action=add, name=tty15, subsystem=tty
Sun Oct  8 18:10:37 2023 daemon.err ModemManager[4124]: hotplug: parent device sysfspath not found
Sun Oct  8 18:10:37 2023 daemon.err zerotier-one[8856]: Starting Control Plane...
Sun Oct  8 18:10:37 2023 daemon.err zerotier-one[8856]: Starting V6 Control Plane...
Sun Oct  8 18:10:37 2023 daemon.debug ModemManager[4124]: hotplug: cached event found: action=add, name=ttyp5, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttyp5
Sun Oct  8 18:10:37 2023 daemon.debug ModemManager[4124]: hotplug: event reported: action=add, name=ttyp5, subsystem=tty
Sun Oct  8 18:10:37 2023 daemon.err ModemManager[4124]: hotplug: parent device sysfspath not found
Sun Oct  8 18:10:37 2023 daemon.debug ModemManager[4124]: hotplug: cached event found: action=add, name=ttypc, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttypc
Sun Oct  8 18:10:37 2023 daemon.debug ModemManager[4124]: hotplug: event reported: action=add, name=ttypc, subsystem=tty
Sun Oct  8 18:10:37 2023 daemon.err ModemManager[4124]: hotplug: parent device sysfspath not found
Sun Oct  8 18:10:37 2023 daemon.debug ModemManager[4124]: hotplug: cached event found: action=add, name=tty43, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty43
Sun Oct  8 18:10:37 2023 daemon.debug ModemManager[4124]: hotplug: event reported: action=add, name=tty43, subsystem=tty
Sun Oct  8 18:10:37 2023 daemon.err ModemManager[4124]: hotplug: parent device sysfspath not found
Sun Oct  8 18:10:37 2023 daemon.notice netifd: wan (6491): udhcpc: lease of 192.168.1.2 obtained from 192.168.1.1, lease time 86400
Sun Oct  8 18:10:37 2023 daemon.debug ModemManager[4124]: hotplug: cached event found: action=add, name=ttyS3, subsystem=tty, sysfspath=/sys/devices/platform/serial8250/tty/ttyS3
Sun Oct  8 18:10:37 2023 daemon.debug ModemManager[4124]: hotplug: event reported: action=add, name=ttyS3, subsystem=tty
Sun Oct  8 18:10:37 2023 daemon.err ModemManager[4124]: hotplug: parent device sysfspath not found
Sun Oct  8 18:10:37 2023 daemon.debug ModemManager[4124]: hotplug: cached event found: action=add, name=tty33, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty33
Sun Oct  8 18:10:37 2023 daemon.debug ModemManager[4124]: hotplug: event reported: action=add, name=tty33, subsystem=tty
Sun Oct  8 18:10:37 2023 daemon.err ModemManager[4124]: hotplug: parent device sysfspath not found
Sun Oct  8 18:10:37 2023 daemon.debug ModemManager[4124]: hotplug: cached event found: action=add, name=tty61, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty61
Sun Oct  8 18:10:37 2023 daemon.debug ModemManager[4124]: hotplug: event reported: action=add, name=tty61, subsystem=tty
Sun Oct  8 18:10:37 2023 daemon.err ModemManager[4124]: hotplug: parent device sysfspath not found
Sun Oct  8 18:10:37 2023 daemon.info avahi-daemon[4051]: Joining mDNS multicast group on interface eth0.IPv4 with address 192.168.1.2.
Sun Oct  8 18:10:37 2023 daemon.info avahi-daemon[4051]: New relevant interface eth0.IPv4 for mDNS.
Sun Oct  8 18:10:37 2023 daemon.info avahi-daemon[4051]: Registering new address record for 192.168.1.2 on eth0.IPv4.
Sun Oct  8 18:10:37 2023 daemon.notice netifd: Interface 'wan' is now up
Sun Oct  8 18:10:37 2023 daemon.info dnsmasq[1]: reading /tmp/resolv.conf.d/resolv.conf.auto
[... cut ...]
[   43.483854] IPv6: ADDRCONF(NETDEV_CHANGE): veth65bc274: link becomes ready
[   43.484613] br-e6776e6d5f46: port 1(veth65bc274) entered blocking state
[   43.485194] br-e6776e6d5f46: port 1(veth65bc274) entered forwarding state
[   43.485951] IPv6: ADDRCONF(NETDEV_CHANGE): veth4cc1761: link becomes ready
[   43.486637] br-f5cd51eef460: port 2(veth4cc1761) entered blocking state
[   43.487214] br-f5cd51eef460: port 2(veth4cc1761) entered forwarding state
[   43.487920] IPv6: ADDRCONF(NETDEV_CHANGE): br-e6776e6d5f46: link becomes ready
```

jjm2473, 2023-10-08:
目前的驱动不支持，你说的那个驱动也是给cf-912ac用的，而不是什么cf-812ac。根据网上资料，CF-811AC使用的驱动是rtl8822bu https://linux-hardware.org/?id=usb:0bda-b812 ，目前openwrt还没有这个驱动，需要移植。

Jeafly, 2023-10-08:
> 目前的驱动不支持，你说的那个驱动也是给cf-912ac用的，而不是什么cf-812ac。根据网上资料，CF-811AC使用的驱动是rtl8822bu https://linux-hardware.org/?id=usb:0bda-b812 ，目前openwrt还没有这个驱动，需要移植。

好的

jjm2473, 2023-10-27:
已支持

closed 2023-10-27 by jjm2473

huixirun163, 2024-12-07:
Qualcomm Atheros AR928X Wireless Network Adapter (PCI-Express) (rev 01)
谁有这个驱动啊