istoreos/istoreos #1038: 新版istoreos使用时会突然断网甚至会自己重启

NEA147, 2023-10-06:
我的设备是x86 cpu是1125G4 istoreos版本是22.03.5 2023092815目前最新版
在使用的时候会自己突然断网设备也访问不到，甚至有几次会自己重启。
****
这是一次突然重启的详细日志
****
[image]
[image]
[image]

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
希望解决。

jjm2473, 2023-10-06:
没有测试机，你可以试试23.05测试固件看有没有问题
 
EFI： https://fw0.koolcenter.com/iStoreOS/alpha/x86_64_efi/istoreos-23.05-SNAPSHOT-2023083016-x86-64-squashfs-combined-efi.img.gz
非EFI https://fw0.koolcenter.com/iStoreOS/alpha/x86_64/istoreos-23.05-SNAPSHOT-2023083016-x86-64-squashfs-combined.img.gz

还可以刷Openwrt官方的22.03固件试试，可能需要自己安装些驱动

NEA147, 2023-10-06:
> 没有测试机，你可以试试23.05测试固件看有没有问题
>
> EFI： https://fw0.koolcenter.com/iStoreOS/alpha/x86_64_efi/istoreos-23.05-SNAPSHOT-2023083016-x86-64-squashfs-combined-efi.img.gz 非EFI [...]

尝试了测试版istoreos23.05的固件之后解决问题了

shen-lan, 2023-10-07:
我的也是， pve下的istore。昨天又重装了 istore,   看日志依然会偶尔wan接口 down,  up切换， 亚运会看直播  我真是心态爆炸了

jjm2473, 2023-10-09:
@shen-lan pve直通网卡吗？网卡是什么型号？有没有看最新固件的更新说明？

shen-lan, 2023-10-09:
@jjm2473 谢谢老哥解答。🤙🤙🤙是pve网卡直通，n5105的工控机，网卡是i225。现在物理机重装的istore, 一天多了没有重启。观察下有重启再反馈

closed 2024-02-23 by jjm2473