istoreos/istoreos #1042: H69K 5G模块问题

SebaSs007, 2023-10-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；
H69K 5G模块“假死”
Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：H69K 5G模块“假死”，rm520ngl 使用过程中出现断网，但wwan 网口、IP等都存在，但无法上网 却又可以使用at命令可查询模块信息，掉了又没完全掉 。 这时就只能重启解决
以前H69K旧版本 掉模块网口直接提示不存在wwan也没了。

### (2) 路由器型号和固件版本

A：H69K   iStoreOS 22.03.5 2023092815

### (3) 详细日志和/或截图

A：
日志：[ 2913.526064] ------------[ cut here ]------------
[ 2913.526158] NETDEV WATCHDOG: wwan0 (qmi_wwan): transmit queue 0 timed out
[ 2913.526300] WARNING: CPU: 0 PID: 0 at dev_watchdog+0x298/0x2a0
[ 2913.526315] Modules linked in: xt_FULLCONENAT ath9k ath9k_htc ath9k_common rtw88_8822ce rtw88_8822c rtw88_8822be rtw88_8822b rtw88_8723de rtw88_8723d rtl8821ae rtl8192se rtl8192de rtl8192cu rtl8192c_common rtl_usb rtl_pci mt76x0u mt76x0_common iwlmvm iwldvm cdc_mbim btcoexist brcmfmac b43 ath9k_hw ath6kl_usb ath6kl_core ath10k_pci ath10k_core ath wireguard uvcvideo sr9700 snd_usb_audio smsc95xx sierra_net rtw88_pci rtw88_core rtlwifi rtl8xxxu rtl8812au rsi_usb rsi_91x rndis_host qmi_wwan qcserial pppoe plusb option mwifiex_pcie mwifiex mt7921u mt7921e mt7921_common mt7915e mt76x2u mt76x2_common mt76x02_usb mt76x02_lib mt7663u mt7663_usb_sdio_common mt7615e mt7615_common mt76_usb mt76_connac_lib mt76 mmc_spi mcs7830 mac80211 libchacha20poly1305 l2tp_ppp kalmia iwlwifi ipw ipt_REJECT huawei_cdc_ncm dm9601 cfg80211 cdc_subset cdc_ncm cdc_ether cdc_eem ax88179_178a asix aqc111 zstd xt_time xt_tcpudp xt_tcpmss xt_string xt_statistic xt_state xt_socket xt_recent xt_quota2 xt_quota xt_policy
[ 2913.527057]  xt_pkttype xt_physdev xt_owner xt_nat xt_multiport xt_mark xt_mac xt_lscan xt_limit xt_length2 xt_length xt_iprange xt_ipp2p xt_iface xt_hl xt_helper xt_geoip xt_fuzzy xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connlabel xt_connbytes xt_condition xt_comment xt_cgroup xt_bpf xt_addrtype xt_TPROXY xt_TCPMSS xt_REDIRECT xt_NETMAP xt_MASQUERADE xt_LUA xt_LOGMARK xt_LOG xt_IPMARK xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY xt_ACCOUNT xr_usb_serial_common visor videobuf2_v4l2 videobuf2_common usbnet usblp usbhid usbatm usb_wwan usb_serial_simple ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda tulip ts_fsm ts_bm ti_usb_3410_5052 snd_usbmidi_lib smsc sky2 skge sierra sfp sfc_falcon sfc sch_mqprio sch_cake rtl8150 r8152 pptp pppox ppp_synctty ppp_mppe ppp_async poly1305_neon pl2303 pegasus pcnet32 oti6858 of_mmc_spi ntfs3 nlmon niu nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4
[ 2913.527791]  nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_rtsp nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv6 nf_log_ipv4 nf_log_common nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtsp nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount mos7840 mos7720 mdio_i2c mdio_gpio mdio_bitbang mct_u232 macvlan lzo_rle lzo libcurve25519_generic lib80211_crypt_wep lib80211_crypt_tkip lib80211_crypt_ccmp lib80211 keyspan kaweth iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ah ipt_ECN ipheth ip_tables io_edgeport iavf hso hid_generic hid_cp2112 garmin_gps ftdi_sio forcedeth ezusb ethoc et131x dmx_usb cypress_m8 crc_ccitt cp210x cordic compat_xtables compat cm109 cls_flower chaoskey ch341 cdc_wdm cdc_acm brcmutil br_netfilter bnx2x bnx2 belkin_sa be2net ax88796b atl2 atl1e atl1c atl1 at24 ark3116 alx act_vlan
[ 2913.528456]  pktgen 8139too 8139cp sch_teql sch_sfq sch_multiq sch_gred sch_fq sch_dsmark sch_codel em_text em_nbyte em_meta em_cmp act_simple act_pedit act_csum em_ipset cls_bpf act_bpf act_ctinfo act_connmark sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact configs videobuf2_vmalloc videobuf2_memops sg hid videodev evdev ledtrig_usbport trelay siit xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink net2280 st sr_mod ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT ip6t_rt ip6t_mh ip6t_ipv6header ip6t_hbh ip6t_frag ip6t_eui64 ip6t_ah ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6
[ 2913.529105]  pppoatm nfsd bonding 3c59x ip6_gre ip_gre gre r8168 r8125 igc e1000 e100 ifb dummy nat46 l2tp_ip6 l2tp_ip l2tp_eth ip6_vti ip_vti sit sctp mdio l2tp_netlink l2tp_core ipcomp6 xfrm6_tunnel esp6 ah6 xfrm4_tunnel ipcomp esp4 ah4 ipip ip6_tunnel netlink_diag tunnel6 tunnel4 ip_tunnel udp_diag tcp_diag raw_diag inet_diag hfcpci hfcmulti rpcsec_gss_krb5 dm_raid veth tun snd_pcm_oss snd_mixer_oss snd_hwdep snd_compress nbd mISDN_dsp l1oip mISDN_core xfrm_ipcomp af_key udf minix hfsplus hfs cramfs cifs binfmt_misc br2684 atm aoe multipath bcma raid456 async_raid6_recov async_pq async_xor async_memcpy async_tx raid10 raid1 raid0 linear md_mod nls_iso8859_1 zram zsmalloc natsemi vxlan udp_tunnel ip6_udp_tunnel macsec seqiv echainiv deflate cts authenc crypto_acompress ohci_pci uhci_hcd sata_via sata_sil24 sata_sil ata_piix pata_pdc202xx_old sata_nv pata_artop fsl_mph_dr_of ehci_platform ehci_fsl igb gpio_button_hotplug xfs reiserfs jfs f2fs btrfs xor zstd_compress zlib_deflate xor_neon
[ 2913.529986]  raid6_pq lzo_compress dm_mirror dm_region_hash dm_log dm_crypt button_hotplug tg3 ssb ptp pps_core encrypted_keys trusted tpm
[ 2913.530115] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 5.10.176 #0
[ 2913.530127] Hardware name: HINLINK OPC-H69K Board (DT)
[ 2913.530147] pstate: 20400005 (nzCv daif +PAN -UAO -TCO BTYPE=--)
[ 2913.530170] pc : dev_watchdog+0x298/0x2a0
[ 2913.530185] lr : dev_watchdog+0x298/0x2a0
[ 2913.530194] sp : ffffffc0113abda0
[ 2913.530206] x29: ffffffc0113abda0 x28: ffffffc010fdb728 
[ 2913.530230] x27: 00000000ffffffff x26: 0000000000000000 
[ 2913.530252] x25: ffffff81fe394498 x24: 0000000000000140 
[ 2913.530274] x23: ffffff81fe3943e0 x22: ffffff81fcfd7e80 
[ 2913.530296] x21: ffffffc011137000 x20: ffffff81fe394000 
[ 2913.530318] x19: 0000000000000000 x18: 0000000000000473 
[ 2913.530339] x17: 0000000000000000 x16: 0000000000000000 
[ 2913.530360] x15: ffffffc011153eb8 x14: 0000000000000d59 
[ 2913.530381] x13: 0000000000000473 x12: ffffffc0113aba68 
[ 2913.530402] x11: ffffffc011203eb8 x10: 00000000ffffe000 
[ 2913.530423] x9 : ffffffc011203eb8 x8 : 0000000000000000 
[ 2913.530444] x7 : ffffffc011153eb8 x6 : 0000000000000001 
[ 2913.530465] x5 : 0000000000000000 x4 : 0000000000000000 
[ 2913.530485] x3 : 0000000000000040 x2 : 0000000000000008 
[ 2913.530505] x1 : 0000000000000004 x0 : 000000000000003d 
[ 2913.530527] Call trace:
[ 2913.530549]  dev_watchdog+0x298/0x2a0
[ 2913.530577]  call_timer_fn.constprop.0+0x24/0x80
[ 2913.530593]  __run_timers.part.0+0x1d4/0x20c
[ 2913.530605]  run_timer_softirq+0x3c/0x74
[ 2913.530620]  _stext+0x11c/0x280
[ 2913.530638]  irq_exit+0xa0/0xf0
[ 2913.530658]  __handle_domain_irq+0x80/0xe0
[ 2913.530679]  gic_handle_irq+0xc0/0x140
[ 2913.530692]  el1_irq+0xcc/0x180
[ 2913.530714]  cpuidle_enter_state+0x130/0x31c
[ 2913.530726]  cpuidle_enter+0x38/0x50
[ 2913.530745]  do_idle+0x1d4/0x230
[ 2913.530756]  cpu_startup_entry+0x24/0x60
[ 2913.530776]  rest_init+0xbc/0xcc
[ 2913.530796]  arch_call_rest_init+0x10/0x1c
[ 2913.530808]  start_kernel+0x4dc/0x4f8
[ 2913.530821] ---[ end trace fe1df6e38133b052 ]---

SebaSs007, 2023-10-10:
[image]

SebaSs007, 2023-10-10:
[image]
[image]

SebaSs007, 2023-10-10:
网断了，，，但是能发AT命令，，，/dev/里也有
[image]

jjm2473, 2023-10-21:
这个问题不是USB的锅就是5G模块的锅，可能降速使用可以解决。AT有个重启模块的命令，查查手册

closed 2023-10-21 by jjm2473

SebaSs007, 2024-02-04:
@wyb4112 
群大佬给的插件，不过只适配了几个模块。

wyb4112, 2024-02-04:
> @wyb4112 群大佬给的插件，不过只适配了几个模块。
能否把插件我下,谢谢
wyb4112@gmail.com

SebaSs007, 2024-02-05:
@wyb4112 我最近太忙了，有空后我给你想办法找找。

jjm2473, 2024-02-05:
执行AT命令可以使用picocom，软件包里面就能安装，但是自己要知道哪个tty是对应5G模块的（一般是ttyUSB开头）