istoreos/istoreos #1044: zerotier-cli命令报错

GolemZ2K, 2023-10-11:
### (1) 具体问题
zerotier正常运行，想添加moon服务器时，运行zerotier-cli出错
zerotier-cli: missing port and zerotier-one.port not found in /var/lib/zerotier-one

### (2) 路由器型号和固件版本
R2S   22.03.5 2023092815

rumianoesa, 2023-10-12:
最近的更新之后/var/lib路径默认生成的zerotier-one变成zerotier-luci了
临时解决办法：可以先重命名成zerotier-one再运行zerotier-cli
`mv /var/lib/zerotier-luci /var/lib/zerotier-one`

GolemZ2K, 2023-10-14:
> 最近的更新之后/var/lib路径默认生成的zerotier-one变成zerotier-luci了 临时解决办法：可以先重命名成zerotier-one再运行zerotier-cli `mv /var/lib/zerotier-luci /var/lib/zerotier-one`

多谢，搞定了，我用的 cp -r 复制了目录

closed 2023-10-17 by GolemZ2K

caesaranthony, 2024-03-19:
> > 最近的更新之后/var/lib路径默认生成的zerotier-one变成zerotier-luci了 临时解决办法：可以先重命名成zerotier-one再运行zerotier-cli `mv /var/lib/zerotier-luci /var/lib/zerotier-one`
> 
> 多谢，搞定了，我用的 cp -r 复制了目录

但是如果重启路由器，新复制的目录会消失，需要重新复制，挺不方便的。

codingiran, 2024-03-24:
> > 最近的更新之后/var/lib路径默认生成的zerotier-one变成zerotier-luci了 临时解决办法：可以先重命名成zerotier-one再运行zerotier-cli `mv /var/lib/zerotier-luci /var/lib/zerotier-one`
> 
> 多谢，搞定了，我用的 cp -r 复制了目录

重启 iStoreOS 之后都需要操作一遍，建议打开 issue，参考 https://github.com/coolsnowwolf/lede/issues/11974

Git-XY, 2025-05-11:
此问题可以修改luci_zerotier文件配置  使用命令 vi /etc/init.d/luci_zerotier 将CONFIG_PATH=/var/lib/zerotier-luci 改为CONFIG_PATH=/var/lib/zerotier-one，亲测可行，重启路由正常