istoreos/istoreos #1046: 旁路由防火墙设置

demonsaidhelloworld, 2023-10-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
旁路由网络拓扑图：
[image]
众所周知，旁路由在配置的时候一定要添加一条防火墙规则：

**_”iptables -t nat -I POSTROUTING -o eth0 -j MASQUERADE“_**

否则主路由与旁路由网关互指之后局域网内设备均无法上网
### (1) 具体问题

A：那为什么此功能不能直接集成在旁路由的快速配置中呢？

### (2) 路由器型号和固件版本

A：主路由：tp-link ax3000
      旁路由：r2s 
     固件版本：iStoreOS 22.03.5 2023092815 / LuCI istoreos-22.03 branch git-23.271.10849-b445bc4

### (3) 详细日志和/或截图

A：
[image]

jjm2473, 2023-10-21:
众所周知个屁啊，这命令就相当于在防火墙lan区域勾选“IP动态伪装”。不是所有人都需要这个功能，只有某些AP或无线路由器有bug的才需要开。

而且这个问题早就有issue总结过了 https://github.com/istoreos/istoreos/issues/598

closed 2023-10-21 by jjm2473

demonsaidhelloworld, 2023-10-21:
哦哦，早就已经不用这个了，bug太多了