istoreos/istoreos #1048: R68s更新固件后无法上网

mobyoung, 2023-10-13:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
R68s更新固件后无法上网

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [× ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：正在使用的版本是istoreos-202304211-r68s，试过在线升级和手动升级在这版本之后的版本，包括2023092815-r6x版本，升级选择保留设置，成功更新，但无法联网，状态一直为检测中，IP地址空白。

### (2) 路由器型号和固件版本

A：路由器是R68S

### (3) 详细日志和/或截图

A：
Sat Oct 14 00:44:53 2023 daemon.info samba4-server: Legacy Protocols allowed, don't use this option for secure environments!
Sat Oct 14 00:44:53 2023 daemon.info samba4-server: io_uring module found, enabling VFS io_uring. (also needs Kernel 5.4+ Support)
Sat Oct 14 00:45:06 2023 daemon.warn pppd[15942]: Timeout waiting for PADO packets
Sat Oct 14 00:45:06 2023 daemon.err pppd[15942]: Unable to complete PPPoE Discovery
Sat Oct 14 00:45:06 2023 daemon.info pppd[15942]: Exit.
Sat Oct 14 00:45:06 2023 daemon.notice netifd: Interface 'wan' is now down
Sat Oct 14 00:45:06 2023 daemon.notice netifd: Interface 'wan' is setting up now
Sat Oct 14 00:45:06 2023 daemon.err insmod: module is already loaded - pppox
Sat Oct 14 00:45:06 2023 daemon.err insmod: module is already loaded - pppoe
Sat Oct 14 00:45:06 2023 daemon.info pppd[16940]: Plugin pppoe.so loaded.
Sat Oct 14 00:45:06 2023 daemon.info pppd[16940]: PPPoE plugin from pppd 2.4.9
Sat Oct 14 00:45:06 2023 daemon.notice pppd[16940]: pppd 2.4.9 started by root, uid 0
Sat Oct 14 00:45:07 2023 daemon.info samba4-server: Legacy Protocols allowed, don't use this option for secure environments!
Sat Oct 14 00:45:07 2023 daemon.info samba4-server: io_uring module found, enabling VFS io_uring. (also needs Kernel 5.4+ Support)
Sat Oct 14 00:45:15 2023 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Sat Oct 14 00:45:15 2023 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 61 names
Sat Oct 14 00:45:15 2023 daemon.info dnsmasq[1]: read /tmp/hosts/odhcpd - 0 names
Sat Oct 14 00:45:15 2023 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Sat Oct 14 00:45:16 2023 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.1.145 f0:03:8c:9a:27:3b
Sat Oct 14 00:45:16 2023 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.1.145 f0:03:8c:9a:27:3b Mr-Young
Sat Oct 14 00:45:16 2023 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Sat Oct 14 00:45:16 2023 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 61 names
Sat Oct 14 00:45:16 2023 daemon.info dnsmasq[1]: read /tmp/hosts/odhcpd - 2 names
Sat Oct 14 00:45:21 2023 daemon.warn pppd[16940]: Timeout waiting for PADO packets
Sat Oct 14 00:45:21 2023 daemon.err pppd[16940]: Unable to complete PPPoE Discovery
Sat Oct 14 00:45:21 2023 daemon.info pppd[16940]: Exit.

jjm2473, 2023-10-21:
升级以后，后台显示的设备型号正确吗，有没有识别成r66s？

mobyoung, 2023-10-23:
> 升级以后，后台显示的设备型号正确吗，有没有识别成r66s？

我刚刚又试了一下升级到iStoreOS 22.03.5 2023102014，设备型号是正确识别成R68S的，无法上网的问题依然存在。试过重启，重新设置网络，都不行。网络>接口那里eth0显示 尝试连接失败。
受其他问题的启发，我试着将WAN口从eth0换成eth1，结果就能正常联网了，这个问题应该就是合并了R66S和R68S导致的。
但是现在接口那里多出了一个wan_6(pppoe-wan)的虚拟动态接口 (DHCPv6 客户端)，点击修改就提示
TypeError
Cannot read properties of null (reading '.name')
也删除不了。

jjm2473, 2023-10-23:
wan_6是动态生成的，本来就没相应的配置，自然也删不了，只要禁用wan的高级设置中的获取ipv6就不不会出现了。
我怀疑你的eth0的mac地址被运营商加黑了，所以才拨不了号。r68s的eth0和eth1都是内置的千兆网口，理论上没什么区别。

jjm2473, 2023-10-23:
执行 `ip link` 看看mac地址是不是正常的。

jjm2473, 2023-10-23:
istore里面安装“MAC地址生成器”，然后给wan口随机生成一个MAC地址试试

mobyoung, 2023-10-23:
> 执行 `ip link` 看看mac地址是不是正常的。

> wan_6是动态生成的，本来就没相应的配置，自然也删不了，只要禁用wan的高级设置中的获取ipv6就不会出现了。 我怀疑你的eth0的mac地址被运营商加黑了，所以才拨不了号。r68s的eth0和eth1都是内置的千兆网口，理论上没什么区别。

应该不是加黑吧，因为我只要回退到旧版本就能够正常使用eth0拨号。
执行了ip link 命令，结果如下：
` -------------------------------------------------------
 iStoreOS 22.03.5, 2023102014
 -------------------------------------------------------
root@iStoreOS:~# ip link
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP mode DEFAULT group default qlen 1000
    link/ether 32:94:fd:4a:da:69 brd ff:ff:ff:ff:ff:ff permaddr ba:22:a7:7a:15:94
3: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq master br-lan state UP mode DEFAULT group default qlen 1000
    link/ether 32:94:fd:4a:da:68 brd ff:ff:ff:ff:ff:ff permaddr b6:22:a7:7a:15:94
4: ip6tnl0@NONE: <NOARP> mtu 1452 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/tunnel6 :: brd :: permaddr aa6a:339c:25ee::
5: tunl0@NONE: <NOARP> mtu 1480 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/ipip 0.0.0.0 brd 0.0.0.0
6: sit0@NONE: <NOARP> mtu 1480 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/sit 0.0.0.0 brd 0.0.0.0
7: ip_vti0@NONE: <NOARP> mtu 1480 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/ipip 0.0.0.0 brd 0.0.0.0
8: ip6_vti0@NONE: <NOARP> mtu 1332 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/tunnel6 :: brd :: permaddr 46be:a0fa:627::
9: dummy0: <BROADCAST,NOARP> mtu 1500 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/ether 92:e6:48:68:dd:68 brd ff:ff:ff:ff:ff:ff
10: eth2: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq master br-lan state DOWN mode DEFAULT group default qlen 1000
    link/ether 32:94:fd:4a:da:6b brd ff:ff:ff:ff:ff:ff permaddr 22:dd:31:45:ef:43
11: eth3: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq master br-lan state UP mode DEFAULT group default qlen 1000
    link/ether 32:94:fd:4a:da:6d brd ff:ff:ff:ff:ff:ff permaddr 5a:62:2b:9b:09:ed
12: gre0@NONE: <NOARP> mtu 1476 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/gre 0.0.0.0 brd 0.0.0.0
13: gretap0@NONE: <BROADCAST,MULTICAST> mtu 1462 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
14: erspan0@NONE: <BROADCAST,MULTICAST> mtu 1450 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
15: ip6gre0@NONE: <NOARP> mtu 1448 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/gre6 :: brd :: permaddr f697:bf9e:1dc5::
16: siit0: <BROADCAST,MULTICAST,NOARP> mtu 1500 qdisc noop state DOWN mode DEFAULT group default qlen 10
    link/ether 8e:60:fa:77:90:02 brd ff:ff:ff:ff:ff:ff
17: teql0: <NOARP> mtu 1500 qdisc noop state DOWN mode DEFAULT group default qlen 100
    link/void 
18: br-lan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP mode DEFAULT group default qlen 1000
    link/ether 32:94:fd:4a:da:69 brd ff:ff:ff:ff:ff:ff
19: docker0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN mode DEFAULT group default 
    link/ether 02:42:d5:21:72:b4 brd ff:ff:ff:ff:ff:ff
20: pppoe-wan: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1492 qdisc fq_codel state UNKNOWN mode DEFAULT group default qlen 3
    link/ppp `

另外，刚才说的wan_6在没更改为eth1之前是没有的，原来只有四个分别是docker、lan、wan、wan6。wan_6的协议是: 虚拟动态接口 (DHCPv6 客户端)，而且不能重启或停止。

jjm2473, 2023-10-23:
wan_6是pppoe拨号成功以后申请ipv6用的，你之前都拨号不了，当然没有wan_6

mobyoung, 2023-10-23:
> 总之，不要再管wan_6接口，你要是不想看到，就禁用pppoe的ipv6申请

ok。
那后面的版本是否会吧wan口这个情况更改回去？

jjm2473, 2023-10-23:
要先改成eth0作为wan口，然后再生成mac地址

jjm2473, 2023-10-23:
生成mac地址以后，保存应用。然后执行`ip link`看看

mobyoung, 2023-10-23:
> 要先改成eth0作为wan口，然后再生成mac地址
> 生成mac地址以后，保存应用。然后执行`ip link`看看
因为担心已经安装的服务和docker要重装，所以还没试过恢复出厂设置。
按照你的说明，再操作了一遍，情况和刚才没有区别，还是一样。执行命令的结果如下，eth1没插网线：
-------------------------------------------------------
 iStoreOS 22.03.5, 2023102014
 -------------------------------------------------------
root@iStoreOS:~# ip link
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth1: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq master br-lan state DOWN mode DEFAULT group default qlen 1000
    link/ether ba:22:a7:7a:15:94 brd ff:ff:ff:ff:ff:ff
3: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP mode DEFAULT group default qlen 1000
    link/ether 7c:10:c9:12:05:67 brd ff:ff:ff:ff:ff:ff permaddr b6:22:a7:7a:15:94
4: ip6tnl0@NONE: <NOARP> mtu 1452 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/tunnel6 :: brd :: permaddr aa6a:339c:25ee::
5: tunl0@NONE: <NOARP> mtu 1480 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/ipip 0.0.0.0 brd 0.0.0.0
6: sit0@NONE: <NOARP> mtu 1480 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/sit 0.0.0.0 brd 0.0.0.0
7: ip_vti0@NONE: <NOARP> mtu 1480 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/ipip 0.0.0.0 brd 0.0.0.0
8: ip6_vti0@NONE: <NOARP> mtu 1332 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/tunnel6 :: brd :: permaddr 46be:a0fa:627::
9: dummy0: <BROADCAST,NOARP> mtu 1500 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/ether 92:e6:48:68:dd:68 brd ff:ff:ff:ff:ff:ff
10: eth2: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq master br-lan state DOWN mode DEFAULT group default qlen 1000
    link/ether 32:94:fd:4a:da:6b brd ff:ff:ff:ff:ff:ff permaddr 22:dd:31:45:ef:43
11: eth3: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq master br-lan state UP mode DEFAULT group default qlen 1000
    link/ether 32:94:fd:4a:da:6d brd ff:ff:ff:ff:ff:ff permaddr 5a:62:2b:9b:09:ed
12: gre0@NONE: <NOARP> mtu 1476 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/gre 0.0.0.0 brd 0.0.0.0
13: gretap0@NONE: <BROADCAST,MULTICAST> mtu 1462 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
14: erspan0@NONE: <BROADCAST,MULTICAST> mtu 1450 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
15: ip6gre0@NONE: <NOARP> mtu 1448 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/gre6 :: brd :: permaddr f697:bf9e:1dc5::
16: siit0: <BROADCAST,MULTICAST,NOARP> mtu 1500 qdisc noop state DOWN mode DEFAULT group default qlen 10
    link/ether 8e:60:fa:77:90:02 brd ff:ff:ff:ff:ff:ff
17: teql0: <NOARP> mtu 1500 qdisc noop state DOWN mode DEFAULT group default qlen 100
    link/void 
18: br-lan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP mode DEFAULT group default qlen 1000
    link/ether 32:94:fd:4a:da:69 brd ff:ff:ff:ff:ff:ff
19: docker0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN mode DEFAULT group default 
    link/ether 02:42:d5:21:72:b4 brd ff:ff:ff:ff:ff:ff

jjm2473, 2023-10-24:
我想起一个问题，你的固件是不是21版本升级过来的，21版本的uboot缺少一些功能，固件升级的时候又不会升级uboot的，所以可能需要单独升级下uboot才行。我搞个升级脚本给你试试。

mobyoung, 2023-10-24:
> 我想起一个问题，你的固件是不是21版本升级过来的，21版本的uboot缺少一些功能，固件升级的时候又不会升级uboot的，所以可能需要单独升级下uboot才行。我搞个升级脚本给你试试。

是的，需要提供邮箱吗？

jjm2473, 2023-10-24:
[rk3568-uboot-flash.tgz](https://github.com/istoreos/istoreos/files/13110787/rk3568-uboot-flash.tgz)

下载这个文件，注意不要解压，然后把扩展名改成`.run`，然后在istore的手段安装那边上传安装，会自动开始刷uboot，窗口变成绿色就成功了，然后重启下系统

mobyoung, 2023-10-24:
晚上回来上传安装了，安装成功，试过重启和关机过一会再开，问题还是没解决。

jjm2473, 2023-10-24:
你的eth0只是不能拨号，却可以作为lan口使用？

mobyoung, 2023-10-24:
是的，我刚刚确认了一下，eth0可以正常作为lan口连接其他设备使用。

jjm2473, 2023-10-24:
试试执行这个命令`ethtool -K eth0 tx off`，不要重启，然后测试能不能拨号

jjm2473, 2023-10-24:
你把`/etc/config/network`的内容复制到这里，注意删掉拨号的账号信息。

mobyoung, 2023-10-24:
刚刚尝试执行命令后设置eth0为wan，拨号失败。
network的内容如下：

config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fdef:8592:67f7::/48'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'eth2'
	list ports 'eth3'
	list ports 'eth1'

config device
	option name 'eth1'

config device
	option name 'eth2'
	option macaddr '32:94:fd:4a:da:6b'

config device
	option name 'eth3'
	option macaddr '32:94:fd:4a:da:6d'

config interface 'lan'
	option device 'br-lan'
	option proto 'static'
	option netmask '255.255.255.0'
	option ip6assign '60'
	option defaultroute '0'
	option ipaddr '192.168.1.1'

config device
	option name 'eth0'
	option macaddr '7C:10:C9:12:05:67'

config interface 'wan'
	option proto 'pppoe'
	option username '账号'
	option password '密码'
	option ipv6 'auto'
	option device 'eth0'

config interface 'wan6'
	option proto 'dhcpv6'
	option reqaddress 'try'
	option reqprefix 'auto'
	option device 'eth0'

config interface 'docker'
	option device 'docker0'
	option proto 'none'
	option auto '0'

jjm2473, 2023-10-24:
`/etc/config/firewall`的内容也贴出来看看

mobyoung, 2023-10-24:
fiewall内容：

config defaults
	option input 'ACCEPT'
	option output 'ACCEPT'
	option forward 'REJECT'
	option fullcone '1'
	option synflood_protect '1'

config zone
	option name 'lan'
	list network 'lan'
	option input 'ACCEPT'
	option output 'ACCEPT'
	option forward 'ACCEPT'
	option auto_helper '0'

config zone
	option name 'wan'
	list network 'wan'
	list network 'wan6'
	option output 'ACCEPT'
	option forward 'REJECT'
	option masq '1'
	option mtu_fix '1'
	option input 'DROP'

config forwarding
	option src 'lan'
	option dest 'wan'

config rule
	option name 'Allow-DHCP-Renew'
	option src 'wan'
	option proto 'udp'
	option dest_port '68'
	option target 'ACCEPT'
	option family 'ipv4'

config rule
	option name 'Allow-Ping'
	option src 'wan'
	option proto 'icmp'
	option icmp_type 'echo-request'
	option family 'ipv4'
	option target 'ACCEPT'

config rule
	option name 'Allow-IGMP'
	option src 'wan'
	option proto 'igmp'
	option family 'ipv4'
	option target 'ACCEPT'

config rule
	option name 'Allow-DHCPv6'
	option src 'wan'
	option proto 'udp'
	option src_ip 'fc00::/6'
	option dest_ip 'fc00::/6'
	option dest_port '546'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-MLD'
	option src 'wan'
	option proto 'icmp'
	option src_ip 'fe80::/10'
	list icmp_type '130/0'
[... 196 lines cut ...]
	option dest 'docker'

config redirect
	option dest 'lan'
	option target 'DNAT'
	option name 'node-red'
	option src 'wan'
	option src_dport '1880'
	option dest_ip '192.168.1.1'
	option dest_port '1880'
	option enabled '0'

config redirect
	option dest 'lan'
	option target 'DNAT'
	option name 'Vaultwarden'
	option src 'wan'
	option dest_ip '192.168.1.10'
	option dest_port '8508'
	option src_dport '8507'

config include 'gowebdav'
	option type 'script'
	option path '/var/etc/gowebdav.include'
	option reload '1'

jjm2473, 2023-10-24:
配置文件没看出什么问题。你有没有试过关掉ssr插件？还有，你有没有试过修复系统组件？（ https://github.com/istoreos/istoreos/wiki/FAQ 第二点）

mobyoung, 2023-10-24:
刚刚尝试关掉ssr插件，修复系统组件。还是不行，头大。。。

jjm2473, 2023-10-24:
修复系统组件后需要重启下

mobyoung, 2023-10-24:
是的，已经重启过了。

jjm2473, 2023-10-24:
那就不知道啥原因了，目前也没其他人反馈

mobyoung, 2023-10-25:
好吧，只能暂时先用着