istoreos/istoreos #1052: 无法在Dell PowerEdge R620中使用SAS直通硬盘

ArtomYuan, 2023-10-14:
(1) 具体问题
无法在Dell Inc. PowerEdge R620中使用IT模式的直通硬盘

(2) 路由器型号和固件版本
Dell Inc. PowerEdge R620
iStoreOS 22.03.5 2023092815

jjm2473, 2023-10-21:
不太了解SAS，执行`lspci -nnk`把结果贴出来看看

ArtomYuan, 2023-10-22:
> 不太了解SAS，执行`lspci -nnk`把结果贴出来看看

root@iStoreOS:~# lspci -nnk
00:00.0 Host bridge [0600]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 DMI2 [8086:0e00] (rev 04)
        Subsystem: Dell Device [1028:04ce]
lspci: Unable to load libkmod resources: error -12
00:01.0 PCI bridge [0604]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 PCI Express Root Port 1a [8086:0e02] (rev 04)        Subsystem: Dell Device [1028:04ce]
        Kernel driver in use: pcieport
00:01.1 PCI bridge [0604]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 PCI Express Root Port 1b [8086:0e03] (rev 04)        Subsystem: Dell Device [1028:04ce]
        Kernel driver in use: pcieport
00:02.0 PCI bridge [0604]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 PCI Express Root Port 2a [8086:0e04] (rev 04)        Subsystem: Dell Device [1028:04ce]
        Kernel driver in use: pcieport
00:02.2 PCI bridge [0604]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 PCI Express Root Port 2c [8086:0e06] (rev 04)        Subsystem: Dell Device [1028:04ce]
        Kernel driver in use: pcieport
00:03.0 PCI bridge [0604]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 PCI Express Root Port 3a [8086:0e08] (rev 04)        DeviceName: SLOT 3
        Subsystem: Dell Device [1028:04ce]
        Kernel driver in use: pcieport
00:04.0 System peripheral [0880]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 Crystal Beach DMA Channel 0 [8086:0e20] (rev 04)
        Subsystem: Dell Device [1028:04ce]
00:04.1 System peripheral [0880]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 Crystal Beach DMA Channel 1 [8086:0e21] (rev 04)
        Subsystem: Dell Device [1028:04ce]
00:04.2 System peripheral [0880]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 Crystal Beach DMA Channel 2 [8086:0e22] (rev 04)
        Subsystem: Dell Device [1028:04ce]
00:04.3 System peripheral [0880]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 Crystal Beach DMA Channel 3 [8086:0e23] (rev 04)
        Subsystem: Dell Device [1028:04ce]
00:04.4 System peripheral [0880]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 Crystal Beach DMA Channel 4 [8086:0e24] (rev 04)
        Subsystem: Dell Device [1028:04ce]
00:04.5 System peripheral [0880]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 Crystal Beach DMA Channel 5 [8086:0e25] (rev 04)
        Subsystem: Dell Device [1028:04ce]
00:04.6 System peripheral [0880]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 Crystal Beach DMA Channel 6 [8086:0e26] (rev 04)
        Subsystem: Dell Device [1028:04ce]
00:04.7 System peripheral [0880]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 Crystal Beach DMA Channel 7 [8086:0e27] (rev 04)
        Subsystem: Dell Device [1028:04ce]
00:05.0 System peripheral [0880]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 VTd/Memory Map/Misc [8086:0e28] (rev 04)
        Subsystem: Dell Device [1028:04ce]
00:05.2 System peripheral [0880]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 IIO RAS [8086:0e2a] (rev 04)
        Subsystem: Dell Device [1028:04ce]
00:11.0 PCI bridge [0604]: Intel Corporation C600/X79 series chipset PCI Express Virtual Root Port [8086:1d3e] (rev 05)
        Subsystem: Dell Device [1028:04ce]
        Kernel driver in use: pcieport
00:16.0 Communication controller [0780]: Intel Corporation C600/X79 series chipset MEI Controller #1 [8086:1d3a] (rev 05)
        Subsystem: Dell Device [1028:04ce]
00:16.1 Communication controller [0780]: Intel Corporation C600/X79 series chipset MEI Controller #2 [8086:1d3b] (rev 05)
        Subsystem: Dell Device [1028:04ce]
00:1a.0 USB controller [0c03]: Intel Corporation C600/X79 series chipset USB2 Enhanced Host Controller #2 [8086:1d2d] (rev 05)
        Subsystem: Dell Device [1028:04ce]
        Kernel driver in use: ehci-pci
00:1c.0 PCI bridge [0604]: Intel Corporation C600/X79 series chipset PCI Express Root Port 1 [8086:1d10] (rev b5)
        Subsystem: Dell Device [1028:04ce]
        Kernel driver in use: pcieport
00:1c.7 PCI bridge [0604]: Intel Corporation C600/X79 series chipset PCI Express Root Port 8 [8086:1d1e] (rev b5)
        Subsystem: Dell Device [1028:04ce]
        Kernel driver in use: pcieport
00:1d.0 USB controller [0c03]: Intel Corporation C600/X79 series chipset USB2 Enhanced Host Controller #1 [8086:1d26] (rev 05)
        Subsystem: Dell Device [1028:04ce]
        Kernel driver in use: ehci-pci
00:1e.0 PCI bridge [0604]: Intel Corporation 82801 PCI Bridge [8086:244e] (rev a5)
        Subsystem: Dell Device [1028:04ce]
00:1f.0 ISA bridge [0601]: Intel Corporation C600/X79 series chipset LPC Controller [8086:1d41] (rev 05)
        Subsystem: Dell Device [1028:04ce]
        Kernel driver in use: lpc_ich
00:1f.2 SATA controller [0106]: Intel Corporation C600/X79 series chipset 6-Port SATA AHCI Controller [8086:1d02] (rev 05)
        Subsystem: Dell Device [1028:04ce]
        Kernel driver in use: ahci
01:00.0 Ethernet controller [0200]: Broadcom Inc. and subsidiaries NetXtreme BCM5720 Gigabit Ethernet PCIe [14e4:165f]
        DeviceName: NIC1
        Subsystem: Dell Device [1028:1f5b]
        Kernel driver in use: tg3
01:00.1 Ethernet controller [0200]: Broadcom Inc. and subsidiaries NetXtreme BCM5720 Gigabit Ethernet PCIe [14e4:165f]
        DeviceName: NIC2
[... 270 lines cut ...]
        Subsystem: Intel Corporation Device [8086:0000]
7f:1d.3 System peripheral [0880]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 Integrated Memory Controller 1 Channel Target Address Decoder Registers [808... (rev 04)
[... cut ...]
        Kernel driver in use: ivbep_uncore
7f:1e.5 System peripheral [0880]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 Integrated Memory Controller 0 Channel 0-3 Thermal Control 3 [8086:0ef5] (rev 04)
        Subsystem: Intel Corporation Device [8086:0000]
        Kernel driver in use: ivbep_uncore
7f:1e.7 System peripheral [0880]: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 Integrated Memory Controller 0 Channel 0-3 ERROR Registers 3 [8086:0ef7] (rev 04)
        Subsystem: Intel Corporation Device [8086:0000]

ArtomYuan, 2023-10-22:
你好，这个就是输出的结果

jjm2473, 2023-10-23:
@ArtomYuan 应该可以直通硬盘，而不直通sas控制器吧

jjm2473, 2023-10-23:
内核倒是有这个SAS控制器的驱动，只是openwrt都没放在软件包列表里，应该是很少用得上的。我可以弄个驱动给你

ArtomYuan, 2023-10-23:
> 内核倒是有这个SAS控制器的驱动，只是openwrt都没放在软件包列表里，应该是很少用得上的。我可以弄个驱动给你

可以分享一下吗，老哥，我搜了好久的Google和百度，没找到什么解决方法

jjm2473, 2023-10-23:
我看了下，这个驱动依赖内核的功能，所以不能单独安装驱动，我等下给你个固件。

jjm2473, 2023-10-23:
这是集成了SAS2308驱动的固件 https://we.tl/t-sjy7QUnixv ，注意不要在线升级，等下个版本发布才能在线升级。

ArtomYuan, 2023-10-23:
> 这是集成了SAS2308驱动的固件 https://we.tl/t-sjy7QUnixv ，注意不要在线升级，等下个版本发布才能在线升级。

多谢老哥，我的直通硬盘识别出来了，但是你说的不要在线升级是不是指不要用“左侧栏 —— 系统 —— 在线升级”这个，那请问我应该怎么升级呢

jjm2473, 2023-10-24:
@ArtomYuan 我是说你刷了这个固件以后不要在线升级，而且刷固件也不是在线升级，要在“系统”-“备份与升级”里面刷固件。

jjm2473, 2023-10-24:
@ArtomYuan 等到下个版本的固件发布以后你才能升级。如果你现在升级，那就是升级到线上正式版本，线上的正式版本是上周发布的，你说会有什么效果？

jjm2473, 2023-10-24:
都说了等下个版本发布之后你才可以升级，为什么这么简单的东西都要说几遍？！！

closed 2023-10-24 by jjm2473

jjm2473, 2023-10-24:
iStoreOS 一般周五才发布，你现在升级不就升级到上周五的版本了吗，上周五的版本有这个驱动吗？
如果这周五有发布固件，那你周五以后看到有更新，也就是10月27日的版本，那你就可以在线升级了。

这么简单的事，还要解释多少遍？真TM对牛弹琴

ArtomYuan, 2023-10-24:
> iStoreOS 一般周五才发布，你现在升级不就升级到上周五的版本了吗，上周五的版本有这个驱动吗？ 如果这周五有发布固件，那你周五以后看到有更新，也就是10月27日的版本，那你就可以在线升级了。
> 
> 这么简单的事，还要解释多少遍？真TM对牛弹琴

哦哦，现在明白了，我不清楚iStoreOS是周五更新，还以为现在那个在线更新是新版的，没想到是上周五的版本，多谢老哥了