istoreos/istoreos #1053: 升级系统到22.03.5 2023092815后，同一网络下的docker容器直接无法ping通

fujie-xiyou, 2023-10-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：升级系统后 docker容器之间无法ping通 确认是在同一个docker网络下 ，尝试重启系统无法解决

### (2) 路由器型号和固件版本

A：x86_64  固件版本 iStoreOS 22.03.5 2023092815

### (3) 详细日志和/或截图

A：
[image]

[image]

fujie-xiyou, 2023-10-15:
系统日志里面也有很多docker相关的 
[image]

fujie-xiyou, 2023-10-15:
[image]
对应的docker网络配置

closed 2023-10-19 by jjm2473

fujie-xiyou, 2023-10-19:
感谢！全局设置里面允许转发之后确实可以了，但是比较奇怪为啥升级系统会导致这个问题呀  升级之前确认是好的

jjm2473, 2023-10-20:
因为以前21版本的固件还是使用docker自带的防火墙，docker会打开自己创建的桥的内部的转发。但是docker自带防火墙功能是不安全的，不遵守openwrt系统防火墙的规则，会把所有容器的服务开放到公网，所以22版本以后禁用了docker的防火墙，转为使用openwrt的防火墙，但是只有docker0才加入系统防火墙，也就是默认的docker网络是可以桥转发的，其他自定义的docker网络需要自己添加到防火墙列表，或者全局允许转发。

不过今天发布的固件已经把桥过滤也关掉了，之后就算不允许全局转发也没问题，详情看更新日志。

fujie-xiyou, 2023-10-20:
> 因为以前21版本的固件还是使用docker自带的防火墙，docker会打开自己创建的桥的内部的转发。但是docker自带防火墙功能是不安全的，不遵守openwrt系统防火墙的规则，会把所有容器的服务开放到公网，所以22版本以后禁用了docker的防火墙，转为使用openwrt的防火墙，但是只有docker0才加入系统防火墙，也就是默认的docker网络是可以桥转发的，其他自定义的docker网络需要自己添加到防火墙列表，或者全局允许转发。
> 
> 不过今天发布的固件已经把桥过滤也关掉了，之后就算不允许全局转发也没问题，详情看更新日志。

明白了！多谢解答！