istoreos/istoreos #1057: 固件缺少依赖的软件包 libopenssl3，导致很多插件无法安装

CharlieVSAngel, 2023-10-15:
很多插件无法安装，提示：
依赖的软件包 libopenssl3 在所有仓库都未提供。我如何安装libopenssl3

jjm2473, 2023-10-21:
libopenssl3 是23.05以上才有的。这里可以下载 https://downloads.openwrt.org/releases/23.05.0/packages ，选择自己的架构，然后点 base ，里面就可以搜索 libopenssl3。例如 aarch64_cortex-a53 架构的相应 ipk 文件是 https://downloads.openwrt.org/releases/23.05.0/packages/aarch64_cortex-a53/base/libopenssl3_3.0.11-1_aarch64_cortex-a53.ipk

jjm2473, 2023-10-23:
你是在安装什么插件的时候出现这个错误？

bcseputetto, 2023-10-31:
> 你是在安装什么插件的时候出现这个错误？

最近passwall的SDK更换到了 23.05 RC3
```
root@iStoreOS:~# opkg list-installed |grep shadowsocksr
shadowsocksr-libev-ssr-local - 2.5.6-9
shadowsocksr-libev-ssr-redir - 2.5.6-9
shadowsocksr-libev-ssr-server - 2.5.6-9
root@iStoreOS:~# opkg list-installed |grep trojan-plus
trojan-plus - 10.0.3-2
root@iStoreOS:~# opkg list-installed |grep libopenssl
libopenssl-conf - 1.1.1t-3
libopenssl1.1 - 1.1.1t-3
libopenssl3 - 3.0.11-1
```
shadowsocksr-libev 三件套和 trojan-plus 的libopenssl依赖也变成了 3.0，所以很多小白用户没办法从默认22.03.5软件源获取到，就束手无策了。
passwall那边也有反馈libopenssl3.0带来的新问题 [openwrt-passwall issues#2827](https://github.com/xiaorouji/openwrt-passwall/issues/2827)
不过该问题已被关闭，对于还在使用openssl3.0遗弃的一些不安全算法的机场的ssr节点来说，可能不太友好。

jjm2473, 2023-10-31:
你用的ipk是哪里下载的，你安装libopenssl3以后能跑起来吗，联网是不是正常？

如果是小白用户就先用第三方编译的离线包吧，至少是使用老SDK编译的，在22.03上可以使用 https://github.com/AUK9527/Are-u-ok

bcseputetto, 2023-10-31:
> 你用的ipk是哪里下载的，你安装libopenssl3以后能跑起来吗，联网是不是正常？
> 
> 如果是小白用户就先用第三方编译的离线包吧，至少是使用老SDK编译的，在22.03上可以使用 https://github.com/AUK9527/Are-u-ok

哈哈，我是没问题的，我都正常使用，我只是说一下他们最近安装什么插件一直被 libopenssl3给卡住。这个issues其实可以关了

jjm2473, 2023-11-01:
23.05.0 的兼容性只会更差，上次移植23.05-RC2的时候，很多luci的lua的api都不兼容了，对于很多还在使用luci1.x的插件来说更容易出问题，还是等到至少23.05.1以后再考虑跟进。

jjm2473, 2023-11-01:
如果是新加密算法的ssr，应该可以使用xray内核，你可以试试xray有没有问题

cateatlemon, 2023-11-11:
> 很多插件无法安装，提示： 所有仓库中的依赖库 libopenssl3 都未提供。我如何安装 libopenssl3

可以了、果然是缺少依赖！！！谢谢解答

cateatlemon, 2023-11-11:
> libopenssl3 是23.05以上保留的。这里可以下载https://downloads.openwrt.org/releases/23.05.0/packages，选择自己的架构，然后点base ，里面就可以搜索libopenssl3。例如aarch64_cortex-a53 架构的架构相应的ipk文件为https://downloads.openwrt.org/releases/23.05.0/packages/aarch64_cortex-a53/base/libopenssl3_3.0.11-1_aarch64_cortex-a53.ipk

可以了、果然是缺少依赖！谢谢解答

UnsleepingCat, 2023-12-10:
这个问题非常恼人！三方源因为缺少依赖，疯狂报错255

bcseputetto, 2023-12-10:
> 这个问题非常恼人！三方源因为缺少依赖，疯狂报错255

https://github.com/bcseputetto/Are-u-ok/
可以到我仓库release 下载用22.03 SDK编译的新版插件，不需要libopenssl3，我也不推荐去用第三方源

jjm2473, 2023-12-15:
三方源自己注意看看人家编译用的sdk版本，新版本sdk编译的ipk可能在老版本固件没法用

JIUJIUs, 2023-12-19:
> > 这个问题非常恼人！三方源因为缺少依赖，疯狂报错255
> 
> https://github.com/bcseputetto/Are-u-ok/ 可以到我仓库release 下载用22.03 SDK编译的新版插件，不需要libopenssl3，我也不推荐去用第三方源

谢谢老哥，用你编译的成功跑起来了～

前面我自己找依赖安装，但是 PassWall 里面节点老是超时，实在搞不定了，多亏看到这里，看到你🙏

jingmian, 2024-06-30:
> > 这个问题非常恼人！三方源因为缺少依赖，疯狂报错255
> 
> https://github.com/bcseputetto/Are-u-ok/ 可以到我仓库release 下载用22.03 SDK编译的新版插件，不需要libopenssl3，我也不推荐去用第三方源

这个不错，现在还是可以用