istoreos/istoreos #1058: 226网卡直通识别不到

codeyFun, 2023-10-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ X ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：I-226V在EXSI8下直通给istoreOS识别不到，试了其他的openwrt固件能够识别。

### (2) 路由器型号和固件版本

A：x86软路由

设备型号
VMware, Inc. VMware20,1
固件版本
iStoreOS 22.03.5 2023092815
内核版本
5.10.176

### (3) 详细日志和/或截图
直通了两个网口，一个给另外一个openwrt，一个给istoreOS
A：另外一个openwrt配置
[image]

[image]

B：istoreOS配置
[image]
[image]

codeyFun, 2023-10-16:
hello,请问有谁知道是啥原因吗？

jjm2473, 2023-10-21:
openwrt官方固件能识别到吗？其他固件使用的驱动是什么，执行`lspci nnk`看看结果

jjm2473, 2023-10-25:
I226-V，I226-LM，I226-IT是支持的，I226-LMvP估计要新内核或者打补丁才能支持

jjm2473, 2023-10-25:
试试这个固件 https://we.tl/t-WTpvQuT334

jjm2473, 2023-10-26:
测试了没？有没有用

codeyFun, 2023-10-27:
> 测试了没？有没有用

> 试试这个固件 https://we.tl/t-WTpvQuT334

下载连接失效了

codeyFun, 2023-10-27:
> 试试这个固件 https://we.tl/t-WTpvQuT334

这个固件可以识别到

closed 2023-10-27 by jjm2473