istoreos/istoreos #1062: 最新版X86固件降速

qsos, 2023-10-20:
最新版X86固件降速 500M宽带只能跑到350M

jjm2473, 2023-10-23:
执行`lspci -nnk`看看，还有你是怎么测速的，有没有经过wifi？

darling1111222, 2023-10-23:
我也发现类似情况 更新最新固件后 passwall中ping延迟从150升到300+ 刷回21.02.3-2023042111的版本就又回复  弄得不敢升级了

darling1111222, 2023-10-23:
网上找的教程提供的就是这个版本 没办法 升级到最新版本就降速

jjm2473, 2023-10-23:
@darling1111222 你的设备是什么？使用什么网卡？

jjm2473, 2023-10-24:
我昨天用N100机器测试拨号，是可以跑满千兆的

darling1111222, 2023-10-24:
i266网卡

jjm2473, 2023-10-24:
@darling1111222 你说的是i226吧，我测试的就是i226

darling1111222, 2023-10-24:
是的 我打错了 怎么看我的软路由是否正确驱动了呢

darling1111222, 2023-10-24:
istore os 在哪里更新驱动网卡呢 我小白 请教

jjm2473, 2023-10-24:
@darling1111222 `lspci -nnk`，看看相应的网卡后面有没有`Kernel driver in use`。

更新不了网卡驱动，除非你自己编译驱动，就算你编译，你也要找到驱动源码，目前用的是igc驱动，linux内核自带的。

jjm2473, 2023-10-24:
@darling1111222  你可以试试打开“数据包引导”，可能会有帮助，不过我这里没开都能跑满千兆了。 

[image]

darling1111222, 2023-10-24:
可是升级最新固件后链接
谷歌ping真的翻倍。。。

jjm2473, 2023-10-24:
靠，你拿谷歌ping来比较？谷歌不是要翻墙的吗，你测试网速为什么还引入更多的变量？还有，ipv6有没有关掉？

darling1111222, 2023-10-24:
可是对我有价值的就是谷歌ping

darling1111222, 2023-10-24:
ipv6在哪关

darling1111222, 2023-10-24:
[image]
是这么

jjm2473, 2023-10-24:
@darling1111222 翻墙的性能问题不要在这讨论，除非你能排除节点和翻墙插件的问题。

qsos, 2023-10-25:
> 我也发现类似情况 更新最新固件后 软件中ping数值从150升到300+ 刷回21.02.3-2023042111的版本就又恢复 弄得不敢升级了

我也降级了，降级之后就没这个问题了

darling1111222, 2023-10-25:
看来是通病 希望得到解决

jjm2473, 2023-10-25:
我的机器上无法复现

virtually321, 2023-10-30:
果然是网卡驱动问题