istoreos/istoreos #1066: 最近liangbnx86硬盘休眠挂了

sunqin0310, 2023-10-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志和/或截图

A：

jjm2473, 2023-10-21:
istore里面卸载重装这个插件试试

jjm2473, 2023-10-23:
已确认插件没问题。可能是因为你是在老版本的固件安装的插件，现在不兼容了，从istore卸载重装就行了。

closed 2023-10-23 by jjm2473

sunqin0310, 2023-10-23:
[image]
问题依旧，我装的就是目前最新的x86固件，新开的虚拟机，测试的，无论是软件包下载重装，还是istor下载重装，均是这个提示，不存在老版本的问题，我也不清楚是为什么，我只是找固件的时候测试出来的，希望对你们有帮助
[image]

jjm2473, 2023-10-23:
你用的是什么浏览器？有没有试过chrome？

sunqin0310, 2023-10-24:
你们这可真神奇，我主用edge浏览器，刚下载了个谷歌浏览器，进去重新安装还是不行，然后我用手机的uc浏览器进去重新下载试试，天，他居然好了，能进去了，回来又用edge和谷歌也能进去了，用edge重新安装也可以进去了，好神奇，用uc进去重新下载前，提示的是已损坏，然后卸载重新安装就可以了，

jjm2473, 2023-10-24:
可能是某个硬盘有问题，看代码是硬盘信息某个字段为null才会导致这个错误。