istoreos/istoreos #1074: docker容器设置内存等资源信息的问题

jymusic0663, 2023-10-24:
反馈bug/问题

## docker容器设置内存等资源信息的问题

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

- istoreOS后台docker里，设置容器内存信息时，出现下面的错误提示，即表单的数值提交到后台，后台转换出错。表单时写的`2147483648`，但后台判断成了浮点类型的，写`2g`也无法识别

- 出现频率：必现

### (2) 路由器型号和固件版本

- 设备型号：x86 Generic（x86迷你主机刷的istoreOS）
- 固件版本：iStoreOS 22.03.5 2023102014
- 内核版本：5.10.176
- 浏览器：Edge浏览器

### (3) 详细日志和/或截图

``` log
Containers: update d7abb...code:500 json: cannot unmarshal number 2147483648.0 into Go struct field UpdateConfig.Memory of type int64
```

jjm2473, 2023-10-25:
看起来是openwrt那边修改lua导致的。

jjm2473, 2023-10-25:
哦，我知道原因了，lua的int是32位有符号整数，无法表示2G以上的值，会被自动转换成浮点数，于是有了.0。
我看看能不能使用M作为后缀提交

jjm2473, 2023-10-25:
看起来dockerd没提供字符串的接口，也就是没法通过变换单位防止数值溢出，只能使用整数。

提交bug到openwrt了 https://github.com/openwrt/luci/issues/6647

jjm2473, 2023-10-27:
已修复。
并且增加 MemorySwap 参数，单独使用 Memory 参数可能无法成功，Docker 会提示 Memory 需要小于 MemorySwap。如果不需要限制 MemorySwap ，填 `-1` 即可

closed 2023-10-27 by jjm2473