istoreos/istoreos #1082: passwall连接上，但无法上Google，油管

wssfxtx, 2023-10-27:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [✓ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：passwall连接上节点，passwall页面点击google显示ping值，ip111显示国外IP为代理IP，goole IP不现实,无法上Google，油管

### (2) 路由器型号和固件版本

A：设备型号
FastRhino R66s
固件版本
iStoreOS 22.03.5 2023102711

### (3) 详细日志和/或截图

A：[image]
[image]

jjm2473, 2023-10-27:
可能你用了ipv6，但是节点又不支持。关掉电脑的ipv6试试，或者关掉软路由的ipv6解析。

wssfxtx, 2023-10-27:
您的邮件已经收到，我将尽快回复，谢谢！

jjm2473, 2023-10-29:
你把电脑的ipv6关掉试试，然后出现连wifi，并且在passwall的基本设置的dns栏，把dns缓存清空

closed 2023-10-29 by jjm2473