istoreos/istoreos #1085: 修改docker目录后 docker 报错 failed to register layer: operation not supported

blankhang, 2023-10-28:
### (1) 具体问题

A：
系统盘是1TB的 M2 装完后 我把剩下的空间给扩容了 然后就想把在其他盘的docker给迁移回来
改了目录后 重新拉镜像报错

### (2) 路由器型号和固件版本

A：
INTEL N100  iStoreOS 22.03.5 2023102711

### (3) 详细日志和/或截图
[image]
[image]
A：

blankhang, 2023-10-28:
ps 迁移docker 能不能不要限制/ 根目录。。。

jjm2473, 2023-10-29:
本来就不能放在根目录啊，就算你扩容了，也要放在/overlay/upper/opt/docker里面。你遇到的这个报错就是因为放在根目录。

blankhang, 2023-10-30:
为什么非要限制我放根目录？ 做个提示不建议放根目录就可以了吗，实在不能理解 我自己1TB的盘还跑不起docker 我服务器的系统40G都跑的欢

closed 2023-10-30 by blankhang

jjm2473, 2023-10-30:
还有，扩容最好是创建个新分区，这样以后就算不保留配置升级固件，只要再分区到同一个位置，那数据就还在。还有，硬盘的挂载点最好都设置到/mnt里面，避免硬盘没挂载的情况下数据被写到根目录，导致系统配置文件损坏。

jjm2473, 2023-10-30:
还有docker的数据只能放在ext4等linux的文件系统里，不能放在ntfs或fat之类的文件系统