istoreos/istoreos #1088: 软路由下游设备无法获取ip地址

cocollin23, 2023-11-01:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
软路由下游桥接无线路由，通过连接无线路由器上网，软路由重启后无法获取ip地址，造成无法上网

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：之前一直是用dnsmasq进行下游ip地址动态分配，软路由自动重启后，重新连接无线网无法获取ip地址，只能通过手动设置静态地址才可以上网；目前的解决办法是，开启adguardhome的dhcp服务器后，下游设备可以获取到IP地址！一旦关闭adguardhome，网络设备就无法获取ip。

### (2) 路由器型号和固件版本

A：istoreos-22.03.5-2023102711-x86-64-squashfs-combined.img.gz

cocollin23, 2023-11-01:
[image]

jjm2473, 2023-11-02:
adg有bug，而且还修改了dnsmasq的设置，adg没启动的时候dnsmasq也不能正常工作。

jjm2473, 2023-11-02:
试试开关一次adg，看看还有没有问题

closed 2023-11-02 by jjm2473

cocollin23, 2023-11-02:
试过很多次了，问题依旧，我仔细看了adg的日志，发现是占用了53端口，关闭adg后，53端口仍然是被占用状态，所以导致dnsmasq不能正常启动！adg的配置文件被Bypass改过！

cocollin23, 2023-11-02:
已经搞清楚逻辑了，Bypass配置adg时，会直接修改adg用53端口劫持dnsmasq，导致dnsmasq不能正常启动dhcp服务