istoreos/istoreos #1091: 硬路由做有线ap，立马网速降一半怎么解决。

daowaiyaoyao, 2023-11-02:
### (1) 具体问题

A：我家是软路由（istore系统）拨号，带宽是电信1000M。然后硬路由有线接软路由做AP，所有硬件设备均是2.5口，万兆网线没问题。

问题来了∶硬路由做ap时无线只能跑一半网速，协商速度2.4G。 网线直连软路由可以跑满。用硬路由直接拨号无线有线都能跑满，可是只要用硬路由做AP，无线立马降速一半都多。
硬路由我都换了三个了，都没用。怎么办？

[image]

jjm2473, 2023-11-02:
看看网口是不是都协商到千兆以上了，还有你说的2.4G不会是频段吧？

daowaiyaoyao, 2023-11-02:
> 看看网口是不是都协商到千兆以上了，还有你说的2.4G不会是频段吧？

无线协商速率是2400M，不是WIFI频段，我不会犯这种低级错误，哈哈。

网口协商速率没问题，全是2500M.
[image]

daowaiyaoyao, 2023-11-02:
> 看看网口是不是都协商到千兆以上了，还有你说的2.4G不会是频段吧？

怪就怪在，我用硬路由直接拨号，无线直接跑满。 只要我把硬路由连接软路由做AP，无线速度直接打折三分之一。 然后我用软路由网线直连速度也能跑满。 彻底给我整蒙了！

jjm2473, 2023-11-07:
有没有试过安装homebox内网测速

daowaiyaoyao, 2023-11-09:
> 有没有试过安装homebox内网测速

安装了，而且测试内网速度嚄没问题。

jjm2473, 2023-11-09:
@daowaiyaoyao 你有使用翻墙插件吗？

daowaiyaoyao, 2023-11-11:
> @daowaiyaoyao 你有使用翻墙插件吗？

有，但是我测速都是关闭翻墙软件以后测速的。
我用无线路由器拨号就没问题，但是用软路由拨号，无线路由器做ap网速衰减的特别厉害。

z4cki, 2024-03-20:
@daowaiyaoyao 解决了吗大佬，同样的问题。软路由当主路由，其他两台硬路由ap，实测有线跑满，无线最多400+

daowaiyaoyao, 2024-03-31:
刚看见邮件，我还被这个问题困扰呢。  硬路由拨号无线跑满，软路由拨号有线跑满。只要硬路由做ap，就跑不满。有人说是nat的问题，唉迷茫了。发送自我的盖乐世
-------- 原始信息 --------发件人： Zhekai Ji ***@***.***> 日期: 2024/3/21  上午12:57  (GMT+08:00) 收件人： istoreos/istoreos ***@***.***> 抄送： daowaiyaoyao ***@***.***>, Mention ***@***.***> 主题： Re: [istoreos/istoreos] 硬路由做有线ap，立马网速降一半怎么解决。 (Issue #1091) 
@daowaiyaoyao 解决了吗大佬，同样的问题。软路由当主路由，其他两台硬路由ap，实测有线跑满，无线最多400+

jjm2473, 2024-04-03:
有没有尝试关掉wifi6，160Mhz这些功能？有些老设备对这些新特性兼容性不好。

yameizi001, 2024-11-05:
哎，我也遇到了一模一样的问题。硬路由直接拨号，有线和wifi都能跑满千兆；软路由istoreos拨号，硬路由做有线ap，wifi直接掉到三分之一，只能跑到三百兆左右。关键是电脑有线直连软路由可以跑满千兆，但是硬路由wifi就是只能两三百兆，懵逼了。

yameizi001, 2024-11-05:
换了两个硬路由都是一样的情况

z4cki, 2024-11-05:
已更换软路由，稳稳破千兆

yameizi001, 2024-11-05:
> 已更换软路由，稳稳破千兆

hello，你是换了软路由系统不用istoreos了吗？

z4cki, 2024-11-05:
> > 已更换软路由，稳稳破千兆
> 
> hello，你是换了软路由系统不用istoreos了吗？

更换了软路由硬件，原来是j1900，现在是N5105，仍然使用的是istoreos，架设在PVE上

yameizi001, 2024-11-05:
> 更换了软路由硬件，原来是j1900，现在是N5105，仍然使用的是istoreos，架设在PVE上

我是n100，直接物理跑的istoreos，内网也测速了找不到毛病，真奇怪啊